EttusResearch/uhd #654: Cannot Salt Minion cannot start N320 running 4.3.0.0

RyanChaosInc, 2023-01-18:
## Issue Description ##
Cannot Salt Minion cannot start N320 running 4.3.0.0

Per https://files.ettus.com/manual/page_usrp_n3xx.html#n3xx_rasm_salt
Salt minion should be preinstalled on image.  The following command should start the salt minion:

`systemctl start salt-minion`

This results in an error.  

Calling it directly I get the same error:
```
root@SDR1:/usr/lib/python3.7/site-packages/salt# salt-minion
/usr/lib/python3.7/site-packages/salt/utils/odict.py:25: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3,and in 3.9 it will stop working
  from collections import Callable
Traceback (most recent call last):
  File "/usr/bin/salt-minion", line 26, in <module>
    salt_minion()
  File "/usr/lib/python3.7/site-packages/salt/scripts.py", line 158, in salt_minion
    import salt.utils.process
  File "/usr/lib/python3.7/site-packages/salt/utils/process.py", line 375
    def run(self, async=False):
```

## Setup Details ##
ssh to N320.
Already used mender to install 4.3 on the N320.

```
root@SDR1:~# cat /etc/mender/artifact_info
artifact_name=v4.3.0.0_n3xx

```

## Expected Behavior ##
Salt minion should start with no errors.

## Actual Behaviour ##
Salt minion fails to start.

## Steps to reproduce the problem ##
ssh to N320

```
root@SDR1:~# systemctl start salt-minion
Job for salt-minion.service failed because the control process exited with error code.
See "systemctl status salt-minion.service" and "journalctl -xe" for details.
root@SDR1:~# systemctl status salt-minion.service
● salt-minion.service - The Salt Minion
   Loaded: loaded (/lib/systemd/system/salt-minion.service; disabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Thu 2022-09-15 15:16:42 UTC; 5s ago
     Docs: man:salt-minion(1)
           file:///usr/share/doc/salt/html/contents.html
           https://docs.saltstack.com/en/latest/contents.html
  Process: 2693 ExecStart=/usr/bin/salt-minion (code=exited, status=1/FAILURE)
 Main PID: 2693 (code=exited, status=1/FAILURE)

Sep 15 15:16:42 SDR1 salt-minion[2693]:     salt_minion()
Sep 15 15:16:42 SDR1 salt-minion[2693]:   File "/usr/lib/python3.7/site-packages/salt/scripts.py", line 158, in salt_minion
Sep 15 15:16:42 SDR1 salt-minion[2693]:     import salt.utils.process
Sep 15 15:16:42 SDR1 salt-minion[2693]:   File "/usr/lib/python3.7/site-packages/salt/utils/process.py", line 375
Sep 15 15:16:42 SDR1 salt-minion[2693]:     def run(self, async=False):
Sep 15 15:16:42 SDR1 salt-minion[2693]:                       ^
Sep 15 15:16:42 SDR1 salt-minion[2693]: SyntaxError: invalid syntax
Sep 15 15:16:42 SDR1 systemd[1]: salt-minion.service: Main process exited, code=exited, status=1/FAILURE
Sep 15 15:16:42 SDR1 systemd[1]: salt-minion.service: Failed with result 'exit-code'.
Sep 15 15:16:42 SDR1 systemd[1]: Failed to start The Salt Minion.

```

## Additional Information ##
It appears that this is a very old Salt version.

```
root@SDR1:/usr/lib/python3.7/site-packages/salt# cat _version.py
# This file was auto-generated by salt's setup

from salt.version import SaltStackVersion

__saltstack_version__ = SaltStackVersion(2017, 7, 0, 0, '', 0, 0, None)

```

I question the compatibility of the old Salt version with Python 3.7.  Did the 4.3 upgrade also upgrade python to 3.7 which maybe broke Salt?  

Upgrading to the latest version of Salt may fix the compatibility problems.  
But, I'm surprised this wasn't caught in Ettus testing.

RyanChaosInc, 2023-04-21:
Same issue on UHD 4.4

mbr0wn, 2024-06-24:
This has been fixed in our upcoming UHD release. Thanks for reporting!

closed 2024-06-24 by mbr0wn