EttusResearch/uhd #660: usrp2_recovery.py error when running with Python 3.10.6

manvacamp2000, 2023-02-11:
## Issue Description ##
when running usrp2_recovery.py it fails because the structure pack does not properly use byte array

## Setup Details ##
Running on Ubuntu 22.04.1 LTS, Python 3.10.6, UHD-4.4

## Expected Behavior ##
sudo ./usrp2_recovery.py --ifc=enp0s9
Opening raw socket on interface: enp0s9
Loading packet with ip address: 192.168.10.2
Sending packet (22 bytes)
Done

## Actual Behaviour ##
sudo ./usrp2_recovery.py --ifc=enp0s9
Opening raw socket on interface: enp0s9
Loading packet with ip address: 192.168.10.2
Traceback (most recent call last):
  File "/usr/local/lib/uhd/utils/./usrp2_recovery.py", line 47, in <module>
    packet = struct.pack(
struct.error: argument for 's' must be a bytes object

## Steps to reproduce the problem ##

## Additional Information ##
replacing line 52:
IP_RECOVERY_CODE,
with:
bytes(IP_RECOVERY_CODE, 'utf-8'),

fixes the problem

shakthivels300, 2023-02-23:
Even I am facing the same problem with UHD 3.15LTS. Hope I get some help. 
Thanks

closed 2023-03-13 by atomita-ni

mbr0wn, 2023-03-15:
We fixed this on master. The fix will go public soon. You can grab it from master and run it on 3.15, too, once it is live.