EttusResearch/uhd #662: Python checking fail when build uhd

LawsonGu, 2023-02-28:
The first time when I build uhd with cmake command.
What is the meaning of this check failed?
[image]

NI-LAm, 2023-02-28:
From the list of missing packages I assume you want to build MPM. We recommend to install the corresponding SDK, You can get the lastest SDK using
`uhd_images_dowloader -t <your_device>_common_sdk`.
The SDK will be dowloaded to your images folder. Once downloaded make the downloaded .sh file executable and run it. When installed  run
`source /usr/local/oecore-x86_64/environment-setup-aarch64-oe-linux` (assuming you install to default location) to enable the SDK and run the make script in this shell again.

LawsonGu, 2023-02-28:
Hi, Lars
Maybe the question is as it’s my first build, I havn’t got uhd_images_downloader yet. Thus maybe these inform is not necessary.

INTERNAL - NI CONFIDENTIAL

NI-LAm, 2023-02-28:
If you want to build host from source you can find instructions here: https://files.ettus.com/manual/page_build_guide.html (but the dependency list above suggested you tried to build MPM. Can you post a list of commands you did?

LawsonGu, 2023-02-28:
cd uhd && git checkout v4.3.0.0 && cd host && mkdir build && cd build && cmake .. && make && make test && make install

NI-LAm, 2023-02-28:
This should work (given you have installed the dependencies mentioned in the build instructions I mentioned above). 

The error you got in the screenshot you posted will only occur when you try to build the MPM (gevent, pyroute2, pyudev, mprpc are only used in MPM).

LawsonGu, 2023-03-01:
I'm not trying to build the MPM. Maybe this error is not critical. We can just skip it.

NI-LAm, 2023-03-01:
Can you post the full output of the `cmake ..` command instead of a screenshot, please?

LawsonGu, 2023-03-02:
:$ cd uhd && git checkout v4.3.0.0 && cd host && mkdir build && cd build && cmake ..
Note: switching to 'v4.3.0.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 1f8fd3457 uhd: Prepare branch for 4.3.0.0 release
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
--
-- Configuring the Python interpreter...
-- Python interpreter: /usr/bin/python3.8 Version: 3.8.10
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
-- Python runtime interpreter: /usr/bin/python3.8 Version: 3.8.10
-- Override with: -DRUNTIME_PYTHON_EXECUTABLE=<path-to-python>
-- Finding Python Libraries...
-- Python Libraries: /usr/lib/x86_64-linux-gnu/libpython3.8.so
-- Python include directories: /usr/include/python3.8
-- Working off of feature or development branch. Updating version number.
-- Using UHD Images Directory: /usr/local/share/uhd/images
-- Build type not specified: defaulting to release.
-- Performing Test HAVE_VISIBILITY_HIDDEN
-- Performing Test HAVE_VISIBILITY_HIDDEN - Success
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN - Success
--
-- Configuring Boost C++ Libraries...
--
-- Checking for Boost version 1.65 or greater
--   Looking for required Boost components...
--   Enabling Boost Error Code Header Only
--     Disabling boost::asio use of std::string_view
--   Boost version: 1.71.0
--   Boost include directories: /usr/include
--   Boost library directories: /usr/lib/x86_64-linux-gnu
--   Boost libraries: Boost::chrono;Boost::date_time;Boost::filesystem;Boost::program_options;Boost::serialization;Boost::thread;Boost::unit_test_framework;Boost::system
-- Looking for Boost version 1.65 or greater - found
--
-- Python checking for compatible Python version
-- Python checking for compatible Python version - 3.8.10 satisifes minimum required version 3.6
--
-- Python checking for Mako templates module
-- Python checking for Mako templates module - 1.1.0 satisifes minimum required version 0.4.2
--
-- Python checking for requests module
-- Python checking for requests module - 2.22.0 satisifes minimum required version 2.0
[... 263 lines cut ...]
--   * N320
--   * E320
--   * E300
--   * X400
--   * OctoClock
--   * Manual
--   * API/Doxygen
--   * Man Pages
--
-- ######################################################
-- # UHD disabled components                             
-- ######################################################
--   * DPDK
--
-- ******************************************************
-- * You are building a development branch of UHD.
-- * These branches are designed to provide early access
-- * to UHD and USRP features, but should be considered
-- * unstable and/or experimental!
-- ******************************************************
-- Building version: 4.3.0.HEAD-0-g1f8fd345
-- Using install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lawgu/workarea/uhd/host/build

NI-LAm, 2023-03-02:
Lawson, you can ignore this. The cmake just told you that it cannot build the MPM simulator (because of the missing Python packages). Just continue with `make` `make install` and you're good to go.

LawsonGu, 2023-03-03:
Sure, thanks.

closed 2023-03-03 by LawsonGu