EttusResearch/uhd #666: Accessing GPIOs from RFNoC blocks

eltos, 2023-03-17:
## Issue Description ##

We would like to access the GPIOs on an USRP X310 via a custom OOT RFNoC block.
More specifically, we would like to read one pin on either of the GPIO banks (preferrably one of the Daugherboards) and manipulate the data stream of the RFNoC block with a custom FPGA implementation.

The RFNoC specification introduces the concept of `io_ports` which allow access to IO ports like Radio RX/TX, dram etc. However, the GPIOs are currently not exposed to that interface by the core. Thus, the RFNoC block can not access the GPIOs.

## Additional Information ##

It seems that there has been an implementation proposed to access the GPIOs via RFNoC quite a while ago for UHD 3:
- https://github.com/EttusResearch/uhd/pull/206
- https://github.com/EttusResearch/fpga/pull/28

However, it looks like this has not been merged, and is probably not compatible with the new UHD 4.
@mbr0wn could you comment on the current status, since you said that ["We're working on pulling a change close to this into master."](https://github.com/EttusResearch/uhd/pull/206#issuecomment-578250646)?

## Setup Details ##

- USRP X310 with LFRX and LFTX Daughterboards  
- UHD v4.4.0.0-23-g106afff88  
- GNU Radio v3.10.5.1

## Expected Behavior ##

GPIO interface exposed to RFNoC via the `io_ports` mechanism (RFNoC Specification Sec. 2.3.4), i.e. in
https://github.com/EttusResearch/uhd/blob/02558b69f73a03318424a52459c6f87d0fc6e18c/host/include/uhd/rfnoc/core/x310_bsp.yml#L20

Such that a custom RFNoC block can access this interface by likewise definition of it's `io_ports`

## Actual Behaviour ##

GPIOs are not accessible from an RFNoC block.

## Workaround ##

Following conversations from the mailing list [here](https://lists.ettus.com/empathy/thread/ZVODG4LMDOOHQF5CPEEL2BGENDZGGEWQ) and [also here](https://lists.ettus.com/empathy/thread/XGXXOFG2DDQSDK6C3P7DHUEPMYVJOP2I), it seems the current approach is to manually disconnect the GPIOs in the core FPGA sources and re-route them into the custom RFNoC block manually, by editing files inside this UHD Repository. 

## Related ##

- Issue https://github.com/EttusResearch/uhd/issues/605
  > io_ports aren't currently supported by rfnoc_create_verilog, so AXI ports have to be added manually.

eltos, 2025-03-05:
I've implemented GPIO access from RFNoC for UHD4.6 here: https://github.com/eltos/uhd/pull/1

Any interest from Ettus to merge these changes into UHD?

eltos, 2025-05-06:
Using GPIOs as input for trigger (burst transmission) and interlock signal (TX amplitude control).

julinodev, 2026-02-02:
@mbr0wn Would you mid referring to where the change was made/ if there is some documentation on it?