EttusResearch/uhd #673: Wrong python site-packages paths detected in CMake install on Mac M1

nzqo, 2023-04-08:
## Issue Description ##

I am trying to build and install uhd from source but am facing an error during the installation, specifically:

> CMake Error at host/build/python/cmake_install.cmake:41 (file):
>   file INSTALL cannot make directory
>   "/usr/local/../../../../../../lib/python3.11/site-packages/uhd": No such
>   file or directory.
> Call Stack (most recent call first):
>   host/build/cmake_install.cmake:66 (include)

## Setup Details ##

I am on a mac m1, Ventura 13.3 with CMake 3.26.3, python 3.11 (both installed as brew packages) and am trying to build from source on master. I tried setting the `PYTHONPATH` variable but that didnt help, I need to manually fix the paths in the generated `cmake_install` scripts.

The issue is already apparent in the configuration phase, specifically from the CMake output:

> --
> -- Utilizing the python install directory: /usr/local/../../../../../../lib/python3.11/site-packages
> --

And I can indeed confirm that the script used to find the site-packages (see host root `CMakeLists.txt`)

```python
import os,sysconfig
platlib = sysconfig.get_path(name='platlib')
prefix = sysconfig.get_config_var('prefix')
print(os.path.relpath(platlib, prefix))
```

produces this wrong path on my machine, specifically due to

```bash
>>> print(prefix)
/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11
>>> print(platlib)
/opt/homebrew/lib/python3.11/site-packages
```

This seems to be a general issue in portability. Without diving into this more deeply, why isnt the `prefix` (which is in fact just the platlib path) used directly as absolute path? Some more discussion can be found [on this SO post](https://stackoverflow.com/questions/122327/how-do-i-find-the-location-of-my-python-site-packages-directory)

## Expected Behavior ##

To have the uhd components installed properly, specifically the site-package directory `/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages` or equivalently `/opt/homebrew/lib/python3.11/site-packages` be recognized.

## Actual Behaviour ##

Python packages aren't installed properly due to false site-packages path being recognized

## Steps to reproduce the problem ##

I did nothing fancy, just

```bash
cmake -S host -B host/build -DCMAKE_BUILD_TYPE=DEBUG -DENABLE_TESTS=OFF
cmake --build host/build
sudo cmake --install host/build
```

## Additional Information ##

unrelated fyi: It seems that the provided [mailing list link](http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com) from the issue template is broken, I am getting a 404 from it.

mbr0wn, 2023-05-11:
@nzqo Did your build include c385d20eeea717b3859ac6a2bcc247b69fc66003? You may have missed it by a few days. That could be the solution.

mbr0wn, 2023-05-11:
(the code you mention in the top post was updated in that commit)

mbr0wn, 2023-05-12:
@nzqo there's often a delay between the commit date and when we push it to this public remote (we have an internal remote for CI etc.). Seems you got really unlucky based on the date though.

mbr0wn, 2023-05-12:
FYI think this is a dupe of https://github.com/EttusResearch/uhd/issues/628

nzqo, 2023-05-15:
> FYI think this is a dupe of https://github.com/EttusResearch/uhd/issues/628

Not sure about that, my issue came about without the use of a custom install path. But maybe related, havent taken that close of a look into the other issue..

Anyway, you were right. I did not have those changes because this issue does not persist and is fixed on the current master. I am hence closing this.

closed 2023-05-15 by nzqo