EttusResearch/uhd #674: error in building custom fpga includes rfnoc blocks

vahidrezaee, 2023-04-12:
## Issue Description ##

I want to build custom FPGA image that includes RFNoC block.
I have follow the  [USRP Build Documentation](http://files.ettus.com/manual/md_usrp3_build_instructions.html) in the UHD and USRP Manual.
that I have encountered this error 

## Setup Details ##

## Expected Behavior ##

## Actual Behaviour ##

WARNING: [Vivado_Tcl 4-391] The following IPs are missing output products for Synthesis target. These output products could be required for synthesis, please generate the output products using the generate_target or synth_ip command before running synth_design.
WARNING: [Vivado_Tcl 4-391] The following IPs are missing output products for Implementation target. These output products could be required for synthesis, please generate the output products using the generate_target or synth_ip command before running synth_design.
WARNING: [IP_Flow 19-2162] IP 'xge_pcs_pma' is locked:
ERROR: [Synth 8-439] module 'xge_pcs_pma' not found
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/xge_pcs_pma/xge_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/xge_pcs_pma/xge_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/xge_pcs_pma/xge_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/xge_pcs_pma/xge_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/xge_pcs_pma/xge_pcs_pma.xml'
ERROR: [Vivado 12-398] No designs are open
ERROR: [Common 17-69] Command failed: * IP definition '10G/25G Ethernet Subsystem (4.0)' for IP 'xge_pcs_pma' (customized with software release 2021.1) has a different revision in the IP Catalog.
[00:01:30] Current task: Synthesis +++ Current Phase: Starting
[00:01:30] Current task: Synthesis +++ Current Phase: Finished
[00:01:30] Process terminated. Status: Failure

## Steps to reproduce the problem ##

UHD 4.2.0
vivado 2021.2
ubuntu 20.4
## Additional Information ##

wordimont, 2023-05-01:
Sorry for the delayed response. From the log, it looks like you changed something to cause a version mismatch in the IP. This often happens when you try to use the wrong Vivado version. You could try something like this if you're still having a problem:

```
cd fpga/usrp3/top/<target folder>
git checkout v4.2.0.1
git reset --hard     # This revert any changes you made
source setupenv.sh   # This will make sure you have the right Vivado version and patches installed
make cleanall        # This deletes any IP you already generated
make <target>        # Build the target you want
```

vahidrezaee, 2023-05-30:
Hi,
thanks for your replay,
I install Vivado 2021.1 and included the patch.
but the Vivado has an issue with generating ddr4_64bits IP.

* GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)
* Python 3.8.10
* Vivado v2021.1_AR76780 (64-bit)
 
my machine is running Ubuntu 20.04.1 and is capable of compiling X310
builds successfully.

UHD version is "UHD 4.4.0.0-33-g4a77791c"

I searched the mail-list "usrp-user" and found  Mr "Dario Pennisi" had this issue before,
He said "I was actually able to spot the issue... the IP
version doesn't match the one in the Vivado version I have so it wouldn't
compile. I fixed it by manually upgrading the IP and everything worked as
expected. "
unfortunately, he doesn't explain more about it.
link of  mail-list archive [:](https://lists.ettus.com/empathy/thread/4CX5JYHAZQLKREOPS7B7X7NVNP4S5IDV?hash=OPDYF6IJRAKJPHU5J7JBVMSAFVLH7B7U#OPDYF6IJRAKJPHU5J7JBVMSAFVLH7B7U)

below is a dump of the failing bit:
`========================================================
BUILDER: Building IP ddr4_64bits
========================================================
BUILDER: Staging IP in build directory...
BUILDER: Reserving IP location: /home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits
BUILDER: Retargeting IP to part zynquplusRFSOC/xczu28dr/ffvg1517/-1/e...
BUILDER: Building IP...
[00:00:00] Executing command: vivado -mode batch -source /home/vahid/workarea/uhd/fpga/usrp3/tools/scripts/viv_generate_ip.tcl -log ddr4_64bits.log -nojournal
WARNING: [BD 5-699] No address segments matched 'get_bd_addr_segs -of_object /ilmb_cntlr/SLMB/Mem'
WARNING: [BD 5-699] No address segments matched 'get_bd_addr_segs -of_object /second_ilmb_cntlr/SLMB/Mem'
WARNING: [BD 5-699] No address segments matched 'get_bd_addr_segs -of_object /dlmb_cntlr/SLMB/Mem'
WARNING: [BD 5-699] No address segments matched 'get_bd_addr_segs -of_object /second_dlmb_cntlr/SLMB/Mem'
WARNING: [BD 5-699] No address segments matched 'get_bd_addr_segs -of_object /iomodule_0/SLMB/Reg'
WARNING: [BD 5-699] No address segments matched 'get_bd_addr_segs -of_object /iomodule_0/SLMB/IO'
[00:00:52] Current task: Initialization +++ Current Phase: Starting
[00:00:52] Current task: Initialization +++ Current Phase: Finished
[00:00:52] Executing Tcl: synth_design -top ddr4_64bits -part xczu28dr-ffvg1517-1-e -mode out_of_context
[00:00:52] Starting Synthesis Command
WARNING: [Synth 8-7071] port 'LMB_Rst' of module 'bd_97f3_dlmb_0' is unconnected for instance 'dlmb' [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:259]
WARNING: [Synth 8-7023] instance 'dlmb' of module 'bd_97f3_dlmb_0' has 25 connections declared, but only 24 given [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:259]
WARNING: [Synth 8-7071] port 'LMB_Rst' of module 'bd_97f3_ilmb_0' is unconnected for instance 'ilmb' [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:307]
WARNING: [Synth 8-7023] instance 'ilmb' of module 'bd_97f3_ilmb_0' has 25 connections declared, but only 24 given [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:307]
WARNING: [Synth 8-7071] port 'rsta_busy' of module 'bd_97f3_lmb_bram_I_0' is unconnected for instance 'lmb_bram_I' [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:375]
WARNING: [Synth 8-7071] port 'rstb_busy' of module 'bd_97f3_lmb_bram_I_0' is unconnected for instance 'lmb_bram_I' [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:375]
WARNING: [Synth 8-7023] instance 'lmb_bram_I' of module 'bd_97f3_lmb_bram_I_0' has 16 connections declared, but only 14 given [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:375]
WARNING: [Synth 8-7071] port 'Interrupt_Ack' of module 'bd_97f3_microblaze_I_0' is unconnected for instance 'microblaze_I' [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:392]
WARNING: [Synth 8-7023] instance 'microblaze_I' of module 'bd_97f3_microblaze_I_0' has 54 connections declared, but only 53 given [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:392]
WARNING: [Synth 8-7071] port 'interconnect_aresetn' of module 'bd_97f3_rst_0_0' is unconnected for instance 'rst_0' [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:446]
WARNING: [Synth 8-7071] port 'peripheral_aresetn' of module 'bd_97f3_rst_0_0' is unconnected for instance 'rst_0' [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:446]
WARNING: [Synth 8-7023] instance 'rst_0' of module 'bd_97f3_rst_0_0' has 10 connections declared, but only 8 given [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:446]
WARNING: [Synth 8-7071] port 'rsta_busy' of module 'bd_97f3_second_lmb_bram_I_0' is unconnected for instance 'second_lmb_bram_I' [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/bd_0/synth/bd_97f3.v:498]
[... cut ...]
WARNING: [Designutils 20-1567] Use of 'set_false_path' with '-hold' is not supported by synthesis. The constraint will not be passed to synthesis. [/home/vahid/workarea/uhd/fpga/usrp3/top/x400/build-ip/xczu28drffvg1517-1e/ddr4_64bits/par/ddr4_64bits.xdc:258]
[00:02:48] Current task: Synthesis +++ Current Phase: Handling Custom Attributes
[00:02:48] Current task: Synthesis +++ Current Phase: Finished
[00:02:48] Process terminated. Status: Failure

========================================================
Warnings:           53
Critical Warnings:  0
Errors:             0

`

vahidrezaee, 2023-05-30:
dear wade,
I did try ``make cleanall``` then `make`. it generate same issue.
please read this mail-list archive, in this post Mr Dario had faced same issue and fix it.
[link to archive with similar issue](https://lists.ettus.com/empathy/thread/4CX5JYHAZQLKREOPS7B7X7NVNP4S5IDV?hash=OPDYF6IJRAKJPHU5J7JBVMSAFVLH7B7U#OPDYF6IJRAKJPHU5J7JBVMSAFVLH7B7U)

vahidrezaee, 2023-07-08:
I discovered that the process of building a default FPGA image was consuming a large amount of RAM. To fix this issue, I increased the RAM capacity ( to 24GB RAM) of my virtual machine and this resolved the problem. Thank you for your assistance.

closed 2023-07-08 by vahidrezaee