EttusResearch/uhd #675: X310 warning

SpongeBobaa, 2023-04-14:
ubuntu 18.04 
uhd 3.15.0
gnuradio 3.7.13.4

I have received the following display when using X310.

[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.HEAD-0-gaea0e2de
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1302 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1317 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
[WARNING] [X300] Cannot update master clock rate! X300 Series does not allow changing the clock rate during runtime.
[INFO] [MULTI_USRP] 1) catch time transition at pps edge
[INFO] [MULTI_USRP] 2) set times next pps (synchronously)
[WARNING] [MULTI_USRP] At least one channel is using an external LO.Using a manual DSP frequency policy is recommended to ensure the same frequency shift on all channels.

How can I solve it ?

SpongeBobaa, 2023-04-14:
Sorry , how can I solve the warning ?

SpongeBobaa, 2023-04-19:
> @SpongeBobaa if this is still an issue, we need more info to help you.
> 
> * Are you using more than 1 X310, or just
> * What daughterboard/s in the X310 / each X310?
> * What is the actual full command being issued?
> 
> These are a good starting point. Have you considered emailing "[support@ettus.com](mailto:support@ettus.com)"? They might be able to help faster than here.

Thanks for your help @michaelld  

1. I am using one X310
2. I have two TwinRX in the X310
3. I am running a project of gr-doa named 'estimate_X310_TwinRX_constant_phase_offsets_and_save.grc' in Gnuradio.The project should be received by four channels, but my X310 has only one sub-board dual-channel receiving signal, and the dual -channel receiving light of the other sub -board is on, but it does not receive the signal. I have tested the receiving signals of all channels alone.

Thanks for your suggestion
 I will email "support@ettus.com" for some help.

michaelld, 2023-04-19:
Add me on CC to that email if you've sent it already: michael.dickens@ettus.com ... I'll see what I can do to help

closed 2023-04-19 by michaelld