EttusResearch/uhd #678: Issue with timed commands when initializing TX streamer

RoelandOsaer, 2023-05-11:
## Issue Description ##

When synchronizing multiple USRPs(b210) by using the ` usrp->set_time_next_pps(uhd::time_spec_t(0.0));` waiting two seconds to be sure the next pps arrives so every USRPshas its internal timer set to 0 seconds. Afterward, a timed command is set to toggle the gpio pins high after a certain time. Without initilazing the TX streamer on the TX-USRPs the toggle on the gpio pins is done at the exact same time. When the TX streamer is initialized (done before the synchronization) there is a shift in the toggle of the gpio pins between different USRPs. The lines of code that we think are giving the issue are marked in the TX script.

## Setup Details ##

tested with UHD version 4.3 and 4.1.0.6 (same issue). Custom FPGA version where only the PPS comes outside on gpio pin 7. 

code for RX and TX USRP are below.

Note: the USRPs are waiting for a synchronization signal from a server to start the synchronization on the next pps.

------------------------------------------------------------=
script for RX USRP:
------------------------------------------------------------=

    #include <zmq.hpp>
    #include <uhd/usrp/multi_usrp.hpp>
    #include <uhd/utils/safe_main.hpp>
    #include <uhd/utils/thread.hpp>
    #include <boost/format.hpp>
    #include <boost/program_options.hpp>
    #include <iostream>
    #include <fstream>
    #include <string>
    #include <chrono>
    #include <thread>
    #include <cmath>
    
    #define FMT_HEADER_ONLY
    #include <fmt/format.h>
    #include <fmt/ranges.h>
    
    namespace po = boost::program_options;
    
    zmq::context_t context(1); 

    void ready_to_go(std::string id, std::string server_ip)
    {
        zmq::socket_t socket(context, zmq::socket_type::req);

        std::cout << "Connecting to server..." << std::endl;
        socket.connect("tcp://"+server_ip+":5555");

        zmq::message_t request(id.size());
        memcpy(request.data(), id.data(), id.size());
        std::cout << "Sending ID "
                  << id
                  << "..." << std::endl;
        socket.send(request, zmq::send_flags::none);

        //  Get the reply.
        zmq::message_t reply;
        auto res = socket.recv(reply, zmq::recv_flags::none);
        std::cout << "Received" << std::endl;
    }
    
    void wait_till_go_from_server(std::string server_ip)
    {
        // TODO check if received message is SYNC
        //   Socket to receive messages on
        zmq::socket_t subscriber(context, ZMQ_SUB);
        subscriber.connect("tcp://"+server_ip+":5557");
        zmq_setsockopt(subscriber, ZMQ_SUBSCRIBE, "", 0);
        zmq::message_t msg;
        auto res = subscriber.recv(&msg);
        std::string msg_str = std::string(static_cast<char *>(msg.data()), msg.size());
        std::cout << "Received '" << msg_str << "'" << std::endl;
    }
    
[... 342 lines cut ...]
    
    
    hello_socket = context.socket(zmq.REP)
    hello_socket.bind("tcp://*:5555")
    
    
    while(True):
    	message = hello_socket.recv()
    	print(f"Node #{message.decode()} is ready")
    	hello_socket.send_string("OK")

	message = hello_socket.recv()
	print(f"Node #{message.decode()} is ready")
	hello_socket.send_string("OK")

	message = hello_socket.recv()
	print(f"Node #{message.decode()} is ready")
	hello_socket.send_string("OK")

	time.sleep(2)
	# Sends a string message
	start_socket.send_string("SYNC")
	print("SYNC")

## Additional Information ##

mbr0wn, 2023-07-12:
@RoelandOsaer You didn't specify the arguments you're using to call your scripts. I'm guessing you're not providing a `master_clock_rate` argument.

Here's why that matters: If you don't provide a `master_clock_rate` argument (MCR), then UHD will choose a master clock rate for you, dynamically. See the manual: https://uhd.readthedocs.io/en/latest/page_usrp_b200.html#b200_auto_mcr

The choice of MCR depends on two things: The sampling rate you want, and how many channels you are using. Because we don't know how many channels you need until you request a streamer, auto-MCR triggers very late in the game (in your script, when the TX streamer is created, but if you were to change the rate afterwards, then it would change the MCR again).

Because you are using 2 channels RX, and 1 channel TX, you will have different MCRs on both devices. UHD will even tell you in the logs (`Actually got clock rate XXX MHz`). If your devices are running at different MCRs, then various minor differences in the clocking architecture will lead to different delays. The difference between your USRPs is in the ns range, so that seems reasonable.

Try running both scripts with `master_clock_rate=16e6`.

mbr0wn, 2023-07-12:
FYI, I'll update the manual to reflect this (it's not obvious, I admit). I'll also close this. If you find that specifying the MCR does not help, please reopen.

closed 2023-07-12 by mbr0wn