EttusResearch/uhd #694: USRO B210 change gain type toNormalized

aldalahmeh, 2023-07-20:
## Issue Description ##

 I am using a USRP B210 with srsRAN 5G Project and COTS UE (a rooted android phone) to setup a 5G network. Of course, the B210 uses the UHD. However, using the standard configuration the transmitted power is around -125dBm, which is sufficient for the UE to detect. 

Just to make sure, I create a flowgraph in GNURadio for an AM transmitter with default USRP setting and got the same low power issue. The flowgrqaph pic is attached below. However, when I changed the RF options [Ch0 Fain Type: Normalized], I got up to -67dBm, see pic below. 

[image]

[image]

I found how to do that in the python code  by using
`self.uhd_usrp_sink_0.set_normalized_gain(1, 0)`

What is the way to do it in C or the C API provided with the UHD?

## Setup Details ##

## Expected Behavior ##

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

NI-LAm, 2023-07-21:
@aldalahmeh There is no `set_normalized_gain` in the UHD API (neither in C/C++ nor Python). But there are functions for RX and TX. Given your example I assume you are looking for the TX version. Just have a look into the [C API definition of UHD](https://github.com/EttusResearch/uhd/blob/0e50a37b5ec6039ed4e8947421dd333047e24408/host/include/uhd/usrp/usrp.h#L1107-L1110).
We also have some examples for the C API that can be found here https://github.com/EttusResearch/uhd/tree/master/host/examples (everthing with .c extension). This could be a good starting point to adapt for your use case.

I will close this as complete. Feel free to reopen, if that doesn't answer your question.

Regards Lars

closed 2023-07-21 by NI-LAm

aldalahmeh, 2023-07-22:
OK, found the C++ function "set_normalized_tx_gain" defined in the (uhd/host/lib/usrp/multi_usrp.cpp). It seems that the default value is 0, which is the smallest gain.