EttusResearch/uhd #695: 90 of 92 tests failed when doing "make test" to test my build. Returns and Error 8 as well.

Haider037, 2023-08-05:
## Issue Description ##

## Setup Details ##

## Expected Behavior ##

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

Haider037, 2023-08-05:
this is what i get when doing make

haider03@raspberrypi:~/uhd/host/build $ make
[  1%] Built target uhd_rpclib
[  4%] Built target uhd-resources
Scanning dependencies of target uhd
[  4%] Building CXX object lib/CMakeFiles/uhd.dir/convert/convert_with_neon.cpp.o
In file included from /home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h: In member function ‘virtual void __convert_fc32_1_sc16_item32_le_1_PRIORITY_SIMD::operator()(const input_type&, const output_type&, size_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h:6744:1: error: inlining failed in call to ‘always_inline’ ‘float32x4_t vdupq_n_f32(float32_t)’: target specific option mismatch
 6744 | vdupq_n_f32 (float32_t __a)
      | ^~~~~~~~~~~
/home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:27:33: note: called from here
   27 |     float32x4_t Q0 = vdupq_n_f32(float(scale_factor));
      |                      ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
In file included from /home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h:10848:1: error: inlining failed in call to ‘always_inline’ ‘void vst1_s16(int16_t*, int16x4_t)’: target specific option mismatch
10848 | vst1_s16 (int16_t * __a, int16x4_t __b)
      | ^~~~~~~~
/home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:56:17: note: called from here
   56 |         vst1_s16((reinterpret_cast<int16_t*>(&output[i + 6])), D15);
      |         ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h:9031:1: error: inlining failed in call to ‘always_inline’ ‘int16x4_t vrev32_s16(int16x4_t)’: target specific option mismatch
 9031 | vrev32_s16 (int16x4_t __a)
      | ^~~~~~~~~~
/home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:55:37: note: called from here
   55 |         int16x4_t D15   = vrev32_s16(D14);
      |                           ~~~~~~~~~~^~~~~
In file included from /home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h:7598:1: error: inlining failed in call to ‘always_inline’ ‘int16x4_t vqmovn_s32(int32x4_t)’: target specific option mismatch
 7598 | vqmovn_s32 (int32x4_t __a)
      | ^~~~~~~~~~
/home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:54:37: note: called from here
   54 |         int16x4_t D14   = vqmovn_s32(Q13);
      |                           ~~~~~~~~~~^~~~~
In file included from /home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h:7444:1: error: inlining failed in call to ‘always_inline’ ‘int32x4_t vcvtq_s32_f32(float32x4_t)’: target specific option mismatch
 7444 | vcvtq_s32_f32 (float32x4_t __a)
      | ^~~~~~~~~~~~~
/home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:53:40: note: called from here
   53 |         int32x4_t Q13   = vcvtq_s32_f32(Q11);
      |                           ~~~~~~~~~~~~~^~~~~
In file included from /home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h:1188:1: error: inlining failed in call to ‘always_inline’ ‘float32x4_t vmulq_f32(float32x4_t, float32x4_t)’: target specific option mismatch
 1188 | vmulq_f32 (float32x4_t __a, float32x4_t __b)
      | ^~~~~~~~~
/home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:52:36: note: called from here
   52 |         float32x4_t Q11 = vmulq_f32(Q10, Q0);
      |                           ~~~~~~~~~^~~~~~~~~
In file included from /home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h:10848:1: error: inlining failed in call to ‘always_inline’ ‘void vst1_s16(int16_t*, int16x4_t)’: target specific option mismatch
10848 | vst1_s16 (int16_t * __a, int16x4_t __b)
      | ^~~~~~~~
/home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:50:17: note: called from here
   50 |         vst1_s16((reinterpret_cast<int16_t*>(&output[i + 4])), D13);
      |         ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h:9031:1: error: inlining failed in call to ‘always_inline’ ‘int16x4_t vrev32_s16(int16x4_t)’: target specific option mismatch
 9031 | vrev32_s16 (int16x4_t __a)
      | ^~~~~~~~~~
/home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:49:36: note: called from here
   49 |         int16x4_t D13  = vrev32_s16(D12);
      |                          ~~~~~~~~~~^~~~~
In file included from /home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h:7598:1: error: inlining failed in call to ‘always_inline’ ‘int16x4_t vqmovn_s32(int32x4_t)’: target specific option mismatch
 7598 | vqmovn_s32 (int32x4_t __a)
      | ^~~~~~~~~~
/home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:48:36: note: called from here
   48 |         int16x4_t D12  = vqmovn_s32(Q9);
[... 273 lines cut ...]
      |                           ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:10:
[... cut ...]
      | ^~~~~~~~~
/home/haider03/uhd/host/lib/convert/convert_with_neon.cpp:29:36: note: called from here
   29 |         float32x4_t Q1  = vld1q_f32(reinterpret_cast<const float*>(&input[i]));
      |                           ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [lib/CMakeFiles/uhd.dir/build.make:553: lib/CMakeFiles/uhd.dir/convert/convert_with_neon.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1262: lib/CMakeFiles/uhd.dir/all] Error 2
make: *** [Makefile:182: all] Error 2

NI-LAm, 2023-08-10:
@Haider037 when compiling for RasPi the solution discussed in #468 should help.

mbr0wn, 2023-09-07:
@Haider037 yeah if you're doing custom compiles for custom archs, you need to make sure the build flags are all set correctly.

closed 2023-09-07 by mbr0wn