EttusResearch/uhd #696: I am building a UHD driver on a raspberry pi 3 B+ Model. When doing sudo make install it says "Building CXX object python/CmakeFiles/pyuhd.dir/pyuhd.cpp.o Then it says: c++: fatal error: killed signal terminated program cc1plus compilation terminated. make[2]: *** [python/cmakefiles/pyuhd.dir/build.make:82: python/Cmakefiles/pyuhd.dir/pyuhd.cpp.o] error 1 make[1]: *** [cmakefiles/makefiles2:5431: python/cmakefiles/pyuhd.dir/all] error2 make: *** [Makefile:182: all] Error 2

Haider037, 2023-08-13:
## Issue Description ##

## Setup Details ##

## Expected Behavior ##

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

Haider037, 2023-08-13:
The last thing it says during the make process before the error comes up is
"[ 98%] Building CXX object python/CMakeFiles/pyuhd.dir/pyuhd.cpp.o"
And it stays at that stage for a long time before it exits the code with the error mentioned above.

NI-LAm, 2023-08-15:
@Haider037 Could please provide some more details, like:
* What Distrubution is installed on the RasPi?
* Which uhd Version are you trying to build?
* What cmake command did you use for configuration?

Please also post full logs of the configure and build process.

Haider037, 2023-08-15:
1) Raspberry Pi OS (Raspbian)
2) Version 4.0.0
3) cmake -DCMAKE_CXX_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a53 -Wno-psabi" -DCMAKE_C_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a53 -Wno-psabi" -DCMAKE_ASM_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a53 -g" ../

The tags are on my raspberry pi and I am currently on my MacI have to run the cmake line above and it's taking a while, so I will try and post It on here if I get it working. But I am pretty sure all the builds pass until it gets to 98% like stated above.

Haider037, 2023-08-16:
Here are the tags 
After running this command: cmake -DCMAKE_CXX_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a53 -Wno-psabi" -DCMAKE_C_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a53 -Wno-psabi" -DCMAKE_ASM_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a53 -g" ../

...@raspberrypi:~/uhd/host/build $ cmake -DCMAKE_CXX_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a53 -Wno-psabi" -DCMAKE_C_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a53 -Wno-psabi" -DCMAKE_ASM_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a53 -g" ../
--
-- Configuring the Python interpreter...
-- Manually determining build Python version...
-- Python interpreter: /usr/bin/python3.9 Version: 3.9.2
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
-- Manually determining runtime Python version...
-- Python runtime interpreter: /usr/bin/python3.9 Version: 3.9.2
-- Override with: -DRUNTIME_PYTHON_EXECUTABLE=<path-to-python>
-- Finding Python Libraries...
-- Python Libraries: /usr/lib/arm-linux-gnueabihf/libpython3.9.so
-- Python include directories: /usr/include/python3.9
-- Operating on master branch.
-- Using UHD Images Directory: /usr/local/share/uhd/images
-- Build type not specified: defaulting to release.
--
-- Configuring Boost C++ Libraries...
--
-- Checking for Boost version 1.65 or greater
--   Looking for required Boost components...
--   Enabling Boost Error Code Header Only
--     Disabling boost::asio use of std::string_view
--   Boost version: 1.74.0
--   Boost include directories: /usr/include
--   Boost library directories: /usr/lib/arm-linux-gnueabihf
--   Boost libraries: Boost::chrono;Boost::date_time;Boost::filesystem;Boost::program_options;Boost::serialization;Boost::thread;Boost::unit_test_framework;Boost::system
-- Looking for Boost version 1.65 or greater - found
--
-- Python checking for compatible Python version
-- Python checking for compatible Python version - 3.9.2 satisifes minimum required version 3.6
--
-- Python checking for Mako templates module
-- Python checking for Mako templates module - 1.1.3 satisifes minimum required version 0.4.2
--
-- Python checking for requests module
-- Python checking for requests module - 2.25.1 satisifes minimum required version 2.0
--
-- Python checking for numpy module
-- Python checking for numpy module - 1.19.5 satisifes minimum required version 1.11
--
-- Python checking for ruamel.yaml module
-- Python checking for ruamel.yaml module - "import ruamel.yaml" failed (is it installed?)
--
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = TRUE
--   Dependency HAVE_PYTHON_MODULE_MAKO = TRUE
--   Enabling LibUHD support.
--   Override with -DENABLE_LIBUHD=ON/OFF
--
-- Configuring LibUHD - C API support...
--   Dependency ENABLE_LIBUHD = ON
--   Enabling LibUHD - C API support.
--   Override with -DENABLE_C_API=ON/OFF
--
-- Configuring LibUHD - Python API support...
--   Dependency ENABLE_LIBUHD = ON
--   Dependency HAVE_PYTHON_MODULE_NUMPY = TRUE
--   Dependency HAVE_PYTHON_LIBS = TRUE
--   Enabling LibUHD - Python API support.
--   Override with -DENABLE_PYTHON_API=ON/OFF
--
-- Configuring Examples support...
--   Dependency ENABLE_LIBUHD = ON
--   Enabling Examples support.
--   Override with -DENABLE_EXAMPLES=ON/OFF
--
-- Configuring Utils support...
[... 336 lines cut ...]
[ 92%] Built target rfnoc_detailgraph_test
[ 92%] Built target octoclock_burn_eeprom
[ 92%] Built target b2xx_fx3_utils
[ 92%] Built target uhd_config_info
[ 93%] Built target uhd_usrp_probe
[ 94%] Built target uhd_find_devices
[ 94%] Built target uhd_image_loader
[ 94%] Built target uhd_adc_self_cal
[ 95%] Built target uhd_cal_rx_iq_balance
[ 95%] Built target uhd_cal_tx_dc_offset
[ 95%] Built target uhd_cal_tx_iq_balance
[ 95%] Built target usrp_burn_db_eeprom
[ 95%] Built target usrp_burn_mb_eeprom
[ 95%] Built target converter_benchmark
[ 95%] Built target query_gpsdo_sensors
[ 95%] Built target fx2_init_eeprom
[ 97%] Built target man_page_gzips
[ 98%] Built target doxygen_docs
[ 98%] Building CXX object python/CMakeFiles/pyuhd.dir/pyuhd.cpp.o

Then after a long time of the machine staying at 98% it crashes and spits this out 

"Building CXX object python/CmakeFiles/pyuhd.dir/pyuhd.cpp.o Then it says: c++: fatal error: killed signal terminated program cc1plus compilation terminated. make[2]: *** [python/cmakefiles/pyuhd.dir/build.make:82: python/Cmakefiles/pyuhd.dir/pyuhd.cpp.o] error 1 make[1]: *** [cmakefiles/makefiles2:5431: python/cmakefiles/pyuhd.dir/all] error2 make: *** [Makefile:182: all] Error 2

I hope this has everything you need.

NI-LAm, 2023-08-16:
So, your config looks good. From the error message you get and the RasPi you are using I assume you run out of memory during compilation (pyuhd.cpp is huge, it contains the full Python interface for UHD). If that's the case, you have several options:
* remove the Python interface if you don't need it (run cmake with `DENABLE_PYTHON_API=OFF`)
* increase the swap size on your PI (https://linuxhint.com/increase-memory-raspberry-pi/), this will further slow down your compilation process on the RasPI, because swap is so much slower than RAM.
* cross compile (e.g. using this https://github.com/tttapa/docker-arm-cross-toolchain) might be more effort at the beginning but compile times will reduce drastically.

As a side node. When building UHD yourself for this specific platform you might want to disable components you don't need (e.g. disable all the devices you do not need support for).

Haider037, 2023-08-17:
Okay after removing the python interface everything installed and make passed with no errors. But now when I am creating my code on gnu radio and running it on my pi this is what comes up 

Generating: '/home/.../uhd/host/build/fmReceiver.py'

Executing: /usr/bin/python3 -u /home/.../uhd/host/build/fmReceiver.py

[INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107100; UHD_3.15.0.0-4
[WARNING] [B200] EnvironmentError: IOError: Could not find path for image: usrp_b200_fw.hex

Using images directory: <no images directory located>

Set the environment variable 'UHD_IMAGES_DIR' appropriately or follow the below instructions to download the images package.

Please run:

 "/lib/arm-linux-gnueabihf/uhd/utils/uhd_images_downloader.py"
Traceback (most recent call last):
  File "/home/.../uhd/host/build/fmReceiver.py", line 263, in <module>
    main()
  File "/home/.../uhd/host/build/fmReceiver.py", line 239, in main
    tb = top_block_cls()
  File "/home/.../uhd/host/build/fmReceiver.py", line 83, in __init__
    self.uhd_usrp_source_0 = uhd.usrp_source(
  File "/usr/lib/python3/dist-packages/gnuradio/uhd/__init__.py", line 125, in constructor_interceptor
    return old_constructor(*args)
  File "/usr/lib/python3/dist-packages/gnuradio/uhd/uhd_swig.py", line 2791, in make
    return _uhd_swig.usrp_source_make(device_addr, stream_args, issue_stream_cmd_on_start)
RuntimeError: LookupError: KeyError: No devices found for ----->
Empty Device Address

>>> Done (return code 1)

I ran the exact same code on my MacBook and used the same usrp device and it connected/ worked perfectly. But on my pi this is what comes up.

Haider037, 2023-08-17:
I have installed all the images necessary, or at least I think I have and I see the files under the utils and the build directories

Haider037, 2023-08-17:
Yes it is connected to my Pi via usb and when I run uhd_find_devices it shows up with the serial number and everything as well

Haider037, 2023-08-17:
are you sure I will need the interface? because I have seen it work without I think, unless I am mistaken.

Haider037, 2023-08-18:
I re installed the uhd with the python interface but it still says that it cant read the device on gnu radio, but when I run uhd_find_devices or uhd_usrp_probe the device comes up.

Haider037, 2023-08-18:
Im using the companion

NI-LAm, 2023-08-18:
Is the companion running on the Pi or on your Mac?

Haider037, 2023-08-18:
pi all of this is on my pi

NI-LAm, 2023-08-21:
The GRC script says it tries to load a 3.15 UHD version. Is that the one you have build on your Pi? How did install the companion? Did you compiled it by youself or via package manager? In both cases can you share some details about the versions you use? You might have to tell GRC where to find your UHD binaries.

mbr0wn, 2024-06-21:
Summarizing this thread:
- RPi3 is not beefy enough to build Python API natively (cross-compiling may work)
- GNU Radio does not require Python API
- Looks like a UHD version mismatch on the Pi

closed 2024-06-21 by mbr0wn