EttusResearch/uhd #697: uhd_usrp_probe shows only two Antennas works

zzystudy, 2023-08-14:
## Issue Description ##

when I run `uhd_usrp_probe `, the result shows my USRP X310 only has two Antennas works. And when I run srsran gnb, I can not receive any signal in my phone, could someone help judge whether the USRP X310 is broken?

## Setup Details ##

- USRP X310
- All OSes (tested and confirmed Windows, Ubuntu and ArchLinux)
- uhd v4.3.0.0

## Expected Behavior ##

I can run srsran gnb and can receive signal in my phone

## Actual Behaviour ##

I can run srsran gnb but can not receive any signal in my phone.

## Additional Information ##

the result of `uhd_usrp_probe` is:
```
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.3.0.HEAD-0-g1f8fd345
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [GPS] No GPSDO found
[INFO] [X300] Radio 1x clock: 200 MHz
  _____________________________________________________
 /
|       Device: X-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: X310
|   |   revision: 8
|   |   revision_compat: 7
|   |   product: 30818
|   |   mac-addr0: 00:80:2f:23:29:2f
|   |   mac-addr1: 00:80:2f:23:29:30
|   |   gateway: 192.168.10.1
|   |   ip-addr0: 192.168.10.102
|   |   subnet0: 255.255.255.0
|   |   ip-addr1: 192.168.20.2
|   |   subnet1: 255.255.255.0
|   |   ip-addr2: 192.168.30.2
|   |   subnet2: 255.255.255.0
|   |   ip-addr3: 192.168.40.2
|   |   subnet3: 255.255.255.0
|   |   serial: 30B33BD
|   |   FW Version: 6.0
|   |   FPGA Version: 39.0
|   |   FPGA git hash: c781f75
|   |   RFNoC capable: Yes
|   |   
|   |   Time sources:  internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|     _____________________________________________________
|    /
|   |       RFNoC blocks on this device:
|   |   
|   |   * 0/DDC#0
|   |   * 0/DDC#1
|   |   * 0/DUC#0
|   |   * 0/DUC#1
|   |   * 0/Radio#0
|   |   * 0/Radio#1
|   |   * 0/Replay#0
|     _____________________________________________________
|    /
|   |       Static connections on this device:
|   |   
|   |   * 0/SEP#0:0==>0/DUC#0:0
|   |   * 0/DUC#0:0==>0/Radio#0:0
|   |   * 0/Radio#0:0==>0/DDC#0:0
|   |   * 0/DDC#0:0==>0/SEP#0:0
|   |   * 0/Radio#0:1==>0/DDC#0:1
|   |   * 0/DDC#0:1==>0/SEP#1:0
|   |   * 0/SEP#2:0==>0/DUC#1:0
|   |   * 0/DUC#1:0==>0/Radio#1:0
|   |   * 0/Radio#1:0==>0/DDC#1:0
|   |   * 0/DDC#1:0==>0/SEP#2:0
|   |   * 0/Radio#1:1==>0/DDC#1:1
|   |   * 0/DDC#1:1==>0/SEP#3:0
|   |   * 0/SEP#4:0==>0/Replay#0:0
|   |   * 0/Replay#0:0==>0/SEP#4:0
|   |   * 0/SEP#5:0==>0/Replay#0:1
|   |   * 0/Replay#0:1==>0/SEP#5:0
|     _____________________________________________________
|    /
|   |       TX Dboard: 0/Radio#0
|   |   ID: SBX-120 (0x0082)
|   |   Serial: 308FFBE
|   |     _____________________________________________________
|   |    /
|   |   |       TX Frontend: 0
|   |   |   Name: SBX-120 TX
|   |   |   Antennas: TX/RX, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 400.000 to 4400.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   Bandwidth range: 120000000.0 to 120000000.0 step 0.0 Hz
|   |   |   Connection Type: QI
|   |   |   Uses LO offset: No
|     _____________________________________________________
|    /
|   |       RX Dboard: 0/Radio#0
|   |   ID: SBX-120 (0x0083)
|   |   Serial: 308FFBE
|   |     _____________________________________________________
|   |    /
|   |   |       RX Frontend: 0
|   |   |   Name: SBX-120 RX
|   |   |   Antennas: TX/RX, RX2, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 400.000 to 4400.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   Bandwidth range: 120000000.0 to 120000000.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No
|     _____________________________________________________
|    /
|   |       TX Dboard: 0/Radio#1
|   |   ID: SBX-120 (0x0082)
|   |   Serial: 308FFC0
|   |     _____________________________________________________
|   |    /
|   |   |       TX Frontend: 0
|   |   |   Name: SBX-120 (0x0082) - 0
|   |   |   Antennas: 
|   |   |   Sensors: 
|   |   |   Freq range: 0.000 to 0.000 MHz
|   |   |   Gain Elements: None
|   |   |   Bandwidth range: 0.0 to 0.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No
|     _____________________________________________________
|    /
|   |       RX Dboard: 0/Radio#1
|   |     _____________________________________________________
|   |    /
|   |   |       RX Frontend: 0
|   |   |   Name: Unknown (0xffff) - 0
|   |   |   Antennas: 
|   |   |   Sensors: 
|   |   |   Freq range: 0.000 to 0.000 MHz
|   |   |   Gain Elements: None
|   |   |   Bandwidth range: 0.0 to 0.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No

```

snehasish1991, 2023-08-30:
you can use the below command to verify if your usrp is working or not.

 ./benchmark_rate --rx_rate 10e6 --tx_rate 10e6

It will create a loopback and display the number of trasmitted and received samples, along with other messages like number of dropped samples or timeout. 

Please share the output of the above command.

mbr0wn, 2023-09-07:
This part does look like a bad board:
```
|     _____________________________________________________
|    /
|   |       TX Dboard: 0/Radio#1
|   |   ID: SBX-120 (0x0082)
|   |   Serial: 308FFC0
|   |     _____________________________________________________
|   |    /
|   |   |       TX Frontend: 0
|   |   |   Name: SBX-120 (0x0082) - 0
|   |   |   Antennas: 
|   |   |   Sensors: 
|   |   |   Freq range: 0.000 to 0.000 MHz
|   |   |   Gain Elements: None
|   |   |   Bandwidth range: 0.0 to 0.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No
|     _____________________________________________________
|    /
|   |       RX Dboard: 0/Radio#1
|   |     _____________________________________________________
|   |    /
|   |   |       RX Frontend: 0
|   |   |   Name: Unknown (0xffff) - 0
|   |   |   Antennas: 
|   |   |   Sensors: 
|   |   |   Freq range: 0.000 to 0.000 MHz
|   |   |   Gain Elements: None
|   |   |   Bandwidth range: 0.0 to 0.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No
```

I recommend making sure that the SBX is properly plugged in (tighten the screws, but not too much). Then, you can use `usrp_burn_db_eeprom` to communicate with the dboard EEPROM. If that works, but the dboard doesn't come up as expected, you might be able to write the PID `0x0083` to the EEPROM and cross your fingers.

Other than that, I'm afraid your slot-B SBX might be busted. Not much else we can do from here -- closing this.

closed 2023-09-07 by mbr0wn

PHDsaoudi, 2023-09-29:
Hi;
Please i need to know how to connect USRP X310 to my computer and configure it