EttusResearch/uhd #698: X310 Firmware (not FPGA Image) Load

johnwstanford, 2023-08-27:
I'm trying to figure out how to update the firmware on the X310 and I can't find it documented anywhere. I'm not talking about the FPGA image. That process is well-documented. I'm talking about the code that runs on the ZPU:

https://github.com/EttusResearch/uhd/blob/master/firmware/usrp3/x300/x300_main.c

Is this stored in the EEPROM?  Is there a way to access wherever it's stored over JTAG?  Can you do a non-persistent load like you can do for the FPGA image?  If you point me in the right direction to get started, I can figure out the details and I'm happy to update the documentation with whatever I find out. Thanks

John

wordimont, 2023-08-27:
The firmware is part of the FPGA bitstream. See here for details:

https://kb.ettus.com/X300/X310#Firmware

johnwstanford, 2023-08-27:
Thanks!  That's exactly what I was looking for

closed 2023-09-07 by mbr0wn