EttusResearch/uhd #709: Source build failure for UHD 4.5.0.0 on Windows when enabling Python API

icyveins7, 2023-10-19:
## Issue Description ##
When enabling Python API during the source build, the post-build event trigger fails to properly build/copy the Python API files & libraries.

```
2>Including packages in pyuhd: []
2>E:\Python\Python311\envs\main\Lib\site-packages\setuptools\_distutils\cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
2>!!
2>
2>        ********************************************************************************
2>        Please avoid running ``setup.py`` directly.
2>        Instead, use pypa/build, pypa/installer or other
2>        standards-based tools.
2>
2>        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
2>        ********************************************************************************
2>
2>!!
2>  self.initialize_options()
2>E:\Python\Python311\envs\main\Lib\site-packages\setuptools\_distutils\cmd.py:66: EasyInstallDeprecationWarning: easy_install command is deprecated.
2>!!
2>
2>        ********************************************************************************
2>        Please avoid running ``setup.py`` and ``easy_install``.
2>        Instead, use pypa/build, pypa/installer or other
2>        standards-based tools.
2>
2>        See https://github.com/pypa/setuptools/issues/917 for details.
2>        ********************************************************************************
2>
2>!!
2>  self.initialize_options()
2>EXEC : warning : install_lib: 'build\lib' does not exist -- no Python modules to install
2>Done building project "INSTALL.vcxproj".
========== Build: 2 succeeded, 0 failed, 139 up-to-date, 0 skipped ==========
========== Build started at 5:51 pm and took 01.433 seconds ==========
```

Note that the build is considered to have succeeded, but no UHD related .py files / libraries will actually be installed.

## Setup Details ##
Source for 4.5.0.0 downloaded directly from 'Releases' page.
CMake 3.27.7
Python 3.11 (Virtual environment)
Visual Studio 2022
Windows 10

## Expected Behavior ##
UHD python libraries are installed properly into the virtual environment (or somewhere at least).

## Actual Behaviour ##
Error manifests as only a warning as above. See Additional Information for my hotfix.

## Steps to reproduce the problem ##
Source for 4.5.0.0 downloaded directly from 'Releases' page.
Build with Python API enabled.

## Additional Information ##
The fix in this case was to simply run the post-build event myself. As seen in ```host\build\python\cmake_install.cmake```, which is run as part of the post-build event, it tries to do

```
E:/Python/Python311/envs/main/Scripts/python.exe E:/uhd-4.5.0.0/host/build/python/setup.py -q install --force
```

The outermost ```INSTALL.vcxproj``` attempts to do this from the outermost ```host\build``` folder. I stepped into the ```host\build\python``` folder (where there is a ```build\lib``` folder containing the .py files & libs), and then ran the above command again. This worked and successfully wrote a ```uhd``` folder into my ```E:\Python\Python311\envs\main\Lib\site-packages\uhd-4.5.0-py3.11.egg``` folder.

I suspect the outermost ```INSTALL.vcxproj``` is not correctly stepping down into the subfolder before running the command.
However, this doesn't fix everything. Attempts to ```import uhd``` are met with 

```python
ImportError: cannot import name 'libpyuhd' from partially initialized module 'uhd' (most likely due to a circular import) (E:\Python\Python311\envs\main\Lib\site-packages\uhd-4.5.0-py3.11.egg\uhd\__init__.py)
```

The fix for this was to manually copy ```host\build\python\uhd\libpyuhd.pyd``` into my ```E:\Python\Python311\envs\main\Lib\site-packages\uhd-4.5.0-py3.11.egg``` folder.  I am not sure what went wrong along the builds for this to not have been copied into the ```host\build\python\build\lib\uhd``` folder like the rest of the ```.py``` files.

Then the consequent error from ```import uhd``` became 

```python
ImportError: DLL load failed while importing libpyuhd: The specified module could not be found.
```

which is solved on Windows by doing the standard

```
os.add_dll_directory("path\to\install\bin")
```
where my ```uhd.dll``` file is.

1) Need a fix to properly move the .pyd file.
2) Need a fix to properly run the setup.py file in the python subfolder so that it actually installs to site-packages.
I suggest also reworking the build to no longer use ```setup.py install``` as it is already deprecated.

lokkelvin2, 2024-05-24:
I am facing a similar issue in UHD 4.6.0.0-166-g041eef34, Windows 10, Python 3.10.

The `libpyuhd.pyd` isn't copied into site-packages because setup.py expects a Unix `*.so` file and not a windows `*.pyd` file.

https://github.com/EttusResearch/uhd/blob/041eef3472e0440730708053d47b1fb7b793c682/host/python/setup.py.in#L31-L35

My workaround is to change line 32 from `'uhd': ['*.so'],` to `'uhd': ['*.so', '../path/to/Release/libpyuhd.pyd'],` so that `libpyuhd.pyd` is installed into site-packages.

cfe-NI, 2025-02-07:
@icyveins7 
With [UHD Release 4.8.0.0](https://github.com/EttusResearch/uhd/releases/tag/v4.8.0.0) we changed the Windows build and installation flow for the UHD Python API.

The source build will not fail anymore, as described in this issue. As a post build step you currently need to install the binary distribution package (wheel) manually using the following commands:
```
(venv) cd <build directory>\python
(venv) python -m poetry install
```
Alternatively you can use:
```
(venv) python -m pip install <build directory>\dist\uhd-4.8.0.0-cp3xx-cp3xx-win_amd64.whl
```
Please also refer to https://files.ettus.com/manual/page_python.html#python_install_windows

Please also note that we made the Python portion of UHD available on [pypi.org](https://pypi.org/project/uhd/) as binary distribution package (wheel) to install this binary packages please follow this link: https://files.ettus.com/manual/page_install.html#install_win_python

For Linux the installation flow is still as it was before. Longer term we would like to harmonize both build flows for Python.

closed 2025-02-07 by cfe-NI