EttusResearch/uhd #711: rx_stream for 8 bit samples not working all the times when uisng c++

MikoMikarro, 2023-11-02:
## Issue Description ##
I'm creating a monitoring tool for radios and when I try to do some recordings using the sc8 format sometimes I just receive 0s, after I do a recording using sc16 the sometimes the sc8 works fine.

## Setup Details ##
I'm using the binary installation of the UHD 4.5.0.0-release and happened to me too with the 4.3-pre-release. I've tried it on a B205 mini-i and a X300, in both happens the same problem. sc16 works fine, fc32 and fc64 too, but when I use OTW sc8 and CPU format sc8 doesn't work reliably.

## Expected Behavior ##
When you configure the rx_streamer to use sc8 I expect to receive sc8 samples.
## Actual Behavior ##
Receive full array of 0s.

## Steps to reproduce the problem ##
Using the rx_to_file.exe example you can recreate this behavior. Just use big sample rates and you can see that some of the recordings you do are just full of 0s. 

## Additional Information ##

mbr0wn, 2024-06-21:
`sc8` is not currently supported on X3x0 devices. Is it possible that it works for B2xx, and not for X3xx?

mbr0wn, 2025-02-17:
Hi,

I was not able to reproduce this, and therefore will close it. Some info that might be helpful:
- sc8 is only supported on N2x0/USRP2 and B200 series, not X300.
- `rx_samples_to_file` (assuming that's the example that you're referring to) does not natively support CPU format sc8, only wire format sc8. I tried reproducing this with wire format sc8, and CPU format sc8/sc16/fc32, but couldn't see any issues.

If you can provide some additional info to more reliably reproduce your issue, please let us know and you can reopen this. Until then this will be closed.

closed 2025-02-17 by mbr0wn