EttusResearch/uhd #712: Tx tuning bug in UHD 4.4 for X310/UBX following commit 682a358b

rkossler, 2023-11-13:
## Issue Description ##
UHD does not tune to the requested Tx frequency and reports wrong "actual frequency"

## Setup Details ##
X310/UBX-160.  UHD-4.4. Use "tx_waveforms" example or really any example that includes Tx tuning

## Expected Behavior ##
Tx tuning should go to requested frequency and report an actual frequency that is close to the requested frequency

## Actual Behaviour ##
For a requested tune frequency of 2450e6, UHD tune result reports actual frequency of 450e6.  Additionally, the RF output is not at the requested frequency (that it, this is not just a reporting error)

## Steps to reproduce the problem ##
Run the following with X310/UBX-160.  Take a look at the resulting "Actual Tx Freq"

$ tx_waveforms --rate 25e6 --freq 2450e6 

Creating the usrp device with: ...
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.4.0.0-33-g4a77791c
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.932
[INFO] [X300] Radio 1x clock: 200 MHz
Using Device: Single USRP:
  Device: X-Series Device
  Mboard 0: X310
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: UBX RX
  RX Channel: 1
    RX DSP: 1
    RX Dboard: B
    RX Subdev: UBX RX
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: UBX TX
  TX Channel: 1
    TX DSP: 1
    TX Dboard: B
    TX Subdev: UBX TX

Setting TX Rate: 25.000000 Msps...
Actual TX Rate: 25.000000 Msps...

Setting TX Freq: 2450.000000 MHz...
Setting TX LO Offset: 0.000000 MHz...
Actual TX Freq: 449.999998 MHz...

[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
Setting device timestamp to 0...
Checking TX: TXLO: locked ...
Press Ctrl + C to stop streaming...
^C
Done!

## Additional Information ##
This error was introduced in commit B6447234 (UBX: Add VCO band calibration and map access)

In order to show that this is not just a reporting error but an actual Tx tuning error, you can run the following with an RF loopback cable & 30dB attenuator between Tx-0 and Rx-1.  Take a look at the resulting capture file which shows a 2MHz tone for code versions prior to the bug and shows no tone for code versions after the bad commit

txrx_loopback_to_file --tx-rate 12.5e6 --rx-rate 12.5e6 --tx-freq 2450e6 --rx-freq 2450e6 --tx-gain 20 --rx-gain 20 --rx-channels 1 --nsamps 102400 --wave-type SINE --wave-freq 2000000 --file /media/ramfolder/usrp_samples.dat --settling 1

nzqo, 2023-11-26:
Any updates on this? I am facing the same issue. x300 reports a transmission frequency of -50 MHz

rkossler, 2023-11-26:
Try uhd 4.5 or 4.6 if you can. Otherwise, I think it is still broken on 4.4
such that you would need to manually add the fix (late commit) and then re
build

nzqo, 2023-11-26:
> Try uhd 4.5 or 4.6 if you can. Otherwise, I think it is still broken on 4.4 such that you would need to manually add the fix (late commit) and then re build
> […](#)

I tried on whatever the lattest version shipped with apt is today and had met the issue there. Id need to check the version but I assume its at least 4.6

rkossler, 2023-11-26:
For me, it worked fine on both 4.5 and 4.6

nzqo, 2023-11-27:
> For me, it worked fine on both 4.5 and 4.6
> […](#)

I just checked and the latest version on Ubuntu 22.04 seems to be `4.1.0.5-3`. I am honestly very confused as to why that would happen there.

mbr0wn, 2024-03-19:
If this is an issue with UHD 4.4 only, we won't be fixing it there. If it's still an issue in recent UHD versions, that's a different story. Thanks for the report.

gmeyerho, 2024-03-19:
This should be fixed by 996e7ab, which is in UHD 4.5 and addresses one of the changes made in the commit referenced that caused this issue (b644723). So this should only be an issue in UHD 4.4 as that was when the change that introduced this issue was committed.

closed 2024-03-19 by mbr0wn