EttusResearch/uhd #715: using vivado ip core in rfnocmod tool

vahidrezaee, 2023-12-20:
## Issue Description ##

## Setup Details ##

## Expected Behavior ##

 I am currently working on an X410 USRP platform. I am in the process of developing an out-of-tree (OOT) RFNoC block and would like to incorporate Vivado IP cores such as clock_wiz, binary counter, and others into my RFNoC module.

While exploring the `<uhd dir>/host/example/rfnoc-example` directory, I found the `gain_rfnoc` block, which utilizes a complex multiplier IP core. The example is insightful, but it relies solely on UHD's C++ API. I am keen to have my custom RFNoC block accessible through GNU Radio API.

In my research, I came across the RFNoC 4 Workshop presented by Jonathon Pendulum at GRCon 2020, where he demonstrated using `rfnocmodtool` to build a "gain" OOT RFNoC block. Following the build process and integrating it into the FPGA image allowed seamless use in GNU Radio. However, when attempting to add a complex multiplier IP core to this gain RFNoC block, I encountered numerous errors during the FPGA image building process.

Could you please provide guidance or direct me to a tutorial on how to effectively integrate Vivado IP cores into RFNoCModTool projects? Your assistance in resolving these challenges would be greatly appreciated.
## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

2444616578, 2024-01-11:
When I use rfnocmodtool to creat a self-build module, I did not modify Cmakelist.txt. When using  the command"rfnocmodttol add ", the Cmakelist is changed  automatically. I do not know whether you use the 'add' command ., If not I recommand it to you . If you have used it, I suppose you not to change Cmakelist.txt. Just follow the "get start with RFNoC" and "RFNoC 4.0", the two websites. And I suggest you that you should check you verilog.v file. You may have some unrealizable functions in it. Make sure that you have modified XXX.yml, nlock_XXX.yml, et. al. rightly

vahidrezaee, 2024-02-12:
I did something interesting.
I create a simple oot module by  rfnocmodtool.
when I want to build the FPGA image core, I use project mode by adding "GUI=1  CHECK=1" to make command,
after  vivado  opens I navigate to my oot module code and add vivado IP core as many as I want.
for example for my x410_200_rfnoc_image_core.yml
```
make  x410_200_rfnoc_image_core GUI=1 CHECK=1
```

closed 2024-02-12 by vahidrezaee

feyzaozdemir, 2024-05-28:
Hi @vahidrezaee 

If this method works, can you explain the steps in detail? Thank you.