EttusResearch/uhd #718: RFNoC4.0 endpoint buffer size

hakegit, 2024-01-11:
## Issue Description ##

## Setup Details ##

## Expected Behavior ##

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

I have some issues when using RFNoC4.0. When I was using RFNoC3.0, I needed to input a series of modules make by myself when building an image using IMAGE_BUILDER.py. When I am using RFNoC4.0, I need to input the modules in the image_core.yml. The problem is that my own modules need a large amount of BRAM in FPGA. When I am using RFNoC3.0, if I use more than four my own modules, building failure will arise, which is clearly due to insufficient BRAM. But when I use RFNoC4.0, I need to determind the buffer size of ENDPOINT (ep) in image_core.yml for each module  when creating an image .And the connection need to be written by myself ,too. So I don't know how the BRAM used by my module corresponds to the cache size specified by me for ep. That is, how do I set the cache size for ep so that when my own module is added between ddc and radio, and between radio and duc. Currently, I am constantly trying to set the cache size of different EPs to different sizes, which may lead to issues with OOOOO or UUUUUU. I hope developers can provide me with a detailed explanation of how to set the cache size when my own module requires a large BRAM (approximately 8000 complex floating numbers). The specific names of my two own modules are sig1 and sig2. Each module requires a cache area of 8192 complex floating-point numbers. 
My module connection methods are radio-sig2-ddc-ep and ep-duc-sig2-radio, with two identical links for sending and receiving. I would greatly appreciate it if I could receive guidance.

hakegit, 2024-01-25:
1

mbr0wn, 2024-02-16:
Hey @hakegit, I think you are misunderstanding how the buffering works in RFNoC 4. The individual RFNoC blocks have **zero** buffering in the NoC shell, only whatever they have internally. All buffering for streaming purposes is in the stream endpoint.

See here for example. The first two stream endpoints have 64kib buffer, and zero buffer, respectively: https://github.com/EttusResearch/uhd/blob/080b1baac3af63842fa598729ed5442b4a79285a/fpga/usrp3/top/x300/x310_rfnoc_image_core.yml#L15-L23

(zero buffer because EP1 is used for RX only)

The blocks have no buffers: https://github.com/EttusResearch/uhd/blob/080b1baac3af63842fa598729ed5442b4a79285a/fpga/usrp3/top/x300/x310_rfnoc_image_core.yml#L43-L55

(Note that the radio block, for example, has an internal buffer for collecting samples to packetize them, but the framework doesn't need to know that).

closed 2024-02-16 by mbr0wn