EttusResearch/uhd #722: SPI API for Python

berk-acikgoz, 2024-01-19:
We are developing a program to utilize the front-panel GPIO pins of USRP X410 for SPI applications. The examples written in C++ proved useful for most purposes. However, we would like to carry the SPI functionality over to Python, for better integration with our existing code base.

The UHD development manual points to the C++ documentation for equivalent Python APIs. However, the spi_getter_iface class does not exist in Python, neither does a module with equivalent functionality (like the gpio_power class in Python equivalent to the gpio_power_iface class in C++). None of the methods of the uhd.libpyuhd.rfnoc.mb_controller (which was how I got the gpio_power object) seem to have anything to do with SPI. Is SPI functionality already implemented in Python? If so, how can it be utilized to write user applications for front panel GPIO? If not, what are the current plans on releasing it on later versions of UHD?

Thanks in advance!

manderseck, 2024-02-13:
@berk-acikgoz Sorry for the late reply. The `spi_getter_iface` class is indeed not yet implemented in Python. It is in our backlog, it just wasn't prioritized yet. I'll add a note to the backlog item helping us to prioritize it but I cannot promise any specific release for this feature yet.

manderseck, 2026-11-16:
@berk-acikgoz It's been a while but in UHD 4.11 we finally included the SPI API for Python through the addition of discoverable features: https://github.com/EttusResearch/uhd/commit/538a2e3fb3f5122e53170221ae4fc779d0ef3049
The usage should be pretty much the same as in C++ as you can see in the Python SPI example: https://github.com/EttusResearch/uhd/blob/master/host/examples/python/spi.py
I'm closing this issue now. In case anything should still be missing please feel free to re-open or to open a new issue.

closed 2026-11-16 by manderseck