EttusResearch/uhd #726: ERROR: [IMPL 213-28] Failed to generate IP.

JANEYUBAI, 2024-02-01:
## Issue Description ##
I created the project in windows, and after a lot of IPs were passed, an error was suddenly reported.( I followed the build_instructions in the zip file.)

## Setup Details ##

UHD:fpga-UHD-3.15.LTS
vivado:2018.3 and patches
cygwin and all relevant plug-ins are installed.

## Expected Behavior ##
build project

## Actual Behaviour ##
ERROR: [IMPL 213-28] Failed to generate IP.

## Steps to reproduce the problem ##

INFO: [SYSC 207-301] Generating SystemC RTL for addsub_hls.
INFO: [VHDL 208-304] Generating VHDL RTL for addsub_hls.
INFO: [VLOG 209-307] Generating Verilog RTL for addsub_hls.
INFO: [IMPL 213-8] Exporting RTL as a Vivado IP.

****** Vivado v2018.3 (64-bit)
  **** SW Build 2405991 on Thu Dec  6 23:38:27 MST 2018
  **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source run_ippack.tcl -notrace
bad lexical cast: source type value could not be interpreted as target
    while executing
"rdi::set_property core_revision 2402010821 {component component_1}"
    invoked from within
"set_property core_revision $Revision $core"
    (file "run_ippack.tcl" line 877)
INFO: [Common 17-206] Exiting Vivado at Thu Feb  1 08:21:25 2024...
ERROR: [IMPL 213-28] Failed to generate IP.
command 'ap_source' returned error code
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [HLS 200-112] Total elapsed time: 76.999 seconds; peak allocated memory: 82.341 MB.
INFO: [Common 17-206] Exiting vivado_hls at Thu Feb  1 08:21:25 2024...
BUILDER: Releasing IP location: /cygdrive/e/fpga-UHD/usrp3/top/n3xx/build-ip/xc7z100ffg900-2/addsub_hls
make[1]: *** [/cygdrive/e/fpga-UHD/usrp3/lib/hls/addsub_hls/Makefile.inc:20: /cygdrive/e/fpga-UHD/usrp3/top/n3xx/build-ip/xc7z100ffg900-2/addsub_hls/solution/impl/ip/hdl/verilog] Error 1
make[1]: Leaving directory '/cygdrive/e/fpga-UHD/usrp3/top/n3xx'
make: *** [Makefile:138: N310_RFNOC_XG] Error 2

mbr0wn, 2024-02-29:
Hi @JANEYUBAI, we don't support 3.15 any longer so we won't be able to help much. This looks like a Vivado issue, so maybe clearing out your build dir, and retrying will do the trick.

closed 2024-02-29 by mbr0wn

JANEYUBAI, 2024-03-13:
Problem solved, thank you.