EttusResearch/uhd #729: UHD showing duplicate device for USRP X310

junkyardhokie, 2024-02-17:
## Bug Description
The uhd library is reporting a duplicate device for the USRP X310. There should only be a total of two radios reported, (1) USRP X310 and (1) USRP E310. The X310 does have two UBX160 daughterboards. 

### A simple program to discover the devices shows three devices:

``` ./uhd_bug 
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107600; UHD_4.6.0.0-0-g50fa3baa
========================================
"type": "x300"
"addr": "192.168.10.2"
"fpga": "HG"
"name": ""
"serial": "32B0751"
"product": "X310"
========================================
"type": "x300"
"addr": "192.168.10.2"
"fpga": "HG"
"name": ""
"serial": "32B0751"
"product": "X310"
========================================
"mgmt_addr": "10.48.30.104"
"type": "e3xx"
"product": "e310_sg3"
"serial": "32F34C4"
"name": "ni-e31x-32F34C4"
"fpga": "n/a"
"claimed": "False"
"addr": "10.48.30.104"
========================================
``` 

### Source code to generate the output:

``` #include <stdio.h>
#include <string>
#include <uhd.h>
#include <uhd/device.hpp>

int main() {
    try {
        // List devices
        uhd::device_addr_t hint;
        uhd::device_addrs_t devList = uhd::device::find(hint);

        // Display all properties
        printf("========================================\n");
        for (const auto& devAddr : devList) {
            for (const auto& key : devAddr.keys()) {
                printf("\"%s\": \"%s\"\n", key.c_str(), devAddr[key].c_str());
            }
            printf("========================================\n");
        }
    }
    catch (const std::exception& e) {
        fprintf(stderr, "ERROR: %s\n", e.what());
        return -1;
    }
    return 0;
}
``` 
## Configuration
Operating System: Ubuntu 20.04.6 LTS
UHD: UHD_4.6.0.0-0-g50fa3baa

AlexandreRouma, 2024-02-17:
I'll take responsibility for writing the test code. If there's an issue with the way it asks UHD for the device list, it's my fault.

Aang23, 2024-03-25:
I can confirm that issue. Same here, except I forgot report it earlier.

mbr0wn, 2024-10-18:
Thanks all. We have a fix on the way.

closed 2024-10-24 by joergho