EttusResearch/uhd #731: Calibration on X310 with WBX-120: NO CAL antenna

keukenrol, 2024-02-28:
## Issue Description ##

I am using an X310 with WBX-120 and TwinRX daughterboards. The transmission seems to have a lot of carrier leakage (when sending 0, there is still a frequency spike at the carrier frequency). I found out about calibration methods, which are supported by the TX board (WBX-120). https://files.ettus.com/manual/page_calibration.html

When running the uhd_usrp_probe command, I can see the board gives me the following info:

TX Frontend: 0

| | | Name: WBX-120 TX+GDB

| | | Antennas: TX/RX, CAL

| | | Sensors: lo_locked

| | | Freq range: 25.000 to 2200.000 MHz

| | | Gain range PGA0: 0.0 to 31.0 step 1.0 dB

| | | Bandwidth range: 120000000.0 to 120000000.0 step 0.0 Hz

| | | Connection Type: IQ

| | | Uses LO offset: No

However, when running uhd_cal_tx_iq_balance or uhd_cal_tx_dc_offset it gives me the following error:

Running calibration for WBX-120 TX+GDB

Daughterboard serial: [censored]

**Error: This board does not have the CAL antenna option, cannot self-calibrate.**

## Setup Details ##

UHD version 4.6.0.0
FPGA version of X310 is HG

## Expected Behavior ##
Calibration of TX (and RX) for both DC offset and IQ.

## Actual Behaviour ##
This board does not have the CAL antenna option, cannot self-calibrate.

## Steps to reproduce the problem ##

Using a X310 with a WBX-120 daughterboard, run the following command:
uhd_cal_tx_iq_balance --verbose --args="addr=192.168.10.2"
remark: i have the default IP from the HG fw version.

## Additional Information ##

manderseck, 2024-05-15:
@keukenrol Sorry for the late response. I don't have such a device at hand and I'm not too familiar with this product series, but from what the cal_utils do it seems like they check both TX and the RX antennas and if one of them fails (and only in that case) they come up with that error. In your snipped I only see the TX output. Can you confirm that the RX frontend reports to have a CAL antenna, too? That may bring us a little bit further.

keukenrol, 2024-05-15:
Hello @manderseck. This is the output of uhd_find_devices:
|     _____________________________________________________
|    /
|   |       TX Dboard: 0/Radio#1
|   |   ID: WBX-120 (0x0080)
|   |   Serial: 3263BB2
|   |   Revision: 2
|   |   ID: WBX + Simple GDB, WBX v3 + Simple GDB, WBX v4 + Simple GDB, WBX-120 + Simple GDB (0x004f)
|   |   Serial: 32617AF
|   |   Revision: 2
|   |     _____________________________________________________
|   |    /
|   |   |       TX Frontend: 0
|   |   |   Name: WBX-120 TX+GDB
|   |   |   Antennas: TX/RX, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 25.000 to 2200.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.0 step 1.0 dB
|   |   |   Bandwidth range: 120000000.0 to 120000000.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No
|     _____________________________________________________
|    /
|   |       RX Dboard: 0/Radio#1
|   |   ID: WBX-120, WBX-120 + Simple GDB (0x0081)
|   |   Serial: 3263BB2
|   |   Revision: 2
|   |     _____________________________________________________
|   |    /
|   |   |       RX Frontend: 0
|   |   |   Name: WBX-120 RX+GDB
|   |   |   Antennas: TX/RX, RX2, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 25.000 to 2200.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   Bandwidth range: 120000000.0 to 120000000.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No
It does show a CAL antenna option on the RX frontend.

manderseck, 2024-05-17:
Thanks for providing the full output. From that it looks like it should work, indeed. So to get further I'd take the `usrp_cal_utils.hpp` and add some print or log statements to see which antennas it thinks are available. As mentioned earlier, unfortunately I don't have such a device available with these particular daughterboards for testing and debugging it myself.

keukenrol, 2024-05-18:
> Thanks for providing the full output. From that it looks like it should work, indeed. So to get further I'd take the `usrp_cal_utils.hpp` and add some print or log statements to see which antennas [...]

I suppose I need to rebuild the whole uhd package to get the print statements working? Suggestions where to add them and what to print? Thanks!

manderseck, 2024-05-22:
@keukenrol Yes, rebuilding will be required, but if you built your current installation yourself already, then rebuilding from the same directory will be pretty fast.
To see which antennas it thinks it has when checking if the CAL antenna is among them I propose to add the following debug lines after line 244 ("set the antennas to cal"):
```
    auto rx_ant = usrp->get_rx_antennas();
    auto tx_ant = usrp->get_tx_antennas();
    std::cout << "RX antennas:" << std::endl;
    for (const auto& ant : rx_ant) {
        std::cout << "  " << ant << std::endl;
    }
    std::cout << "TX antennas:" << std::endl;
    for (const auto& ant : tx_ant) {
        std::cout << "  " << ant << std::endl;
    }
```
The error message that you get comes a few lines later, so this is the place where something seems to go wrong. Thanks for your effort!

keukenrol, 2024-08-09:
This is the output:

uhd_cal_tx_iq_balance

Creating the usrp device with: ,ignore_cal_file=1,ignore-cal-file=1...
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.HEAD-0-g50fa3baa
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
Running calibration for WBX-120 TX+GDB
Daughterboard serial: [CENSORED]
RX antennas:
  RX1
  RX2
TX antennas:
  TX/RX
  CAL
Error: This board does not have the CAL antenna option, cannot self-calibrate.

manderseck, 2024-08-09:
@keukenrol Thanks for getting back with this information. Interesting to see that usrp_probe reports a CAL antenna back for RX and TX while the getter doesn't find a CAL antenna in the RX path. Will check where this difference comes from.

keukenrol, 2024-09-04:
@manderseck Do you have any news on this? Thanks!

manderseck, 2024-09-04:
@keukenrol I'm sorry, still didn't find a WBX to test this with. It's EOL which is why it's hard to find them and why it's difficult to prioritize this. I've created a bug for this in our internal bug tracker, though.
What happens here is that there seem to be two sources of information about the available antennas. The RX path claims to not have a `CAL` antenna option while in theory it should have this. In case the query for available antenna names just returns a wrong list you could try what happens if you open a session to the device (for instance in Python to make things easy) and then set the rx antenna to `CAL`:
```
import uhd
usrp = uhd.usrp.MultiUSRP("addr=DEVICE_IP")
usrp.set_rx_antenna('CAL',0)
```
If that doesn't complain you could simply remove the check for available antenna ports in `usrp_cal_utils.hpp`, however my hopes are low.