EttusResearch/uhd #732: Can the N310 temporarily burn bit files via jtag?

JANEYUBAI, 2024-03-13:
Hello, I generated the N310 engineering code through vivado. Now I want to capture the signal by adding ila and then generate bit to download to usrp device via JTAG.
    However, after downloading, the device did not respond. The spf and serial port are not working, and the device cannot be connected.
I tried to download the official image to the device via jtag, and the same thing happened.
    Assuming I have no problem loading the official default image with uhd_image_loader.
    I would like to ask whether the N310 supports downloading custom bits with JTAG? If yes, is there any operation guide? If not, is there any other way to download the image with ila added to the device? Thanks for replying ~

mbr0wn, 2024-03-15:
In theory, yes, but in practice, no. That's because loading the bitfile needs to be synchronized with other operations, like unloading/loading kernel drivers.

If you want to load a bitfile with ILAs, just load it using the regular `uhd_image_loader`. The ILAs will then show up in Vivado hardware manager.

closed 2024-03-15 by mbr0wn

JANEYUBAI, 2024-03-15:
这是来自小马同学的自动回复邮件~
 
你的邮件已经收到~请不要着急,我一定会尽快回复您的!

JANEYUBAI, 2024-03-20:
> In theory, yes, but in practice, no. That's because loading the bitfile needs to be synchronized with other operations, like unloading/loading kernel drivers.
>
> If you want to load a bitfile with [...]

Thank you very much for solving my problem!Have a nice day!