EttusResearch/uhd #733: import uhd throws a error

patilraj1707, 2024-03-13:
## Issue Description ##
I installed the uhd library using the steps provided on the EttusResearch website. All the dependencies were also installed. I've also tried installing it using the alternavite method given in the website.
I've also tried the steps from this [link](https://github.com/EttusResearch/uhd/issues/259)
## Setup Details ##

## Actual Behaviour ##
Traceback (most recent call last):
  File "/home/raj/Downloads/DroneSecurity-public_squash/src/droneid_receiver_live.py", line 4, in <module>
    import uhd
ModuleNotFoundError: No module named 'uhd'

## Steps to reproduce the problem ##
sudo apt-get install autoconf automake build-essential ccache cmake cpufrequtils doxygen ethtool \
g++ git inetutils-tools libboost-all-dev libncurses5 libncurses5-dev libusb-1.0-0 libusb-1.0-0-dev \
libusb-dev python3-dev python3-mako python3-numpy python3-requests python3-scipy python3-setuptools \
python3-ruamel.yaml 

sudo add-apt-repository ppa:ettusresearch/uhd
sudo apt-get update
sudo apt-get install libuhd-dev uhd-host

Alternative methods tried
sudo apt-get install libuhd-dev uhd-host

wordimont, 2024-03-14:
You may also need `sudo apt install python3-uhd`. If that doesn't fix it, please check that your `PYTHONPATH` variable includes the location where the UHD site-packages for python were installed. You might gain some insight by searching the mailing list. https://lists.ettus.com/empathy/list/usrp-users.lists.ettus.com

mbr0wn, 2024-03-19:
`python3-uhd` is almost certainly required.

closed 2024-03-19 by mbr0wn