EttusResearch/uhd #734: RX TX Switching in B200 mini by GNU Radio

gourav-2002, 2024-03-18:
## Issue Description ##

I have an B200 mini and I try to "switch RX TX manually" on  the single SMA port of the B200 mini with the help of "GNURC ( GNU Radio Companion) Software", but I am not able to switching.

and If I use the Automatically Switch on the TRX SMA port of the B200 mini I gets an error of Overflow and Underflow in the GNU Radio Code.

Please help me in solving the error of the Overflow and Underflow in the Automatic Switch of TX and RX on the TRX port of SMA Cable by GNU Radio.
and
In Manually switching between the RX and TX on the Single TRX  SMA port of the B200 mini by GNU Radio.
-->
] --!>

## Setup Details ##

USRP B200 mini and GNU Radio -->

## Expected Behavior ##

 Automatic Switch work properly with out Error. 
and Manual Switch has to be TX for a 2 min and then RX for 2 min and continue] -->

## Actual Behaviour ##

Automatic switch Give Error and Manual Switch not working] -->

## Steps to reproduce the problem ##

## Additional Information ##

mbr0wn, 2024-03-19:
If you are using the TX/RX port, there is no manual switching. When TX'ing, you will always TX, otherwise, the switch will be set to RX. If you have a bursty transmitter and a continuous receiver, then you are already done. The LED colours will help you see if it' sworking.

mbr0wn, 2024-03-19:
Also, please use this bug tracker for bugs only. Thanks!

closed 2024-03-19 by mbr0wn