EttusResearch/uhd #735: building UHD not success

djl2020-ddd, 2024-03-23:
hi,i want set up uhd to use gnuradio-companion but i met this probleme:

## Issue Description ##
dingo@dingo-System-Product-Name:~/workarea/uhd/host/build$ make
[  2%] Built target uhd_rpclib
Consolidate compiler generated dependencies of target uhd
[  2%] Building CXX object lib/CMakeFiles/uhd.dir/rfnoc/dma_fifo_block_ctrl_impl.cpp.o
/home/dingo/workarea/uhd/host/lib/rfnoc/dma_fifo_block_ctrl_impl.cpp: In constructor ‘dma_fifo_block_ctrl_impl::dma_fifo_block_ctrl_impl(const uhd::rfnoc::make_args_t&)’:
/home/dingo/workarea/uhd/host/lib/rfnoc/dma_fifo_block_ctrl_impl.cpp:61:21: error: ‘_1’ was not declared in this scope
   61 |                     _1,
      |                     ^~
/home/dingo/workarea/uhd/host/lib/rfnoc/dma_fifo_block_ctrl_impl.cpp:61:21: note: suggested alternatives:
In file included from /usr/include/boost/mpl/aux_/include_preprocessed.hpp:37,
                 from /usr/include/boost/mpl/placeholders.hpp:43,
                 from /usr/include/boost/iterator/iterator_categories.hpp:16,
                 from /usr/include/boost/iterator/iterator_adaptor.hpp:14,
                 from /usr/include/boost/iterator/reverse_iterator.hpp:10,
                 from /usr/include/boost/range/reverse_iterator.hpp:21,
                 from /usr/include/boost/range/rend.hpp:19,
                 from /usr/include/boost/foreach.hpp:82,
                 from /home/dingo/workarea/uhd/host/include/uhd/property_tree.ipp:13,
                 from /home/dingo/workarea/uhd/host/include/uhd/property_tree.hpp:263,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/block_ctrl_base.hpp:11,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/sink_block_ctrl_base.hpp:11,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/dma_fifo_block_ctrl.hpp:11,
                 from /home/dingo/workarea/uhd/host/lib/rfnoc/dma_fifo_block_ctrl_impl.cpp:9:
/usr/include/boost/mpl/aux_/preprocessed/gcc/placeholders.hpp:29:16: note:   ‘mpl_::_1’
   29 | typedef arg<1> _1;
      |                ^~
In file included from /usr/include/boost/function.hpp:23,
                 from /home/dingo/workarea/uhd/host/include/uhd/convert.hpp:13,
                 from /home/dingo/workarea/uhd/host/lib/rfnoc/dma_fifo_block_ctrl_impl.cpp:8:
/usr/include/c++/11/functional:225:34: note:   ‘std::placeholders::_1’
  225 |     extern const _Placeholder<1> _1;
      |                                  ^~
In file included from /usr/include/boost/mpl/aux_/include_preprocessed.hpp:37,
                 from /usr/include/boost/mpl/placeholders.hpp:43,
                 from /usr/include/boost/iterator/iterator_categories.hpp:16,
                 from /usr/include/boost/iterator/iterator_adaptor.hpp:14,
                 from /usr/include/boost/iterator/reverse_iterator.hpp:10,
                 from /usr/include/boost/range/reverse_iterator.hpp:21,
                 from /usr/include/boost/range/rend.hpp:19,
                 from /usr/include/boost/foreach.hpp:82,
                 from /home/dingo/workarea/uhd/host/include/uhd/property_tree.ipp:13,
                 from /home/dingo/workarea/uhd/host/include/uhd/property_tree.hpp:263,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/block_ctrl_base.hpp:11,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/sink_block_ctrl_base.hpp:11,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/dma_fifo_block_ctrl.hpp:11,
                 from /home/dingo/workarea/uhd/host/lib/rfnoc/dma_fifo_block_ctrl_impl.cpp:9:
/usr/include/boost/mpl/aux_/preprocessed/gcc/placeholders.hpp:29:16: note:   ‘mpl_::_1’
   29 | typedef arg<1> _1;
      |                ^~
In file included from /usr/include/boost/bind/bind.hpp:2356,
                 from /usr/include/boost/thread/detail/thread.hpp:33,
                 from /usr/include/boost/thread/thread_only.hpp:22,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /home/dingo/workarea/uhd/host/include/uhd/utils/log.hpp:13,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/node_ctrl_base.hpp:14,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/block_ctrl_base.hpp:15,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/sink_block_ctrl_base.hpp:11,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/dma_fifo_block_ctrl.hpp:11,
                 from /home/dingo/workarea/uhd/host/lib/rfnoc/dma_fifo_block_ctrl_impl.cpp:9:
/usr/include/boost/bind/placeholders.hpp:46:38: note:   ‘boost::placeholders::_1’
   46 | BOOST_STATIC_CONSTEXPR boost::arg<1> _1;
      |                                      ^~
In file included from /usr/include/boost/mpl/aux_/include_preprocessed.hpp:37,
                 from /usr/include/boost/mpl/placeholders.hpp:43,
                 from /usr/include/boost/iterator/iterator_categories.hpp:16,
                 from /usr/include/boost/iterator/iterator_adaptor.hpp:14,
                 from /usr/include/boost/iterator/reverse_iterator.hpp:10,
                 from /usr/include/boost/range/reverse_iterator.hpp:21,
                 from /usr/include/boost/range/rend.hpp:19,
                 from /usr/include/boost/foreach.hpp:82,
                 from /home/dingo/workarea/uhd/host/include/uhd/property_tree.ipp:13,
                 from /home/dingo/workarea/uhd/host/include/uhd/property_tree.hpp:263,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/block_ctrl_base.hpp:11,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/sink_block_ctrl_base.hpp:11,
                 from /home/dingo/workarea/uhd/host/include/uhd/rfnoc/dma_fifo_block_ctrl.hpp:11,
                 from /home/dingo/workarea/uhd/host/lib/rfnoc/dma_fifo_block_ctrl_impl.cpp:9:
/usr/include/boost/mpl/aux_/preprocessed/gcc/placeholders.hpp:29:16: note:   ‘mpl_::_1’
   29 | typedef arg<1> _1;
      |                ^~
make[2]: *** [lib/CMakeFiles/uhd.dir/build.make:1119：lib/CMakeFiles/uhd.dir/rfnoc/dma_fifo_block_ctrl_impl.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:697：lib/CMakeFiles/uhd.dir/all] 错误 2
make: *** [Makefile:166：all] 错误 2

reliably -->
uhd 3.15.0.0 /ubuntu22.04

gmeyerho, 2024-03-26:
Hello, this seems to be an issue with this older version of UHD using a now deprecated practice for boost, see [here that "The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated..."](https://www.boost.org/doc/libs/1_80_0/boost/bind.hpp).

Moving to a newer version of UHD (like the most recent release: [UHD-4.6](https://github.com/EttusResearch/uhd/releases)) would no longer have this issue when building. If you need to stay on UHD 3.15, then you likely need to either downgrade your version of boost or workaround with the suggested change in boost/bind.hpp: "Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior."

mbr0wn, 2024-04-09:
@djl2020-ddd I agree, please move to newer version of UHD. We won't be patching old versions of UHD to work with new versions of Boost. Thanks!

closed 2024-04-09 by mbr0wn