EttusResearch/uhd #744: Switch from mprpc to a maintained alternative

doronbehar, 2024-04-30:
## Issue Description ##

I'm not using this package directly, but rather I'm the downstream maintainer of uhd at Nixpkgs. When I build uhd on our platform with Python API and utilities enabled, I notice this `usrp_hwd.py` script that depends on [mprpc](http://mprpc.readthedocs.org/) (also mentioned during the `cmake` phase). The author of [mprpc](http://mprpc.readthedocs.org/) recommends to use [gRPC](https://grpc.io/) instead. I noticed that mprpc doesn't compile on Python > 3.8 due to [`longintrepr.h` file included there](https://github.com/studio-ousia/mprpc/blob/6076f68a16f78e0010307344afa253e0956f2a9d/mprpc/client.c#L181) that has been renamed since Python3.8.

Excuse me for not filling exactly the bug report, as this is not exactly a bug report.

## Additional Information ##

Perhaps relevant is our attempt to fix our UHD Python API support at https://github.com/NixOS/nixpkgs/pull/307435 .

mbr0wn, 2024-06-21:
@doronbehar Thanks for reaching out. We do want to replace mprpc with something else, but that's a lot of work -- short-term, we might simply have to disable all uses of `mprpc`.

doronbehar, 2024-06-21:
Fine with me, I just want to distribute our `uhd` package with all of it's scripts fully working, and that's not possible if we don't (and we won't because it is deprecated) package [mprpc](http://mprpc.readthedocs.org/). Currently I commented about that in our Nix expression:

https://github.com/NixOS/nixpkgs/pull/307435/files#diff-34786da75d31da49632cde89a31fe4b87e378c96bba22353049c654f40dc2b1eR76

So nothing is urgent from our side.

mbr0wn, 2024-06-25:
Thanks for providing that feedback!

WilfredoLora, 2025-05-07:
@doronbehar Try this as i just modernised it to work with python3, unless mprpc has been disabled from UHD.

https://github.com/WilfredoLora/mprpc.git

doronbehar, 2025-05-07:
Thanks @WilfredoLora for the update and the link. I'm sure Nix users might benefit from it. If the UHD team will decide to officially point `usrp_hwd.py` users to your fork, I will package it in our distribution and make our `uhd` build use it. However at the moment I have no guarantee that it will be maintained properly for uhd in the near future.

mbr0wn, 2026-11-16:
The latest UHD release moved to gRPC.

closed 2026-11-16 by mbr0wn