EttusResearch/uhd #745: USRP N310: How to download XG fpga image and install over SFP port 0

GershomSandeep, 2024-05-03:
I used this command to install the fpga image for USRP N310: "sudo uhd_images_downloader -t mender -t n3xx --yes". But problem is by default it will install HG image so please provide a advice that how to install XG image for USRP N310 over SFP port 0.

wordimont, 2024-05-03:
This is covered in the getting started guide:

https://kb.ettus.com/USRP_N300/N310/N320/N321_Getting_Started_Guide#Updating_the_FPGA_Image

1. Download the FPGA images from the internet
```
$ sudo uhd_images_downloader
```
2. Download the desired N310 image (XG, HG, etc.) to your device
```
$ uhd_image_loader --args "type=n3xx,addr=<N3xx_IP_ADDR>,fpga=XG"
```
Notice I specified "XG" in this example, since that's the one you want.

You mentioned specifically downloading over SFP port 0. Any functional port should work. Generally, I prefer to specify the RJ45 port using `mgmt_addr=<RJ45 IP>`, since it doesn't go down while the FPGA is programming. Regardless, try specifying the address of the port you want to use.

```
$ uhd_image_loader --args "type=n3xx,mgmt_addr=<RJ45 IP>,fpga=XG"
```

You can usually use `uhd_find_devices` to find the IP address to use if you don't know it. You should also be able to ping the IP addresses to verify you're communicating properly with the port.

closed 2024-05-03 by wordimont

GershomSandeep, 2024-05-04:
Thanks for the previous reply, it worked well and now fpga image set to XG.
If i connect internet to SFP port 0 its was blinking  green light. suddenly its stopped blinking same in port 1 so not able to get desired IP address. I checked by pressing reset button and rebooted the device also but it remained same Please advice me on 
this issue.

[image]
[image]
[image]
[image]