EttusResearch/uhd #746: N310:SFP0

GershomSandeep, 2024-05-04:
Hi

If i connect internet to SFP port 0 earlier its was blinking green light. suddenly its stopped blinking same in port 1 also, so not able to get desired IP address. I checked by pressing reset button and rebooted the device but it remained same Please advice me on
this issue.
[image]
[image]
[image]
[image]

wordimont, 2024-05-04:
From the comment in https://github.com/EttusResearch/uhd/issues/745 with the pictures you posted (now deleted?), it looked like it had 1 GbE equipment connected to the SFP ports. So I would expect the link LED to be on for SFP0 until you download the XG image, then it would no longer work. If using the XG image, please ensure you have a 10 GbE SFP cable connected to a 10 GbE Ethernet card on the connected computer. You need to make sure the devices link (indicated by the green LED next to the SFP port) before you try to ping.

GershomSandeep, 2024-05-06:
thank you so much:-)

GershomSandeep, 2024-05-06:
Today I was trying N310 with srsRAN, i don't no what happened suddenly its restarting continuously. Even i tried by resetting the devices but no use. so please advise me on this issue.

wordimont, 2024-05-06:
You probably need to re-image your SD card to a known good state. At this point I suggest your reach out to technical support. https://www.ettus.com/support/

GershomSandeep, 2024-05-07:
sure thank you.

closed 2024-06-17 by wordimont