EttusResearch/uhd #748: Build warnings with Python 3.12

drmpeg, 2024-05-09:
## Issue Description ##
When building UHD 4.6.0.0 with Python 3.12, the following warnings are produced.

/home/ubuntu/xfer/uhd/host/lib/ic_reg_maps/common.py:379: SyntaxWarning: invalid escape sequence '\w'
  x = re.match('^~(\w*)\s+(.*)\s*$', mreg_des)

## Setup Details ##
Ubuntu 24.04

## Expected Behavior ##
No Python warnings.

## Actual Behaviour ##
The warning occurs many times.

## Steps to reproduce the problem ##
Build UHD 4.6.0 on Ubuntu 24.04 (with Python 3.12.3).

## Additional Information ##

drmpeg, 2024-05-09:
Additional note. The .hpp files generated by this Python script are identical to earlier versions. So this is a very minor issue.

mbr0wn, 2024-06-17:
Hey @drmpeg, thanks for reporting! This was fixed in 89091c1edd075a7104a7f075ba1831c0b1ab86f5.

closed 2024-06-17 by mbr0wn