EttusResearch/uhd #751: B2xx Repeated transmit decaying amplitudes starting with 4.5.0

thomasteisberg, 2024-05-13:
## Issue Description ##

I use a b205mini as part of a radar system. It repeatedly transmits a short chirp (in this case 10us) at a fixed pulse repetition interval (in this case 100 us).

Starting with UHD 4.5.0, the first transmitted chirp looks fine but subsequent pulses decrease rapidly in amplitude until they disappear into the noise floor. The problem can be reproduced across a range of pulse lengths, output power levels, etc. It only occurs at relatively high duty cycles (which we define as receive duration / pulse repetition interval). By relatively high, I mean 50% or more.

This issue did not occur on UHD 4.4.0. It happens with the UHD 4.5.0 and UHD 4.6.0. Screenshots of the behavior in the respective sections below.

## Setup Details ##

UHD and our user code runs on a Raspberry Pi 5. The output of the b205mini connects through a 30 dB attenuator and and RF splitter to an oscilloscope and back to the RX port of the b205mini. (So roughly 33 dB of attenuation from TX to both the scope and the RX port.)

The full codebase we run is available here: https://github.com/radioglaciology/uhd_radar

## Expected Behavior ##

Here is an example screenshots from the oscilloscope while running with UHD 4.4.0. This is what we expect to see.

[image]

Note that the measurement on the right side is a gated measurement of the first chirp. The scope is set to 50 ohm impedance and this corresponds to the expected output power level after the 33 dB of attenuation.

## Actual Behaviour ##

With UHD 4.5.0 and above (this screenshot is with 4.5.0 but I've also tested with 4.6.0 and it has the same behavior), the amplitude of the first chirp remains roughly as expected. Subsequent transmitted chirps have lower amplitudes until they rapidly decay into the noise floor. They remain beneath the noise floor until something causes a longer interval between chirps. (This can happen in our code if an ERROR_CODE_LATE_COMMAND is detected. When this happens, a longer interval is commanded between chirps. With UHD 4.5.0+, the amplitude then returns to normal for one pulse and proceeds to decay again as shown here.)

[image]

## Steps to reproduce the problem ##

Unfortunately, I don't have a small self-contained example. Our full project source code is available here: https://github.com/radioglaciology/uhd_radar

Any duty cycle above 50% seems to clearly reproduce this issue.

## Additional Information ##

This is basically just speculation, but the only commit I can find since 4.4.0 that seems like it would have mattered is this: https://github.com/EttusResearch/uhd/commit/69080b45e01f636b340d617e3b1cdb06c5c8e3c4

It seems plausible that this introduced an unexpected additional lag time in configuring the output switches after the end of receiving period.

I'm certainly open to suggestions about how to better architect our transmit and receive code. Perhaps there's a way to avoid changing the state of the RF switches? Relevant code would be here: https://github.com/radioglaciology/uhd_radar/blob/main/sdr/main.cpp

Thanks!

mmatthebi, 2024-06-24:
Probably you can run a [git bisect](https://git-scm.com/docs/git-bisect) between v4.4.0 und 4.5.0 to identify the actual commit and within this commit identify the problematic change. I think this can help the developers a lot. Git bisect sounds more complicated to use than it actually is.

thomasteisberg, 2026-10-06:
Since this bug, the [ORCA](https://github.com/HI-SNR-Lab/uhd_radar) project (now maintained by the HiSNR lab at CU Boulder) has just been pinning `uhd<=4.4.0`. With the introduction of the b206mini (which is only supported starting with `4.9.0`), we now have a need to upgrade to more recent versions. (And I'm sure there's lot of other fixes and improvements.)

As I speculated in the original issue, the problem (for us) is a behavior change in https://github.com/EttusResearch/uhd/commit/69080b45e01f636b340d617e3b1cdb06c5c8e3c4 that changed the state of the T/R switch connected to the `TX/RX` port on the B20xmini-i boards. (I would assume this applies to the B210 as well, but I don't have one.)

I believe that commit was aimed at avoiding a state where the two T/R switches (or 4 on the B210) were left in an undefined state. The side effect was that the T/R switch on the `TX/RX` port is now being set to it's RX mode when the system is in idle or RX-only on the `RX2` port.

I'm still not entirely clear on why this is a problem, given that the T/R switch [(SKY13350-385LF)](https://mm.digikey.com/Volume0/opasdata/d220001/medias/docus/400/SKY13350-385LF.pdf) has a 10-to-90 switching time of 45 ns. Somehow, however, it seems that rapidly cycling it causes the transmit power to drop by more than 20 dB.

The simple (4 line) fix is to leave the T/R switch in the the transmit configuration during idle and RX on `RX2` modes. This preserves the original intent (I think) of the commit in setting this switch to a defined state.

It would be great to understand a bit more about why this T/R switching is issue, given that the switching is happening at an internal of microseconds relative to the switching time of nanoseconds, but that probably requires more detailed knowledge of the schematics than I have.