EttusResearch/uhd #752: n310, EnvironmentError: IOError

houshangaziziaccelleran, 2024-05-21:
Hello
Any suggestion of this please?
```
sudo uhd_usrp_probe
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.HEAD-0-g50fa3baa
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=10.10.0.100,type=n3xx,product=n310,serial=32000F1,name=ni-n3xx-32000F1,fpga=XG,claimed=False,addr=10.10.2.100
[INFO] [MPM.PeriphManager] init() called with device args `fpga=XG,mgmt_addr=10.10.0.100,name=ni-n3xx-32000F1,product=n310,clock_source=internal,time_source=internal'.
[ERROR] [RFNOC::MGMT] EnvironmentError: IOError: recv error on socket: Connection refused
[ERROR] [RFNOC::GRAPH] IO Error during GSM initialization. EnvironmentError: IOError: recv error on socket: Connection refused
[ERROR] [RFNOC::GRAPH] Caught exception while initializing graph: EnvironmentError: IOError: recv error on socket: Connection refused
Error: RuntimeError: Failure to create rfnoc_graph.
```

Considering:
```
uhd_find_devices
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.HEAD-0-g50fa3baa
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 32000F1
    addr: 10.10.2.100
    claimed: False
    fpga: XG
    mgmt_addr: 10.10.0.100
    mgmt_addr: 10.10.1.100
    mgmt_addr: 10.10.2.100
    name: ni-n3xx-32000F1
    product: n310
    type: n3xx

```

wordimont, 2024-05-21:
Can you check if the cables are swapped? Make sure the subnets match for each connection. See this mailing list thread:

https://lists.ettus.com/empathy/thread/2JYP4CKEHLJYZHCV6DI4KOCJ6USAVSB6

houshangaziziaccelleran, 2024-05-22:
Hi @wordimont 
It helped
Thank you

closed 2024-05-23 by wordimont