EttusResearch/uhd #754: uhd installation issue when DPDK=ON

houshangaziziaccelleran, 2024-05-29:
Hello
Can you please help with the attached uhd installation issue?
Thanks
Houshang
[uhd_installation.txt](https://github.com/EttusResearch/uhd/files/15483392/uhd_installation.txt)

NI-LAm, 2024-05-29:
Looks like you miss some DPDK packages. Did you installed them?
```
sudo apt install dpdk dpdk-dev
```
This is the UHD issue tracker. Questions like this are better asked in the mailing list (more readers, quicker response time). https://kb.ettus.com/Mailing_Lists#usrp-users.

houshangaziziaccelleran, 2024-05-29:
Hello
It did not help. Please see the new screenshot attached.
Thanks
[uhd-error.txt](https://github.com/EttusResearch/uhd/files/15483734/uhd-error.txt)

houshangaziziaccelleran, 2024-05-29:
in line 270 on `uhd/host/lib/transport/uhd-dpdk/dpdk_io_service.cpp`
removed:

`    rte_thread_setname(pthread_self(), name);`

and added below instead

```
#if (RTE_VER_YEAR >= 23)
        rte_thread_set_name(rte_thread_self(), name);
#else
            rte_thread_setname(pthread_self(), name);
#endif
```

joergho, 2024-10-02:
Hi @houshangaziziaccelleran,

When using the UHD-4.6 branch, please cherry-pick 85e4b53a0d0715500a74db1f9b9cc234e5c6f4ef to fix the compile issue you are seeing. In the UHD-4.7 branch, the fix is already included.

However, for certain network cards, UHD 4.6 and 4.7 has an incompatibility with DPDK >= 22.11. In case you see the following error when initializing the device

```
[ERROR] [UHD] Device discovery error: vector::_M_range_check: __n (which is 4294967295) >= this->size() (which is 1)
```

then you additionally need to cherry-pick 8305d925176fed1fd7f375e3ae8d9de0b72b04d4. This fix was added to the `master` branch recently.

closed 2024-10-02 by mbr0wn