EttusResearch/uhd #758: Build fails: `error: expected id-expression before '(' token`

barracuda156, 2024-06-10:
This is from the latest 041eef3472e0440730708053d47b1fb7b793c682 commit, gcc 13.2.1, boost 1.81, macOS 10.6:

```
:info:build [ 56%] Building CXX object examples/CMakeFiles/rx_multi_samples.dir/rx_multi_samples.cpp.o
:info:build cd /opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/build/examples && /opt/local/bin/g++-mp-powerpc -DBOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW -DBOOST_ASIO_DISABLE_STD_STRING_VIEW -DBOOST_ERROR_CODE_HEADER_ONLY -DHAVE_CONFIG_H -DUHD_LOG_CONSOLE_COLOR -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_MIN_LEVEL=1 -I/opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/build/include -I/opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/uhd-041eef3472e0440730708053d47b1fb7b793c682/host/include -I/opt/local/include -I/opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/uhd-041eef3472e0440730708053d47b1fb7b793c682/host/examples/getopt -I/opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/build/_cmrc/include -pipe -I/opt/local/libexec/boost/1.81/include -Os -std=gnu++11 -D_GLIBCXX_USE_C99_MATH_TR1 -DNDEBUG -I/opt/local/libexec/boost/1.81/include -D_GLIBCXX_USE_CXX11_ABI=0 -fvisibility=hidden -fvisibility-inlines-hidden -std=gnu++14 -arch ppc -mmacosx-version-min=10.6   -Wall -Wextra -Wsign-compare -MD -MT examples/CMakeFiles/rx_multi_samples.dir/rx_multi_samples.cpp.o -MF CMakeFiles/rx_multi_samples.dir/rx_multi_samples.cpp.o.d -o CMakeFiles/rx_multi_samples.dir/rx_multi_samples.cpp.o -c /opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/uhd-041eef3472e0440730708053d47b1fb7b793c682/host/examples/rx_multi_samples.cpp
:info:build In file included from /opt/local/libexec/boost/1.81/include/boost/filesystem.hpp:18,
:info:build                  from /opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/uhd-041eef3472e0440730708053d47b1fb7b793c682/host/examples/rx_samples_to_file.cpp:14:
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp: In constructor 'boost::filesystem::recursive_directory_iterator::recursive_directory_iterator(const boost::filesystem::path&, boost::filesystem::symlink_option)':
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp:580:5: warning: 'boost::filesystem::recursive_directory_iterator::recursive_directory_iterator(const boost::filesystem::path&, boost::filesystem::symlink_option)' is deprecated: Use directory_options instead of symlink_option [-Wdeprecated-declarations]
:info:build   580 |     }
:info:build       |     ^
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp:577:5: note: declared here
:info:build   577 |     recursive_directory_iterator(path const& dir_path, BOOST_SCOPED_ENUM_NATIVE(symlink_option) opts)
:info:build       |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp: In constructor 'boost::filesystem::recursive_directory_iterator::recursive_directory_iterator(const boost::filesystem::path&, boost::filesystem::symlink_option)':
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp:580:5: warning: 'boost::filesystem::recursive_directory_iterator::recursive_directory_iterator(const boost::filesystem::path&, boost::filesystem::symlink_option)' is deprecated: Use directory_options instead of symlink_option [-Wdeprecated-declarations]
:info:build   580 |     }
:info:build       |     ^
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp:577:5: note: declared here
:info:build   577 |     recursive_directory_iterator(path const& dir_path, BOOST_SCOPED_ENUM_NATIVE(symlink_option) opts)
:info:build       |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp: In constructor 'boost::filesystem::recursive_directory_iterator::recursive_directory_iterator(const boost::filesystem::path&, boost::filesystem::symlink_option, boost::system::error_code&)':
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp:586:5: warning: 'boost::filesystem::recursive_directory_iterator::recursive_directory_iterator(const boost::filesystem::path&, boost::filesystem::symlink_option, boost::system::error_code&)' is deprecated: Use directory_options instead of symlink_option [-Wdeprecated-declarations]
:info:build   586 |     }
:info:build       |     ^
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp:583:5: note: declared here
:info:build   583 |     recursive_directory_iterator(path const& dir_path, BOOST_SCOPED_ENUM_NATIVE(symlink_option) opts, system::error_code& ec) BOOST_NOEXCEPT
:info:build       |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp: In constructor 'boost::filesystem::recursive_directory_iterator::recursive_directory_iterator(const boost::filesystem::path&, boost::filesystem::symlink_option, boost::system::error_code&)':
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp:586:5: warning: 'boost::filesystem::recursive_directory_iterator::recursive_directory_iterator(const boost::filesystem::path&, boost::filesystem::symlink_option, boost::system::error_code&)' is deprecated: Use directory_options instead of symlink_option [-Wdeprecated-declarations]
:info:build   586 |     }
:info:build       |     ^
:info:build /opt/local/libexec/boost/1.81/include/boost/filesystem/directory.hpp:583:5: note: declared here
:info:build   583 |     recursive_directory_iterator(path const& dir_path, BOOST_SCOPED_ENUM_NATIVE(symlink_option) opts, system::error_code& ec) BOOST_NOEXCEPT
:info:build       |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
:info:build In file included from /opt/local/libexec/boost/1.81/include/boost/process/detail/posix/handles.hpp:11,
:info:build                  from /opt/local/libexec/boost/1.81/include/boost/process/detail/used_handles.hpp:14,
:info:build                  from /opt/local/libexec/boost/1.81/include/boost/process/detail/posix/async_in.hpp:19,
:info:build                  from /opt/local/libexec/boost/1.81/include/boost/process/async.hpp:43,
:info:build                  from /opt/local/libexec/boost/1.81/include/boost/process.hpp:23,
:info:build                  from /opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/uhd-041eef3472e0440730708053d47b1fb7b793c682/host/examples/rx_samples_to_file.cpp:17:
:info:build /opt/local/libexec/boost/1.81/include/boost/process/detail/posix/handles.hpp: In function 'std::vector<int> boost::process::detail::posix::get_handles(std::error_code&)':
:info:build /opt/local/libexec/boost/1.81/include/boost/process/detail/posix/handles.hpp:36:20: error: expected id-expression before '(' token
:info:build    36 |     auto my_fd = ::dirfd(dir.get());
:info:build       |                    ^~~~~
:info:build /opt/local/libexec/boost/1.81/include/boost/process/detail/posix/handles.hpp: In member function 'void boost::process::detail::posix::limit_handles_::on_exec_setup(Executor&) const':
:info:build /opt/local/libexec/boost/1.81/include/boost/process/detail/posix/handles.hpp:120:24: error: expected id-expression before '(' token
:info:build   120 |         auto my_fd = ::dirfd(dir);
:info:build       |                        ^~~~~
:info:build /opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/uhd-041eef3472e0440730708053d47b1fb7b793c682/host/examples/rx_samples_to_file.cpp: In function 'double disk_rate_check(size_t, size_t, size_t, const std::string&)':
:info:build /opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/uhd-041eef3472e0440730708053d47b1fb7b793c682/host/examples/rx_samples_to_file.cpp:54:37: warning: unused parameter 'sample_type_size' [-Wunused-parameter]
:info:build    54 | double disk_rate_check(const size_t sample_type_size,
:info:build       |                        ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
:info:build /opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/uhd-041eef3472e0440730708053d47b1fb7b793c682/host/examples/rx_samples_to_file.cpp:55:18: warning: unused parameter 'channel_count' [-Wunused-parameter]
:info:build    55 |     const size_t channel_count,
[... cut ...] e=ON
:info:build make[2]: *** [examples/CMakeFiles/rx_samples_to_file.dir/rx_samples_to_file.cpp.o] Error 1
:info:build make[2]: Leaving directory `/opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_science_uhd/uhd-devel/work/build'
:info:build make[1]: *** [examples/CMakeFiles/rx_samples_to_file.dir/all] Error 2
```

The bug is there for quite some time: I initially tried to build an earlier version (which was in `uhd-devel` port), it failed, I proceeded with the master, it failed likewise.

mbr0wn, 2024-06-17:
Thanks for reporting! We have a Mac OS CI pipeline using Boost 1.85 which has not shown this issue, but I'm not sure we're running the same system. Nevertheless, there are at least some obvious unnecessary includes.

closed 2024-06-25 by joergho

mbr0wn, 2024-06-27:
This should be good to go on the upcoming release.

barracuda156, 2024-06-27:
Great, thanks for addressing it.