EttusResearch/uhd #759: RuntimeError: RuntimeError: Expected FPGA compatibility number 36, but got 39:

panitsasi, 2024-06-13:
## Issue Description ##

I am trying to interact with my USRP-X310 and to run the following commands uhd_fft, uhd_find_devices, uhd_image_loader, uhd_images_downloader, uhd_rx_cfile, uhd_rx_nogui, uhd_siggen, uhd_siggen_gui, uhd_usrp_probe.

The commands uhd_find_devices, uhd_image_loader, uhd_images_downloader, uhd_usrp_probe
are working properly but when i am trying to run some commands like the following:  

uhd_usrp_probeuhd_fft -s 10e6 -g 100 -f 2.437e9

I am receiving this error: RuntimeError: RuntimeError: Expected FPGA compatibility number 36, but got 39:
The FPGA image on your device is not compatible with this host code build.
Download the appropriate FPGA images for this version of UHD.
Please run:

 "/usr/lib/uhd/utils/uhd_images_downloader.py"

Then burn a new image to the on-board flash storage of your
USRP X3xx device using the image loader utility. Use this command:

"/usr/bin/uhd_image_loader" --args="type=x300,addr=192.168.10.2"

I followed the steps but it did not work. 
[image]

wordimont, 2024-06-13:
Compat number 36 corresponds to UHD 3.15. Compat number 39 corresponds to UHD 4.3 and later. You must have a mix of versions on your system, and/or you're running code that was compiled for an older version of UHD.

I see in your screen shot that when you ran uhd_image_loader, it reported UHD 4.6. But the program that gives you the compat error message must have been built on UHD 3.15 or earlier if it's expecting compat 36. Unfortunately, there were major changes between those two versions, so they're not compatible.

panitsasi, 2024-06-14:
So, if i delete all the pre-installed uhd drivers and re-install UHD 3.15, you think that it will solve the problem ?

wordimont, 2024-06-14:
Make sure you update the FPGA image using the version you settle on. UHD 3.15 is getting pretty old, so you might consider upgrading, but you do need the FPGA version and UHD version to match.

panitsasi, 2024-06-14:
Okay, i fixed the issue, thank you.

closed 2024-06-14 by wordimont