EttusResearch/uhd #767: Make E320_AA error

Emreyldz06, 2024-07-03:
## Issue Description ##

I am using the vivado 2022.1. I will use the USRP model E320 and first I want to implement and boot the on of default design in directory /fpga/usrp3/top/e320 
I run the following commands : 
source setupenv.sh --vivado-path=/media/user/disk2/Xilinx/Vivado
make E320_AA
But I got an error as described at below 

## Setup Details ##

I have build the following UHD version 
$ uhd_config_info  --version 
UHD 4.0.0.0-240-gb38c9d83
 vivado 2022.1.
I am using ubuntu18.04 

## Expected Behavior ##

I am expecting to get output as described on UHD documentation page 
## Actual Behaviour ##

I am getting the following error:user@user-HP-Z420-Workstation:/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320$ make E320_AA
make -f Makefile.e320.inc bin NAME=E320_AA ARCH=zynq PART_ID=xc7z045/ffg900/-3 SFP_AURORA=1 BUILD_AURORA=1  E320=1 TOP_MODULE=e320 EXTRA_DEFS="SFP_AURORA=1 BUILD_AURORA=1  E320=1" DEFAULT_RFNOC_IMAGE_CORE_FILE=e320_rfnoc_image_core.v DEFAULT_EDGE_FILE=/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/e320_static_router.hex
make[1]: Entering directory '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320'
BUILDER: Checking tools...
* GNU bash, version 4.4.20(1)-release (x86_64-pc-linux-gnu)
* Python 3.6.9
* Vivado v2022.1_AR76780 (64-bit)
========================================================
BUILDER: Building IP ten_gig_eth_pcs_pma
========================================================
BUILDER: Staging IP in build directory...
BUILDER: Reserving IP location: /media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma
BUILDER: Retargeting IP to part zynq/xc7z045/ffg900/-3...
BUILDER: Building IP...
[00:00:00] Executing command: vivado -mode batch -source /media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/tools/scripts/viv_generate_ip.tcl -log ten_gig_eth_pcs_pma.log -nojournal
WARNING: [Common 17-306] Update version (2021.1_AR76780) does not match product version (2022.1).
WARNING: [IP_Flow 19-2162] IP 'ten_gig_eth_pcs_pma' is locked:
WARNING: [Vivado 12-13651] The IP file '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xci' has been moved from its original location, as a result the outputs for this IP will now be generated in '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma'. Alternatively a copy of the IP can be imported into the project using one of the 'import_ip' or 'import_files' commands.
CRITICAL WARNING: [filemgmt 20-1366] Unable to reset target(s) for the following file is locked: /media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xci
CRITICAL WARNING: [filemgmt 20-1365] Unable to generate target(s) for the following file is locked: /media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xci
[00:00:53] Current task: Initialization +++ Current Phase: Starting
WARNING: [Vivado 12-13651] The IP file '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xci' has been moved from its original location, as a result the outputs for this IP will now be generated in '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma'. Alternatively a copy of the IP can be imported into the project using one of the 'import_ip' or 'import_files' commands.
[00:00:53] Current task: Initialization +++ Current Phase: Finished
[00:00:53] Executing Tcl: synth_design -top ten_gig_eth_pcs_pma -part xc7z045ffg900-3 -mode out_of_context
[00:00:53] Starting Synthesis Command
WARNING: [Vivado_Tcl 4-391] The following IPs are missing output products for Synthesis target. These output products could be required for synthesis, please generate the output products using the generate_target or synth_ip command before running synth_design.
WARNING: [IP_Flow 19-2162] IP 'ten_gig_eth_pcs_pma' is locked:
ERROR: [Designutils 20-414] HRTInvokeSpec : No Verilog or VHDL sources specified
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xml'
ERROR: [Vivado 12-398] No designs are open
ERROR: [Common 17-69] Command failed: * IP definition '10G Ethernet PCS/PMA (10GBASE-R/KR) (6.0)' for IP 'ten_gig_eth_pcs_pma' (customized with software release 2019.1) has a different revision in the IP Catalog.
[00:00:58] Current task: Synthesis +++ Current Phase: Starting
[00:00:58] Current task: Synthesis +++ Current Phase: Finished
[00:00:58] Process terminated. Status: Failure

========================================================
Warnings:           6
Critical Warnings:  7
Errors:             9

BUILDER: Releasing IP location: /media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma
/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/ip/ten_gig_eth_pcs_pma/Makefile.inc:41: recipe for target '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xci' failed
make[1]: *** [/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320/build-ip/xc7z045ffg900-3/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xci] Error 1
make[1]: Leaving directory '/media/user/disk2/sdr_emre/src/uhd/fpga/usrp3/top/e320'
Makefile:70: recipe for target 'E320_AA' failed
make: *** [E320_AA] Error 2

## Steps to reproduce the problem ##

## Additional Information ##

wordimont, 2024-07-04:
Building the FPGA image for UHD 4.0.0 requires Vivado 2019.1. Vivado is giving you that error because the version of the IP in the GitHub repo doesn't match the IP version in Vivado 2022.1 which you're trying to use. Unfortunately, Vivado tends to break compatibility between versions so it's recommended using the Vivado version indicated in the UHD manual and setupenv.sh for the FPGA you're trying to build.

mbr0wn, 2024-07-04:
@Emreyldz06 what @wordimont says is important -- if you start using the right version of Vivado, things will start working. Also, why not use a newer version of UHD? Recent UHD requires Vivado 2021.1.

closed 2024-07-04 by mbr0wn

Emreyldz06, 2024-07-04:
Thanks for your reply @wordimont and @mbr0wn. Today I downgraded my vivado version to 2021.1 and also upgraded the UHD to newest version. Finally I solved the problem with adding AR76780_Vivado_2021_1_preliminary_rev1 patch. Thanks for your answers.