EttusResearch/uhd #776: Ettus X410 UHD 4,4 LMX2572 Unable to communicate with LMX2572 Expected R125==0x2288 got: FFFF

jres-mil, 2024-07-31:
## Issue Description ##

Trying to get any output from Ettus X410. Running uhd_usrp_probe returns error message. device is found using uhd_find_devices
I am able to login to stm32 microcontroller, and linux partition using screen. When trying to collect any data from x410 I receive the error. Any --args that are included does not fix the issue. 
## Setup Details ##

UHD 4.4.0.0
FPGA X4_200

## Expected Behavior ##

## Actual Behaviour ##

[ERROR] [LMX2572] Unable to communicate with LMX2572! Expected R125==0x2288, got: **FFFF**
[ERROR] [RFNOC::GRAPH] Error during initialization of block 0/Radio#0!
[ERROR] [RFNOC::GRAPH] Caught exception while initializing graph: RuntimeError: Unable to communicate to LMX2572!
Error: RuntimeError: Failure to create rfnoc_graph.

[ERROR] [LMX2572] Unable to communicate with LMX2572! Expected R125==0x2288, got: **0000**
[ERROR] [RFNOC::GRAPH] Error during initialization of block 0/Radio#0!
[ERROR] [RFNOC::GRAPH] Caught exception while initializing graph: RuntimeError: Unable to communicate to LMX2572!
Error: RuntimeError: Failure to create rfnoc_graph.

These are two separate error messages I have received. One is got: 0000 and the other got: FFFF

## Steps to reproduce the problem ##

uhd_usrp_probe
## Additional Information ##

In the --help on the linux partition lists a way --interactive-iface to poke and peek registers, which may be a way to rewrite the R125 register but I cannot find any documentation about the command.

manderseck, 2024-08-01:
@jres-mil In general it looks like a communication problem between the daughterboard CPLD and the LO synthesizer. We poke and peek some registers during session setup to ensure the communication works properly and that obviously fails with some random return value (0xFFFF and 0x0000) here. May I ask you for the following things? 
Can you please login to the device and run `eeprom-dump db0` and paste the output into here (you may remove the serial number). Since it may have to make with the CPLD not communicating properly you could try to rewrite the CPLD image by using the `zbx_update_cpld` utility which is available on the device, too, see here: https://files.ettus.com/manual/page_zbx.html#zbx_updating_cpld
Ensure to power-cycle the device afterwards. If that all doesn't help, a hardware failure may be possible, too, unfortunately.

jres-mil, 2024-08-01:
@manderseck  I am using a standalone development computer to run X410. I am not able to copy and paste outputs, I will type them out.

Output from eeprom-dump db0 as follows:
usrp_eeprom_board_info (0x10) pid: 0x4002, rev: 0x0005, compat_rev: 0x0002, serial: 328XXXX
usrp_eeprom_db_pwr_seq (0x12) (20, 0x07) (20, 0x18) (0, 0x00) (0, 0x00) (0, 0x00) (0, 0x00) (0, 0x00) (0, 0x00)

Running zbx_update_cpld I receive the error
NotImplementedError: The CPLD update Strategy for rev or CPLD model 5 is not available.

manderseck, 2024-08-01:
@jres-mil Thanks for providing more information. You could still try to update the CPLD by installing a later file system version, UHD 4.5 or 4.6 should work, in UHD 4.7 I just discovered that the CPLD updater got broken, we're working on this. While the operation of that CPLD (normally) works in older UHD versions, the CPLD image file for this particular component is only shipped from 4.5 on. However to me it looks like a case for RMA unfortunately, so I'd ask you to get in touch with NI's support/RMA. I'm sorry for that.

jres-mil, 2024-08-01:
@manderseck Thank you for the update I will try a newer version. The platform we are building for requires 4.4, however I will be able to make sure the equipment is working or not.

manderseck, 2024-08-02:
For now I'm going to close this as it doesn't seem to be a software issue. I expect the CPLD flashing to work properly in later UHD versions (except for 4.7) but at the same time I fear that doesn't fix your (hardware) issue. If anything new comes up in that issue feel free to re-open. Thanks!

closed 2024-08-02 by manderseck

brt99, 2024-09-19:
@manderseck I have the same problem with my x410 as @jres-mil.  I have installed UHD version 4.6 and flashed the daughterboard firmware with zbx_update_cpld as described. However, this did not solve the problem and the error message still exists. Is there already a solution? Thanks!

manderseck, 2024-09-19:
@brt99 We are still looking into the issue internally. May I ask you to open a service request, so we can get more information from your device, too? 
https://www.ni.com/en/support.html