EttusResearch/uhd #778: Error in macOS15.0 : pointer being freed was not allocated

doctormin, 2024-08-02:
## Issue Description ##
I have been successfully using the UHD Python API with USRP devices on macOS. However, recently, after upgrading to the macOS 15.0 developer beta system, my UHD API returns an error when running the following line of code:
https://github.com/EttusResearch/uhd/blob/a5ed1872be6d0fc36de9a7e0b508933da1f119bc/host/python/uhd/usrp/multi_usrp.py#L36

The error is as follows:
```
[INFO] [UHD] Mac OS; Clang version 16.0.6 ; Boost_108400; UHD_4.7.0.0-release
python(28956,0x1f0f218c0) malloc: *** error for object 0x1eee29cc0: pointer being freed was not allocated
python(28956,0x1f0f218c0) malloc: *** set a breakpoint in malloc_error_break to debug
```

## Setup Details ##
Host OS: macOS 15.0
Python: 3.12.4
UHD: 4.7.0.0-release

ENV setup:
I used miniforge to install gnuradio from conda-forge, which automatically installs Python and UHD. For example, I am currently using the latest version:
[image]

Suspecting this bug might be related to the versions of Python and UHD, I tried the following:
`mamba install gnuradio=3.10.7` (UHD=4.5.0.0, Python=3.11.9) 
but encountered the same error.

## Steps to reproduce the problem ##

1. Prepare a host with macOS 15.0
2. install miniforge
3. `mamba create -n <env_name>`
4. `mamba activate <env_name>`
5. `mamba install gnuradio`
6. Write a python script to evoke UHD API:
```python
import uhd
import time
usrp = uhd.usrp.MultiUSRP()
```

This error will emerge:
[image]

kgarrels, 2024-11-26:
Same issue here, macOS 15.1 Apple silicon, radioconda installation.
Strange: 
- when I build gqrx with cmake, it works fine, no crash
- when building with Xcode, this crash happens.

gqrx is here: https://github.com/gqrx-sdr/gqrx

joergho, 2024-11-29:
Thanks, @doctormin , @rchmielarz , @kgarrels for reporting the issue.

For further analysis, could you please
1. Update to the latest build of `uhd` in conda (just to be sure)
2. provide the output of the following commands
    1. `UHD_LOG_CONSOLE_LEVEL=debug uhd_find_devices`
    2. `UHD_LOG_CONSOLE_LEVEL=debug uhd_find_devices --args type=x3xx`

kgarrels, 2024-11-29:
```
mamba list uhd
# packages in environment at /Users/kai/radioconda:
#
# Name                    Version                   Build  Channel
gnuradio-uhd              3.10.11.0       py310h2193f99_1    conda-forge
uhd                       4.7.0.0         py310hd42cda6_0    conda-forge
UHD_LOG_CONSOLE_LEVEL=debug uhd_find_devices
[INFO] [UHD] Mac OS; Clang version 16.0.6 ; Boost_108400; UHD_4.7.0.0-release
[DEBUG] [MPMD] Discovering MPM devices on port 49600
[DEBUG] [MPMD] Discovering MPM devices on port 49600
No UHD Devices Found
UHD_LOG_CONSOLE_LEVEL=debug uhd_find_devices --args type=x3xx
No UHD Devices Found
python
Python 3.10.15 | packaged by conda-forge | (main, Oct 16 2024, 01:24:20) [Clang 17.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import uhd
>>> usrp = uhd.usrp.MultiUSRP()
[INFO] [UHD] Mac OS; Clang version 16.0.6 ; Boost_108400; UHD_4.7.0.0-release
python(6099,0x1f15e7840) malloc: *** error for object 0x1ef4c5cc0: pointer being freed was not allocated
python(6099,0x1f15e7840) malloc: *** set a breakpoint in malloc_error_break to debug
zsh: abort      python
```

kgarrels, 2024-11-29:
would you expect different results than for gqrx?
https://github.com/EttusResearch/uhd/issues/778#issuecomment-2501844003

I can try, needs some time...

joergho, 2024-11-29:
> would you expect different results than for gqrx?

If UHD is built as part of gqrx (I have not checked if it is) then you are right - then the assumption would be that the UHD examples (built using cmake) also work.

You mentioned that with Xcode it crashes. What do you mean? Did you build gqrx using Xcode? If this is the case then this would be an indication that somehow the build using Xcode is causing the problem. Then it would be interesting to see if the UHD examples built using Xcode show the issue.

kgarrels, 2024-11-29:
Yes, 
build with Xcode --> crash when gqrx starts with the same errors as we see in the python example.
build with cmake --> no crash, gqrx works fine.

I create the Xcode project with cmake -G Xcode .. , I found no difference in the cmake logs.
Xcode is difficult to understand, to say the least :-)

It could well be that we have two different things - I was very surprised to see the python example delivers the same error messages.

I will try the examples, and send  a picture of the Xcode behaviour.

kgarrels, 2024-11-30:
libuhd is linked dynamically.

I did not manage to build the examples yet, but radioconda has them installed:
```
./rx_samples_to_file 

Creating the usrp device with: ...
[INFO] [UHD] Mac OS; Clang version 16.0.6 ; Boost_108400; UHD_4.7.0.0-release
Error: LookupError: KeyError: No devices found for ----->
Empty Device Address
```
I connected my B200: 
```
./rx_samples_to_file

Creating the usrp device with: ...
[INFO] [UHD] Mac OS; Clang version 16.0.6 ; Boost_108400; UHD_4.7.0.0-release
[INFO] [B200] Detected Device: B200
[INFO] [B200] Loading FPGA image: /Users/kai/radioconda/share/uhd/images/usrp_b200_fpga.bin...
[INFO] [B200] Operating over USB 2.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] No GPSDO found
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
Using Device: Single USRP:
  Device: B-Series Device
  Mboard 0: B200
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2

Setting RX Rate: 1.000000 Msps...
[INFO] [B200] Asking for clock rate 32.000000 MHz... 
[INFO] [B200] Actually got clock rate 32.000000 MHz.
Actual RX Rate: 1.000000 Msps...

Setting RX Freq: 0.000000 MHz...
Setting RX LO Offset: 0.000000 MHz...
Actual RX Freq: 34.000000 MHz...

Locking LO on channel 0
Waiting for "lo_locked": ++++++++++ locked.

Press Ctrl + C to stop streaming...
^C
Done!
```

kgarrels, 2024-11-30:
crash with Xcode:

```
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.11.0
built-in source types: file rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
rx_nb_cc :info: set_min_output_buffer on block 12 to 32768
Resampling audio 96000 -> 48000
BandPlanFile is /Users/kai/.config/gqrx/bandplan.csv
BookmarksFile is /Users/kai/.config/gqrx/bookmarks.csv
AddInstanceForFactory: No factory registered for id <CFUUID 0x600001f8a880> F8BB1C28-BAE8-11D6-9C31-00039315CD46
Number of audio input devices: 4
Number of audio output devices: 6
[0;32m[INFO] [UHD] [0;39mMac OS; Clang version 16.0.6 ; Boost_108400; UHD_4.7.0.0-release
gqrx(21948,0x170cbf000) malloc: *** error for object 0x1ef4c5cc0: pointer being freed was not allocated
gqrx(21948,0x170cbf000) malloc: *** set a breakpoint in malloc_error_break to debug
```
no crash with cmake:
```
./src/gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.11.0
built-in source types: file rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
rx_nb_cc :info: set_min_output_buffer on block 12 to 32768
Resampling audio 96000 -> 48000
BandPlanFile is /Users/kai/.config/gqrx/bandplan.csv
BookmarksFile is /Users/kai/.config/gqrx/bookmarks.csv
Number of audio input devices: 4
Number of audio output devices: 6
[INFO] [UHD] Mac OS; Clang version 16.0.6 ; Boost_108400; UHD_4.7.0.0-release
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.11.0
built-in source types: file rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
Using libairspyhf1.6.8, samplerates: 0.192M 0.228M 0.384M 0.456M 0.65M 0.768M 0.912M 
Decimation: 4
  stage: 1  ratio: 4
```

weird.

michaelld, 2024-12-01:
@kgarrels Hey Kai! Long time ... hope you're doing well. I finally got a demo working for a conference in a week (whew!) & am jumping to a few macOS issues of recent including this one.

For macOS 15.1.1 and current miniforge, can you list the versions like the author did above? IDK if that's helpful yet ... I've never used Forge, so trying to come up to speed on it alongside Brew and MacPorts & also seeing if macOS provides enough dependencies to do a general build

michaelld, 2024-12-01:
Catching up here ... by default there is no "uhd" in miniconda ... one has to go to radioconda to find the SDR projects like UHD & GNU Radio & others

michaelld, 2024-12-01:
UHD installs a SO in the Python directory ... for me it is `libpyuhd.cpython-312-darwin.so` ... what does `otool -L libpyuhd.cpython-312-darwin.so` (or whatever the name is) return? For me I see some `@rpath` that likely won't work and the self-ID is wrong. I'm guessing this is what you'll see too: some build using `@rpath` and some do not ... those that do not are *much* more likely to work

michaelld, 2024-12-01:
FYI the basic UHD install in miniconda was really easy ... clearly there is some work to do on the Python API for macOS still ...

doctormin, 2024-12-01:
Hi, this issue persists in the latest version:

```
❯ mamba install gnuradio       

Looking for: ['gnuradio']

conda-forge/osx-arm64                               13.2MB @   3.7MB/s  3.5s
conda-forge/noarch                                  17.5MB @   4.4MB/s  4.0s
warning  libmamba Invalid package cache, file '/Users/yiminzhao/.pyenv/versions/miniforge3-latest/pkgs/uhd-4.7.0.0-py312h11b8700_2/lib/python3.12/site-packages/uhd/usrp/libtypes.py' has incorrect size
Transaction

  Prefix: /Users/yiminzhao/.pyenv/versions/miniforge3-latest/envs/latest

  Updating specs:

   - gnuradio

  Package                           Version  Build                 Channel           Size
───────────────────────────────────────────────────────────────────────────────────────────
  Install:
───────────────────────────────────────────────────────────────────────────────────────────

  + libusb                           1.0.27  h93a5062_100          conda-forge     Cached
  + libzlib                           1.3.1  h8359307_2            conda-forge     Cached
  + gstreamer-orc                    0.4.40  hd74edd7_0            conda-forge     Cached
  + libcxx                           19.1.4  ha82da77_0            conda-forge     Cached
  + icu                                75.1  hfee45f7_0            conda-forge     Cached
  + libogg                            1.3.5  h99b78c6_0            conda-forge     Cached
  + fribidi                          1.0.10  h27ca646_0            conda-forge     Cached
  + hicolor-icon-theme                 0.17  hce30654_2            conda-forge     Cached
  + libjpeg-turbo                     3.0.0  hb547adb_1            conda-forge     Cached
  + ca-certificates               2024.8.30  hf0a4a13_0            conda-forge     Cached
  + xz                                5.2.6  h57fd34a_0            conda-forge     Cached
  + libffi                            3.4.2  h3422bc3_5            conda-forge     Cached
  + libiconv                           1.17  h0d3ecfb_2            conda-forge     Cached
  + epoxy                            1.5.10  h1c322ee_1            conda-forge     Cached
  + libopus                           1.3.1  h27ca646_1            conda-forge     Cached
  + python_abi                         3.12  5_cp312               conda-forge     Cached
  + libsodium                        1.0.20  h99b78c6_0            conda-forge     Cached
  + libcodec2                         1.2.0  hd74edd7_2            conda-forge     Cached
  + libdeflate                         1.22  hd74edd7_0            conda-forge     Cached
  + libwebp-base                      1.4.0  h93a5062_0            conda-forge     Cached
  + llvm-openmp                      19.1.4  hdb05f8b_0            conda-forge     Cached
  + ncurses                             6.5  h7bae524_1            conda-forge     Cached
  + yaml                              0.2.5  h3422bc3_2            conda-forge     Cached
  + bzip2                             1.0.8  h99b78c6_7            conda-forge     Cached
  + libexpat                          2.6.4  h286801f_0            conda-forge     Cached
  + pthread-stubs                       0.4  hd74edd7_1002         conda-forge     Cached
  + xorg-libxdmcp                     1.1.5  hd74edd7_0            conda-forge     Cached
  + xorg-libxau                      1.0.11  hd74edd7_1            conda-forge     Cached
  + lame                              3.100  h1a8c8d9_1003         conda-forge     Cached
  + libbrotlicommon                   1.1.0  hd74edd7_2            conda-forge     Cached
  + tk                               8.6.13  h5083fa2_1            conda-forge     Cached
  + libsqlite                        3.47.0  hbaaea75_1            conda-forge     Cached
  + zlib                              1.3.1  h8359307_2            conda-forge     Cached
  + libpng                           1.6.44  hc14010f_0            conda-forge     Cached
  + zstd                              1.5.6  hb46c0d2_0            conda-forge     Cached
  + mpg123                           1.32.9  hf642e45_0            conda-forge     Cached
  + qhull                            2020.2  h420ef59_5            conda-forge     Cached
  + lerc                              4.0.0  h9a09cb3_0            conda-forge     Cached
  + gmp                               6.3.0  h7bae524_2            conda-forge     Cached
  + libasprintf                      0.22.5  h8414b35_3            conda-forge     Cached
  + graphite2                        1.3.13  hebf3989_1003         conda-forge     Cached
  + pixman                           0.43.4  hebf3989_0            conda-forge     Cached
  + nspr                               4.36  h5833ebf_0            conda-forge     Cached
  + sdl2                             2.30.7  hf9b8971_0            conda-forge     Cached
  + volk                              3.1.2  hebf3989_0            conda-forge     Cached
  + portaudio                        19.7.0  h5833ebf_0            conda-forge     Cached
  + fmt                              11.0.2  h420ef59_0            conda-forge     Cached
[... 133 lines cut ...]
  + urllib3                           2.2.3  pyhd8ed1ab_0          conda-forge     Cached
  + jsonschema-specifications     2024.10.1  pyhd8ed1ab_0          conda-forge     Cached
  + requests                         2.32.3  pyhd8ed1ab_0          conda-forge     Cached
  + jsonschema                       4.23.0  pyhd8ed1ab_0          conda-forge     Cached
[... cut ...]
  + gnuradio-iio                  3.10.11.0  py312hc9ed232_5       conda-forge     Cached
  + gnuradio-grc                  3.10.11.0  py312h14fb442_5       conda-forge     Cached
  + gnuradio-uhd                  3.10.11.0  py312hbfa0caa_5       conda-forge     Cached
  + gnuradio                      3.10.11.0  py312h19d1795_5       conda-forge     Cached

  Summary:

  Install: 194 packages

  Total download: 28MB

───────────────────────────────────────────────────────────────────────────────────────────

Confirm changes: [Y/n] 
```

doctormin, 2024-12-01:
> Catching up here ... by default there is no "uhd" in miniconda ... one has to go to radioconda to find the SDR projects like UHD & GNU Radio & others在这里赶上进度……默认情况下，miniconda 中没有“uhd”……必须去 radioconda 才能找到像 UHD 和 GNU Radio 等 SDR 项目。

By the way, the `conda-forge` channel does have `gnuradio` (https://anaconda.org/conda-forge/gnuradio), so you just use Miniforge (https://github.com/conda-forge/miniforge) and run `mamba install gnuradio`. This method is straightforward and facilitates version management. I even wonder why this has not become the standard installation recommendation for GNURadio or UHD, as using MacPorts and other alternatives often leads to challenges such as difficulties in removal and the inability to coexist with multiple versions.

kgarrels, 2024-12-01:
> @kgarrels Hey Kai! Long time ... hope you're doing well. I finally got a demo working for a conference in a week (whew!) & am jumping to a few macOS issues of recent including this one.

Wow, nice to meet you again! All good here! I met a lot of nice people during the EU gnuradio days in Darmstadt, very cool meeting...

> 
> For macOS 15.1.1 and current miniconda, can you list the versions like the author did above? IDK if that's helpful yet ... I've never used conda (any of them), so trying to come up to speed on it alongside Brew and MacPorts & also seeing if macOS provides enough dependencies to do a general build

I dropped macports some time ago, it seems gnuradio was not getting updates, maybe I just didn't understand it.
Now using radiocoda, which is super easy to use.
I never used homebrew.

The bug start after I upgraded macos 14.x to 15.1.
It also occurred with gnuradio 3.10.10.0 and uhd 4.6.0.0 on macos 15.1.

I think we should forget the Xcode problem for now, that thing is just too complicated to understand.

otool:
```
otool -L /Users/kai/radioconda/lib/python3.10/site-packages/uhd/libpyuhd.cpython-310-darwin.so
/Users/kai/radioconda/lib/python3.10/site-packages/uhd/libpyuhd.cpython-310-darwin.so:
	/Users/runner/miniforge3/conda-bld/uhd_1719620978768/work/host/python/lib/python3.10/site-packages/uhd/libpyuhd.cpython-310-darwin.so (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libuhd.4.7.0.dylib (compatibility version 4.7.0, current version 4.7.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.0.0)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
```

Here is my version listing:
```
mamba list
# packages in environment at /Users/kai/radioconda:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        47.0                     unix_0    conda-forge
airspy                    1.0.10               h3422bc3_0    conda-forge
airspyhf                  1.6.8                h3422bc3_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
archspec                  0.2.3              pyhd8ed1ab_0    conda-forge
armadillo                 14.2.0               h925db70_0    conda-forge
arpack                    3.9.1           nompi_h593882a_101    conda-forge
asciimatics               1.15.0             pyhd8ed1ab_0    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd03087b_2    conda-forge
attrs                     24.2.0             pyh71513ae_0    conda-forge
bcrypt                    4.2.1           py310hde4708a_0    conda-forge
bidict                    0.23.1             pyhd8ed1ab_0    conda-forge
bladerf                   2024.05              h99b78c6_0    conda-forge
blinker                   1.9.0              pyhff2d567_0    conda-forge
boltons                   24.0.0             pyhd8ed1ab_0    conda-forge
boost                     1.84.0               h24597f5_6    conda-forge
brotli                    1.1.0                hd74edd7_2    conda-forge
brotli-bin                1.1.0                hd74edd7_2    conda-forge
brotli-python             1.1.0           py310hb4ad77e_2    conda-forge
bzip2                     1.0.8                h99b78c6_7    conda-forge
c-ares                    1.34.3               h5505292_1    conda-forge
ca-certificates           2024.8.30            hf0a4a13_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               hb4a6bf7_3    conda-forge
certifi                   2024.8.30          pyhd8ed1ab_0    conda-forge
cffi                      1.17.1          py310h497396d_0    conda-forge
charset-normalizer        3.4.0              pyhd8ed1ab_0    conda-forge
click                     8.1.7           unix_pyh707e725_0    conda-forge
click-plugins             1.1.1                      py_0    conda-forge
cmake                     3.31.1               h326f17c_0    conda-forge
codec2                    1.0.3                h99b78c6_1    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
conda                     24.11.0         py310hbe9552e_0    conda-forge
conda-libmamba-solver     24.9.0             pyhd8ed1ab_0    conda-forge
conda-package-handling    2.4.0              pyh7900ff3_0    conda-forge
conda-package-streaming   0.11.0             pyhd8ed1ab_0    conda-forge
[... 311 lines cut ...]
uhd                       4.7.0.0         py310hd42cda6_0    conda-forge
unicodedata2              15.1.0          py310hf9df320_1    conda-forge
urllib3                   2.2.3              pyhd8ed1ab_0    conda-forge
volk                      3.1.2                hebf3989_0    conda-forge
volk-gnss-sdr             0.0.19               h5833ebf_3    conda-forge
watchdog                  6.0.0           py310h078409c_0    conda-forge
wcwidth                   0.2.13             pyhd8ed1ab_0    conda-forge
websocket-client          1.8.0              pyhd8ed1ab_0    conda-forge
werkzeug                  3.1.3              pyhff2d567_0    conda-forge
wheel                     0.45.1             pyhd8ed1ab_0    conda-forge
wsproto                   1.2.0              pyhd8ed1ab_0    conda-forge
[... cut ...]
zeromq                    4.3.5                hc1bb282_7    conda-forge
zipp                      3.21.0             pyhd8ed1ab_1    conda-forge
zlib                      1.3.1                h8359307_2    conda-forge
zope.event                5.0                pyhd8ed1ab_1    conda-forge
zope.interface            7.2             py310h078409c_0    conda-forge
zstandard                 0.23.0          py310h2665a74_1    conda-forge
zstd                      1.5.6                hb46c0d2_0    conda-forge
```

kgarrels, 2024-12-01:
otool:
I think /usr/lib/libSystem.B.dylib should not be there, it does not exist on my system

michaelld, 2024-12-01:
@kgarrels I've never been to UE gnuradio days; would love to go some day. Ettus/NI/Emerson has folks in Germany & I think some of them go since they are much closer. I'm heading to AOC'24 Dec 9-13 to give the demos showing off the X440 phase coherence & 1.6 GHz IBW (2 GS/s). This demo was created a couple years ago & I've inherited it; had to reverse engineer lots of things on how it both works and is used. Glad to have success. with 1 week to go!

> Hi Michael @michaelld, thank you for paying attention to this issue! By the way, the `conda-forge` channel does have `gnuradio` (https://anaconda.org/conda-forge/gnuradio) and `uhd` (https://anaconda.org/conda-forge/uhd), so one could use Miniforge (https://github.com/conda-forge/miniforge) and run `mamba install gnuradio` to install both GNURadio and the corresponding version of `uhd`. This method is straightforward, cross-platform and facilitates version management. I even wonder why this has not become the standard installation recommendation for GNURadio or UHD, as using MacPorts and other alternatives often leads to challenges such as difficulties in removal and the inability to coexist with multiple versions.

@kgarrels @doctormin I have never used miniconda or miniforge;l trying the minimum of the former to be able to do debugging (I will also try MacPorts for diversity sake). I'm guessing the error I see if the same as y'all are seeing, regardless of how UHD was installed ... Python crashes when one issues "import uhd" with some sort of memory access issue. I ran "lldb python3.12", then "run", then "import uhd" and here's what I see:
```
(lldb) run
Process 66079 launched: '/Users/uhd/miniconda3/bin/python3.12' (arm64)
Python 3.12.7 | packaged by Anaconda, Inc. | (main, Oct  4 2024, 08:22:19) [Clang 14.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import uhd
Process 66079 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x68)
    frame #0: 0x0000000101aee0b4 libpython3.12.dylib`take_gil + 92
libpython3.12.dylib`take_gil:
->  0x101aee0b4 <+92>:  ldr    x8, [x9, #0x68]
    0x101aee0b8 <+96>:  mov    w10, #0xd948 ; =55624
    0x101aee0bc <+100>: movk   w10, #0x5, lsl #16
    0x101aee0c0 <+104>: ldr    x20, [x9, x10]
Target 0: (python3.12) stopped.
```
... which is not hugely informative unfortunately without debugging symbols. I'd be curious if y'all see the same.

In the past (when I was active in MacPorts -- I've not been active for coming up 2 years, which is why the GNU Radio & related ports have not been updated very much) there were 2 primary issues to get UHD working with Python: (1) make sure the library dependencies daisy-chained correctly ... for example IIRC removing the Python library from the .SO being loaded because one is already in Python and trying to reload the library caused DYLD issues; (2) make sure the same libraries and includes -- both installed dependencies as well as system ones -- are from the same source ... for example sometimes CMake would use Apple Python headers but port libraries or vice versa, some sort of mix-and-match, which is never a good thing.

I wonder here whether the issue is that the Python is built using "Clang 14.0.6" whereas we're using "Clang 16.0.0" (or 16.0.6; doesn't really matter) for UHD. While the Clang libraries will be similar between versions, they will be different in subtle ways & those could be enough to throw things off.

Is there a way to force Python to be built from source in conda or mamba? That would be worth a try. Let' keep at it & eventually we'll figure this out!

kgarrels, 2024-12-01:
2GS/s sounds like fun :-)

I think conda's packages are always pre-built, I could remove single packages and try to build them locally, but where to start, uhd? Then I would need some help...

What can we learn from the otool results?
The line ```/Users/runner/miniforge3/conda-bld/uhd_1719620978768/work/host/python/lib/python3.10...``` looks weird

michaelld, 2024-12-01:
yes 2 GS/s *is* fun! The X440 is a pretty amazing radio! Unless one uses bursts of data then it takes quite a beastly server to receive that data stream even on 1 channel ... the X440 supports 2 (either direction for either stream), which is even more fun. Next up will be adding in 1 or 2 GPUs to see if we can stream to them & then they do real-time processing on the data. Need some beefy GPUs!

I seem to remember reading about Conda/Samba that Apple Silicon support was experimental ... am i misremembering? 

I'm using MacPorts to build everything from source & seeing if that install works. If that works I'll store those & change the install to be from pre-built binary.

`otool` is useful to some degree to make sure the binary (lib or so or exe) has the correct load settings (e.g., RPATH) and the dependencies all make sense. macOS is picky about some parts & not soo much about other parts in a binary.

I'm guessing the line you note is a local build directory (`conda-bld`), which happens sometimes when building an SO ... since it is the self-ID of the SO, it doesn't really matter anyway -- though it's nice to have it match the actual install location.

michaelld, 2024-12-03:
In the issue Jörg notes on conda-forge, Ryan notes that his [UHD feedstock](https://github.com/conda-forge/uhd-feedstock) is up to date & even rebuilt it just in case. Following the instructions provided I added this channel to my miniconda install & then installed UHD through it. This updated the Python 3..12 install & now the compilers match for Python and UHD (Clang 16.0.6), and "import uhd" works. This reinforces my belief that the issue is compiler mismatch. I also built UHD 4.7 release in MacPorts & it works fine all around. I don't know enough about conda/mamba/forge to tell people how to resolve this issue, but it's not an issue with UHD.

joergho, 2024-12-03:
@doctormin @kgarrels As mentioned by @michaelld , there is a new build of uhd available in conda-forge (date: 12/03/2024) which is supposed to solve the issue. Please check and confirm.

doctormin, 2024-12-03:
Thanks! I've upgraded to the newest build of `uhd` from conda-forge. 
mamba update --all
```
- uhd                4.7.0.0  py312h11b8700_2  conda-forge     Cached
+ uhd                4.7.0.0  py312hcf98d66_3  conda-forge        5MB
```
Unfortunately, this issue persists.
```
[INFO] [UHD] Mac OS; Clang version 18.1.8 ; Boost_108600; UHD_4.7.0.0-release
python(4789,0x207010240) malloc: *** error for object 0x204be1cc0: pointer being freed was not allocated
python(4789,0x207010240) malloc: *** set a breakpoint in malloc_error_break to debug
[1]    4789 abort      python usrp_setting.py
```

joergho, 2024-12-03:
> Unfortunately, this issue persists.

That's a pity. However, as we learned that this issue is conda-specific, I suggest to continue the discussion here: https://github.com/conda-forge/uhd-feedstock/issues/59

@doctormin Would you mind copying your latest findings there?

doctormin, 2024-12-03:
Sure :)

closed 2024-12-12 by joergho