EttusResearch/uhd #779: [Python API] Incompatible Function Arguments

CarlosD1119, 2024-08-15:
## Issue Description ##
I am using usrp to transmit signals with python API. The code is as follows. I don't find out any misaligned arguments, but there are mistakes.

Code:
```
import uhd
import numpy as np

usrp = uhd.usrp.MultiUSRP()
st_args = uhd.usrp.StreamArgs("fc32", "sc16")
st_args.channels = [0]
tx_streamer = usrp.get_tx_stream(st_args)
md_tx = uhd.types.TXMetadata
md_tx.has_time_spec = False
samples = 0.1*np.random.randn(10000) + 0.1j*np.random.randn(10000) # create random signal
tx_streamer.send(samples, md_tx)
```

Result:
```Traceback (most recent call last):
  File "/home/iiot/PycharmProjects/usrp_tx/new_main.py", line 11, in <module>
    tx_streamer.send(samples, md_tx)
TypeError: send(): incompatible function arguments. The following argument types are supported:
    1. (self: uhd.libpyuhd.usrp.tx_streamer, np_array: object, metadata: uhd.libpyuhd.types.tx_metadata, timeout: float = 0.1) -> int

Invoked with: <uhd.libpyuhd.usrp.tx_streamer object at 0x7f58aed521f0>, array([ 0.05122881-0.02905669j,  0.06272389+0.04570414j,
        0.10575979-0.08321157j, ..., -0.07249551+0.04663707j,
       -0.05430126-0.1516733j , -0.12245535-0.02743819j]), <class 'uhd.libpyuhd.types.tx_metadata'>```

NI-LAm, 2024-08-15:
Here:
`md_tx = uhd.types.TXMetadata`
you do not create a TXMetadata object but you use the underlying pybind11 class, which is not a valid parameter for the streamer send method. Instead you need:
`md_tx = uhd.types.TXMetadata()`
mind the brackets which will create an instance of the object instead of assigning the class to md_tx!
Then you have a real metadata object and your code should work.

closed 2024-08-15 by NI-LAm

CarlosD1119, 2024-08-15:
It works. Many thanks.