EttusResearch/uhd #781: Constantly throwing the errors in uhd_usrp_probe

YonghoonChung, 2024-08-21:
## Issue Description ##
After completing the installation of UHD on Ubuntu 22.04 LTS, I was able to successfully run uhd_find_devices. However, when I attempt to use uhd_usrp_probe, I encounter the following error:
```
[ERROR] [MPMD::MB_IFACE] Unable to look up clock '1'. Invalid index
[ERROR] [RFNOC::GRAPH] Caught exception while initializing graph: LookupError: KeyError: Unable to look up clock '1'. Invalid index
Error: RuntimeError: Failure to create rfnoc_graph.
```
I have tried loading new images onto my USRP X410, but the same error persists. This issue also occurs when using the Python API with uhd.usrp.MultiUSRP().

## Setup Details ##
    Operating System: Ubuntu 22.04 LTS
    UHD Version: UHD_4.7.0.0
    Hardware: USRP X410

Below is the output of uhd_usrp_probe:
etri@etri-ThinkStation-P3-Ultra:/usr/local/lib/python3.10/dist-packages$ uhd_usrp_probe 
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.7.0.0-0-ga5ed1872
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.20.2,type=x4xx,product=x410,serial=329C697,name=ni-x4xx-329C697,fpga=X4_200,claimed=False,addr=192.168.20.2
[INFO] [MPM.PeriphManager] init() called with device args `fpga=X4_200,mgmt_addr=192.168.20.2,name=ni-x4xx-329C697,product=x410,clock_source=internal,time_source=internal,initializing=True'.

## Additional Information ##
While this may not be a critical issue, I did notice an error during the installation of UHD with the following commands:
cmake -DENABLE_PYTHON_API=ON .
make
make install

The error appeared in the __init__ phase. Could this be related to the issue I'm experiencing?
Any guidance on resolving this would be greatly appreciated.

wordimont, 2024-08-21:
Can you try updating the FPGA image? I'm wondering if the FPGA image is older than the UHD version you're using.

https://kb.ettus.com/USRP_X410/X440_Getting_Started_Guide#Updating_the_FPGA

mbr0wn, 2024-10-01:
I'm also pretty sure that running `uhd_image_loader` with a recent image will fix this. Closing this -- please reopen if you think this is a legitimate issue.

closed 2024-10-01 by mbr0wn