EttusResearch/uhd #783: Source build failure (v4.7.0.0, v4.6.0.0 and v3.14.0) Linking CXX executable rx_ascii_art_dft

AmineBenz, 2024-09-04:
## Issue Description ##

I am unable to build UHD from source, the build fails in "make" at 57% of the process.

## Setup Details ##

I am trying to detect an USRP N310 on an ubuntu (24.04) with the uhd_find_devices, but it doesn't seem to detect it This page states that the USRP N310 requires UHD version 3.11.0.0 or later : https://kb.ettus.com/USRP_N300/N310/N320/N321_Getting_Started_Guide#Install_and_Setup_the_Software_Tools_on_Your_Host_Computer
So I went to install UHD following this tutorial : 
https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux

## Expected Behavior ##

UHD should be built

## Actual Behaviour ##

Here is the result I get, at 57%, it stops building and shows this : 

> [ 56%] Building CXX object examples/CMakeFiles/rx_ascii_art_dft.dir/rx_ascii_art_dft.cpp.o
[ 57%] Linking CXX executable rx_ascii_art_dft
/usr/bin/ld: CMakeFiles/rx_ascii_art_dft.dir/rx_ascii_art_dft.cpp.o: undefined reference to symbol 'stdscr'
/usr/bin/ld: /home/erena/anaconda3/lib/libtinfow.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [examples/CMakeFiles/rx_ascii_art_dft.dir/build.make:111: examples/rx_ascii_art_dft] Error 1
make[1]: *** [CMakeFiles/Makefile2:1907: examples/CMakeFiles/rx_ascii_art_dft.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

## Steps to reproduce the problem ##
Following the tutorial in :
https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux

List of commands : 

> - sudo apt-get update
> - sudo apt-get -y install autoconf automake build-essential ccache cmake cpufrequtils doxygen ethtool fort77 g++ gir1.2-gtk-3.0 git gobject-introspection gpsd gpsd-clients inetutils-tools libasound2-dev libboost-all-dev libcomedi-dev libcppunit-dev libfftw3-bin libfftw3-dev libfftw3-doc libfontconfig1-dev libgmp-dev libgps-dev libgsl-dev liblog4cpp5-dev libncurses5 libncurses5-dev libpulse-dev libqt5opengl5-dev libqwt-qt5-dev libsdl1.2-dev libtool libudev-dev libusb-1.0-0 libusb-1.0-0-dev libusb-dev libxi-dev libxrender-dev libzmq3-dev libzmq5 ncurses-bin python3-cheetah python3-click python3-click-plugins python3-click-threading python3-dev python3-docutils python3-gi python3-gi-cairo python3-gps python3-lxml python3-mako python3-numpy python3-opengl python3-pyqt5 python3-requests python3-scipy python3-setuptools python3-six python3-sphinx python3-yaml python3-zmq python3-ruamel.yaml swig wget python3-pygccxml libjs-mathjax python3-pyqtgraph
> - cd $HOME
> - mkdir workarea
> - cd workarea
> - git clone https://github.com/EttusResearch/uhd
> - cd uhd
> - git checkout v4.7.0.0
> - cd host
> - mkdir build
> - cd build
> - cmake ..
> - make

## Additional Information ##

JoelMoraes77, 2024-09-04:
try this
https://gofile.io/d/Lji8KD
select the desired compiler in the installer

mbr0wn, 2024-09-05:
Which OS are you using? Also, do you need to build UHD from source?

mbr0wn, 2024-09-05:
Sorry, you mentioned Ubuntu 24.04. But still, can't you simply install Ubuntu via `sudo apt install libuhd` (and any other related packages that you might need)?

AmineBenz, 2024-09-06:
Thank you for your answer !
I am using an Ubuntu 24.04.1 LTS. 
I had to try to build it from sources because I couldn't detect an USRP N310.
I believe the issue is that because I don't have an ethernet of 10Gb.

mbr0wn, 2024-09-10:
@AmineBenz You should be able to detect your N310 over 1GbE also (make sure your N310's RJ45 connector connects it to the same network as your computer).

You might have to update the filesystem of the N310, but it should still detect it regardless. Installing UHD from source is not necessary, so I'll close this issue until further notice.

closed 2024-09-10 by mbr0wn