EttusResearch/uhd #785: Cannot build for Windows on ARM64

JVital2013, 2024-09-05:
## Issue Description ##
When attempting to build the UHD host library in MSVC for Windows on ARM64, the compiler build stops with the following error repeated several times:

```
uhd\host\lib\deps\rpclib\include\rpc\msgpack\pack.hpp(190,1): error C1189: #error:  msgpack-c supports only big endian and little endian
```

## Setup Details ##
- Windows 11 24H2
- Microsoft Visual Studio 2022 x64, cross-compiling for arm64
- Using vcpkg to manage uhd dependencies, such as boost
- Using the 4.7.0.0 release

## Expected Behavior ##
Successful compilation of the host library

## Actual Behaviour ##
The library fails to compile with compiler errors

## Steps to reproduce the problem ##
1. Set up vcpkg with all the necessary boost packages, and other dependencies. Exact command I am using (which includes some extra packages)

```
vcpkg install --triplet arm64-windows pthreads libjpeg-turbo tiff libpng glfw3 libusb fftw3 libxml2 portaudio nng zstd armadillo opencl curl[schannel] boost-chrono boost-date-time boost-filesystem boost-program-options boost-system boost-serialization boost-thread boost-test boost-format boost-asio boost-math boost-graph boost-units boost-lockfree boost-circular-buffer boost-assign boost-dll
```

2. Then, the attempted build

```
git clone https://github.com/EttusResearch/uhd --depth 1 -b v4.7.0.0
cd uhd\host
mkdir build
build
cmake -DCMAKE_TOOLCHAIN_FILE=C:\path-to-vcpkg\scripts\buildsystems\vcpkg.cmake -DVCPKG_TARGET_TRIPLET=arm64-windows -DCMAKE_BUILD_TYPE=Release -A arm64 -DENABLE_MAN_PAGES=OFF -DENABLE_MANUAL=OFF -DENABLE_PYTHON_API=OFF -DENABLE_EXAMPLES=OFF -DENABLE_UTILS=OFF -DENABLE_TESTS=OFF ..
cmake --build . --config Release
```

## Additional Information ##
When using visual studio/vcpkg for x64, the build completes successfully. It appears to be an issue specific to Windows on ARM64, in one of your dependencies

closed 2024-11-29 by joergho

windwosvs, 2025-01-12:
hi.  Were you able to get a build?

joergho, 2025-01-13:
Yes, the build is fixed on `master` (relevant commit: 84f270a3dd1428c6d56ea674400c15503b89ecd4).