EttusResearch/uhd #792: [ERROR] [RFNOC::MGMT] EnvironmentError: IOError: recv error on socket: Connection refused

houshangaziziaccelleran, 2024-10-04:
Hi
Can someone help with the following issue please?
```
ad@bm-super11-intel:/usr/local/lib/uhd/examples$ sudo uhd_usrp_probe 
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; DPDK_23.11; UHD_4.7.0.HEAD-0-ga5ed1872
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=10.10.0.100,type=n3xx,product=n310,serial=32000F1,name=ni-n3xx-32000F1,fpga=HG,claimed=False,addr=10.10.2.100
[INFO] [MPM.PeriphManager] init() called with device args `fpga=HG,mgmt_addr=10.10.0.100,name=ni-n3xx-32000F1,product=n310,clock_source=internal,time_source=internal'.
[ERROR] [RFNOC::MGMT] EnvironmentError: IOError: recv error on socket: Connection refused
[ERROR] [RFNOC::GRAPH] IO Error during GSM initialization. EnvironmentError: IOError: recv error on socket: Connection refused
[ERROR] [RFNOC::GRAPH] Caught exception while initializing graph: EnvironmentError: IOError: recv error on socket: Connection refused
Error: RuntimeError: Failure to create rfnoc_graph.
ad@bm-super11-intel:/usr/local/lib/uhd/examples$ 
```

I have an n310. # IP ADDR: ETH for management (10.10.0.100) and two fiber for DPDK use (10.10.1.100 and 10.10.2.100).
```

ad@bm-super11-intel:/usr/local/lib/uhd/examples$ sudo uhd_find_devices
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; DPDK_23.11; UHD_4.7.0.HEAD-0-ga5ed1872
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 32000F1
    addr: 10.10.2.100
    claimed: False
    fpga: HG
    mgmt_addr: 10.10.0.100
    mgmt_addr: 10.10.1.100
    mgmt_addr: 10.10.2.100
    name: ni-n3xx-32000F1
    product: n310
    type: n3xx

ad@bm-super11-intel:/usr/local/lib/uhd/examples$ 
```

This is my UHD version:

```
ad@bm-super11-intel:/usr/local/lib/uhd/examples$ uhd_config_info --version
UHD 4.7.0.HEAD-0-ga5ed1872
ad@bm-super11-intel:/usr/local/lib/uhd/examples$ 
```

And here are my FPGA and MPM versions:

```
ad@bm-super11-intel:/usr/local/lib/uhd/examples$ uhd_usrp_probe --args "addr=10.10.0.100"
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; DPDK_23.11; UHD_4.7.0.HEAD-0-ga5ed1872
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=10.10.0.100,type=n3xx,product=n310,serial=32000F1,name=ni-n3xx-32000F1,fpga=HG,claimed=False,addr=10.10.0.100
[INFO] [MPM.PeriphManager] init() called with device args `fpga=HG,mgmt_addr=10.10.0.100,name=ni-n3xx-32000F1,product=n310,clock_source=internal,time_source=internal'.
  _____________________________________________________
 /
|       Device: N300-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: ni-n3xx-32000F1
|   |   dboard_0_pid: 336
|   |   dboard_0_serial: 31F6BD1
|   |   dboard_1_pid: 336
|   |   dboard_1_serial: 31F6BD5
|   |   eeprom_version: 3
|   |   fs_version: 20240628134248
|   |   mender_artifact: v4.7.0.0_n3xx
|   |   mpm_sw_version: 4.7.0.0-ga5ed1872
|   |   pid: 16962
|   |   product: n310
|   |   rev: 10
|   |   rpc_connection: remote
|   |   serial: 32000F1
|   |   type: n3xx
|   |   MPM Version: 5.3
|   |   FPGA Version: 8.2
|   |   FPGA git hash: c37b318.clean
|   |   RFNoC capable: Yes

```

closed 2024-10-07 by mbr0wn

houshangaziziaccelleran, 2024-10-07:
Thanks for the reply @mbr0wn!
Is the instruction in [this manual ](https://uhd.readthedocs.io/en/latest/page_dpdk.html) valid for my case as well?
My case: I want to use DPDK on my n310 with two SFP dedicated for DPDK.

houshangaziziaccelleran, 2024-10-07:
I get such error message. Is there a compatibility issue (UHD- DPDK)?

mbr0wn, 2024-10-09:
If you're using current master, you're good, but the latest release does have issues with DPDK 23.11.

mbr0wn, 2024-10-09:
@houshangaziziaccelleran like I said, this is not a support forum. This is for bug reports only. In your case, turn off DPDK, make sure communication works with the device, then from there craft a working DPDK configuration. Take further questions to the mailing list.