EttusResearch/uhd #796: Error on using DDC RFNoC block.

NIDHIPANDA, 2024-10-22:
Hello,

I am having USRP X300 device with following tool versions:

Vivado 2021.1 - AR76780n,
GNU radio version - v3.11.0.0git-820-g2adbd4ea
UHD version - UHD_4.7.0.0-84-gbdada1ed

I am trying to use the default example present in the GNU radio example folder "rfnoc_radio_ddc.grc". I am  able to down convert the input signal with the desired frequency however on using frequency shift feature it hives following error:

[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
[ERROR] [RFNOC::GRAPH::DETAIL] 0/DDC#0: RfnocError: ResolveError: Attempting to overwrite property `freq@USER:0' with a new value after it was locked!
Traceback (most recent call last):
  File "/home/quasar/gnuradio/gr-uhd/examples/grc/rfnoc_radio_ddc.py", line 330, in <module>
    main()
  File "/home/quasar/gnuradio/gr-uhd/examples/grc/rfnoc_radio_ddc.py", line 309, in main
    tb.start()
  File "/usr/local/lib/python3.8/dist-packages/gnuradio/gr/top_block.py", line 116, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
RuntimeError: RfnocError: ResolveError: Attempting to overwrite property `freq@USER:0' with a new value after it was locked!

I have used the default .bit file which includes DDC blocks. Please suggest how to resolve this.

Regards,
Nidhi

mbr0wn, 2025-01-23:
@NIDHIPANDA can you specify how exactly you triggered the error? I thought I'd reproduced this bug before, but I can't right now.

This is what I  tried:
- Setting a different default value for the DDC frequency
- Switching the DDC frequency at runtime

Both of these actions work fine.

NIDHIPANDA, 2025-01-23:
Hello,

The issue was with frequency shift operation in DDC block. The output rate
works fine with run time changes in frequency but the offset was not
working with uhd 4.7 version. It works with uhd 4.1

mbr0wn, 2025-01-24:
@NIDHIPANDA I understand the frequency shift operation was the issue; but I could not reproduce this (I was using HEAD of master, which is basically the 4.8 Release Candidate).

mbr0wn, 2025-01-24:
Actually, this bug is related to https://github.com/EttusResearch/uhd/issues/763 (and then the same fix applies for both issues).

I just checked; 635ad362ae60f82b3131970d6f8171505dce6ac7 fixes this issue also.