EttusResearch/uhd #805: Failed to rebuild FPGA image for X310

tomwts, 2024-10-31:
When I rebuilt an example FPGA image for X310, I got a compilation error:

/usr/bin/rfnoc_image_builder -F ${MY_ETTUS_REPO}/fpga -I ${MY_ETTUS_REPO}/host/examples/rfnoc-example -y ${MY_ETTUS_REPO}/uhd/host/examples/rfnoc-example/icores/x310_rfnoc_image_core.yml -t X310_XG
Below is the build log:

> Setting up a 64-bit FPGA build environment for the USRP-X3x0...
> - Vivado: Found (/opt/Xilinx/Vivado/2024.1/bin)
> 
> Environment successfully initialized.
> make -f Makefile.x300.inc bin NAME=X310_XG ARCH=kintex7 PART_ID=xc7k410t/ffg900/-2 BUILD_10G=1 SFP0_10GBE=1 SFP1_10GBE=1  X310=1  TOP_MODULE=x300 EXTRA_DEFS="BUILD_10G=1 SFP0_10GBE=1 SFP1_10GBE=1  X310=1 "
> make[1]: Entering directory '/shares/projectdata/users/johndoe/ettus/fpga/usrp3/top/x300'
> BUILDER: Checking tools...
> * GNU bash, version 5.2.21(1)-release (x86_64-pc-linux-gnu)
> * Python 3.12.3
> * vivado v2024.1 (64-bit)
> ========================================================
> BUILDER: Building IP ten_gig_eth_pcs_pma
> ========================================================
> BUILDER: Staging IP in build directory...
> BUILDER: Reserving IP location: /shares/projectdata/users/johndoe/ettus/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/ten_gig_eth_pcs_pma
> BUILDER: Retargeting IP to part kintex7/xc7k410t/ffg900/-2...
> BUILDER: Building IP...
> [00:00:00] Executing command: vivado -mode batch -source /shares/projectdata/users/johndoe/ettus/fpga/usrp3/tools/scripts/viv_generate_ip.tcl -log ten_gig_eth_pcs_pma.log -nojournal
> WARNING: [Device 21-436] No parts matched 'xc7k410tffg900-2'
> ERROR: [Coretcl 2-106] Specified part could not be found.
> [00:00:04] Current task: Initialization +++ Current Phase: Starting
> [00:00:04] Current task: Initialization +++ Current Phase: Finished
> [00:00:04] Process terminated. Status: Failure
> 
> ========================================================
> Warnings:           1
> Critical Warnings:  0
> Errors:             1

OS: Ubuntu 24.04
UHD: 4.7.0.0
Vivado 2024.1 ML Enterprise

It seems a Vivado license issue. I do have a 30 days eval license installed. Is it really a license issue?
I included FPGA Kintex7 in Vivado 2024.1 installation.

Thanks,
Tom

wordimont, 2024-10-31:
You need to install Vivado 2021.1. You should get an error saying you have the wrong Vivado version if you try it with 2024.1. If you didn't then that would be a bug.

Make sure you clean out all the build and build-* files in the x300 directory to clear out any thing that Vivado 2024.1 might have created, otherwise you'll probably see more errors when switching to the correct version.

Double check during the install that you have 7 Series Kintex 7 support checked.

If that doesn't work, then maybe it's a license issue, but try the above first.

You can find more details in the manual:
https://files.ettus.com/manual/md_usrp3_build_instructions.html

closed 2024-10-31 by wordimont

tomwts, 2024-11-04:
Hi wordimont, thanks for the info. I installed Vivado 2021.1 plus patch. The image build for X310 partially worked. It didn't complain "No parts matched 'xc7k410tffg900-2'". The build made some progress but stopped with another error:

make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/shares/projectdata/users/johnjdoe/ettus/uhd/fpga/usrp3/top/x300'
BUILDER: Checking tools...
* GNU bash, version 5.2.21(1)-release (x86_64-pc-linux-gnu)
* Python 3.12.3
* Vivado v2021.1_AR76780 (64-bit)
Build directory:: /shares/projectdata/users/johndoe/ettus/uhd/fpga/usrp3/top/x300/build-usrp_x310_fpga_HG
Printing MB_XDC:: 
Printing VERILOG_DEFS:: BUILD_1G=1 BUILD_10G=1 SFP0_1GBE=1 SFP1_10GBE=1  BUILD_SEED=0 X310=1  GIT_HASH=32'hfa5ed187 RFNOC_IMAGE_CORE_HDR=/shares/projectdata/users/johndoe/ettus/uhd/fpga/usrp3/top/x300/build-usrp_x310_fpga_HG/rfnoc_image_core.vh 
Using parser configuration from: /shares/projectdata/users/johndoe/ettus/uhd/fpga/usrp3/top/x300/dev_config.json
[00:00:00] Executing command: vivado -mode batch -source /shares/projectdata/users/johndoe/ettus/uhd/fpga/usrp3/top/x300/build_x300.tcl -log build.log -journal x300.jou
ERROR: [Common 17-69] Command failed: File '/shares/projectdata/users/johndoe/ettus/uhd/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/ten_gig_eth_pcs_pma_ex/imports/ten_gig_eth_pcs_pma_example_design.v' does not exist
[00:00:19] Current task: Initialization +++ Current Phase: Starting
[00:00:19] Current task: Initialization +++ Current Phase: Finished
[00:00:19] Process terminated. Status: Failure

I searched through my UHD 4.7 local git repo but couldn't find file ten_gig_eth_pcs_pma_example_design.v.
The file is referenced by ip-core/ten_gig_eth_pcs_pma/Mikefile.inc.

I checked the git online repo for UHD 4.7 but couldn't find that file neither.
I am unsure if there is an issue with UHD 4.7 for X310 support or not.

wordimont, 2024-11-04:
That's missing file is from Xilinx IP that's generated during the build. You probably need to delete the `build-ip` directory and regenerate a clean copy of the IP. You may still have files left over from trying to build with the other Vivado version.

tomwts, 2024-11-04:
I missed the critical step of doing "make X310_IP" first. After it finished successfully, the RFNOC image build: rfnoc_image_builder -y x310_HG_rfnoc_image_core.yml -t X310_HG complete successfully!

Thanks once again, wordimont!

mbr0wn, 2024-11-06:
Running `make X310_IP` first is not required. If you have a clean build, the image builder will do all that for you.

tomwts, 2024-11-07:
mbr0wn, wordimont,

Even though the X310_HG image build was successful. There were many repeated critical warnings:

RITICAL WARNING: [Constraints 18-1056] Clock 'FPGA_CLK' completely overrides clock 'FPGA_CLK_p'.
CRITICAL WARNING: [Vivado 12-4739] set_clock_groups:No valid object(s) found for '-group [get_clocks bus_clk]'. [/shares/projectdata/users/johndoe/ettus/uhd/fpga/usrp3/top/x300/timing.xdc:72]
CRITICAL WARNING: [Vivado 12-4739] set_clock_groups:No valid object(s) found for '-group '. [/shares/projectdata/users/johndoe/ettus/uhd/fpga/usrp3/top/x300/timing.xdc:72]
CRITICAL WARNING: [Vivado 12-4739] set_clock_groups:No valid object(s) found for '-group [get_clocks ioport2_clk]'. [/shares/projectdata/users/johndoe/ettus/uhd/fpga/usrp3/top/x300/timing.xdc:72]
CRITICAL WARNING: [Vivado 12-4739] set_clock_groups:No valid object(s) found for '-group '. [/shares/projectdata/users/johndoe/ettus/uhd/fpga/usrp3/top/x300/timing.xdc:72]

CRITICAL WARNING: [Timing 38-469] The REFCLK pin of IDELAYCTRL u_ddr3_32bit/u_ddr3_32bit_mig/u_iodelay_ctrl/u_idelayctrl_200 has a clock period of 4.998 ns (frequency 200.080 Mhz) but IDELAYE2 u_ddr3_32bit/u_ddr3_32bit_mig/u_memc_ui_top_axi/mem_intfc0/ddr_phy_top0/u_ddr_mc_phy_wrapper/u_ddr_mc_phy/ddr_phy_4lanes_0.u_ddr_phy_4lanes/ddr_byte_lane_A.ddr_byte_lane_A/ddr_byte_group_io/input_[1].iserdes_dq_.idelay_dq.idelaye2 has REFCLK_FREQUENCY of 200.000 Mhz (period 5.000 ns). The IDELAYCTRL REFCLK pin frequency must match the IDELAYE2 REFCLK_FREQUENCY property.

Are these critical warnings normal?

wordimont, 2024-11-08:
Yes, that's expected. See:

https://kb.ettus.com/RFNoC_Frequently_Asked_Questions#I_see_a_.22CRITICAL_WARNING.22_in_the_build_log._Is_this_expected.3F