EttusResearch/uhd #807: Error when creating uhd::stream_args_t: "string too long"

tantanli, 2024-11-04:
**Description:**
I encountered a runtime error while trying to create an instance of `uhd::stream_args_t` with the following parameters:

```cpp
uhd::stream_args_t stream_args("fc32", "sc16");
```
Error Message:
```cpp
string too long
```
UHD Version: uhd_4.7.0.0-release_Win64_VS2017.exe
Compiler: Visual Studio 2017
I have tried different combinations of CPU and OTW formats but still encounter the same error.
Using the default constructor does not produce the error.
Thank you for your help!

tantanli, 2024-11-05:
I ran into some issues with examples that seemed to be fixed before, saying it should work with UHD version 3.14.1.0 or later. But mine is 4.6.0.0, and I’m not sure if that’s the problem.See [#198](https://github.com/EttusResearch/uhd/issues/198)
```C
Setting device timestamp to 0...

Begin streaming 10000 samples, 1.500000 seconds in the future...
[ERROR] [STREAMER] recv packet demuxer unexpected sid 0x1ffff
[ERROR] [STREAMER] recv packet demuxer unexpected sid 0x10004
[ERROR] [STREAMER] recv packet demuxer unexpected sid 0xfffafff9
[ERROR] [STREAMER] recv packet demuxer unexpected sid 0xc0008
[ERROR] [STREAMER] recv packet demuxer unexpected sid 0xfffcfff5
Receive timeout before all samples received...
```
And here is the version
```C
C:\Program Files\UHD\lib\uhd\examples>uhd_config_info --print-all
UHD 4.6.0.0-release
Build date: Tue, 14 Nov 2023 18:12:08
C compiler: MSVC 2015
C++ compiler: MSVC 2015
C flags: /D WIN32_LEAN_AND_MEAN /D VC_EXTRALEAN /D _SCL_SECURE_NO_WARNINGS /D _SCL_SECURE_NO_DEPRECATE /D _CRT_SECURE_NO_WARNINGS /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_WARNINGS /D _CRT_NONSTDC_NO_DEPRECATE /D _WINSOCK_DEPRECATED_NO_WARNINGS /D _WIN32_WINNT=0x0501 /D NOMINMAX /D BOOST_ERROR_CODE_HEADER_ONLY /D BOOST_ASIO_DISABLE_STD_STRING_VIEW /D BOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW /D HAVE_CONFIG_H /D UHD_LOG_MIN_LEVEL=1 /D UHD_LOG_CONSOLE_LEVEL=2 /D UHD_LOG_FILE_LEVEL=2/DWIN32 /D_WINDOWS /W3
C++ flags: /D WIN32_LEAN_AND_MEAN /D VC_EXTRALEAN /D _SCL_SECURE_NO_WARNINGS /D _SCL_SECURE_NO_DEPRECATE /D _CRT_SECURE_NO_WARNINGS /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_WARNINGS /D _CRT_NONSTDC_NO_DEPRECATE /D _WINSOCK_DEPRECATED_NO_WARNINGS /D _WIN32_WINNT=0x0501 /D NOMINMAX /D BOOST_ERROR_CODE_HEADER_ONLY /D BOOST_ASIO_DISABLE_STD_STRING_VIEW /D BOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW /D HAVE_CONFIG_H /D UHD_LOG_MIN_LEVEL=1 /D UHD_LOG_CONSOLE_LEVEL=2 /D UHD_LOG_FILE_LEVEL=2/DWIN32 /D_WINDOWS /W3 /GR /EHsc
Enabled components: LibUHD, LibUHD - C API, Examples, Utils, Tests, USB, B100, B200, USRP1, USRP2, X300, MPMD, N300, N320, E320, E300, X400, OctoClock
Install prefix: C:/Program Files/UHD
Boost version: 1.78
Libusb version: 1.0.26
Library path: C:\Program Files\UHD\bin
Package path: C:\Program Files\UHD
Images directory: C:\Program Files\UHD\share\uhd\images
ABI version string: 4.6.0
```

mbr0wn, 2024-11-06:
@tantanli we can't reproduce this, and it's not clear what your issue is (your first and second comment have different content). I'm going to close this. If you think you have an actual bug in UHD, then please reopen with more detail. Thanks!

closed 2024-11-06 by mbr0wn