EttusResearch/uhd #808: Add dmafifo to RFNOC image for X310

tomwts, 2024-11-04:
I want to add DMAFIFO rfnoc block to X310 HG image. I discovered the UHD 4.7 version x310_XG_rfnoc_image_core.yml, doesn't have duc0.yml, ddc0.yml, duc1.yml, ddc1.yml defined under noc_blocks. Though UHD 4.6 version x310_rfnoc_image_core.yml has those 4 blocks defined. However, UHD 4.6 version doesn't have HG and XG x310_rfnoc_image_core.yml file.

I suspect the prebuilt UHD 4.7 X310 rfnoc image HG was not built from x310_HG_rfnoc_image_core.yml because:

#1.  uhd_usrp_probe output of X310 with prebuilt image installed shows it has DUC and DDC rfnoc blocks available.

#2.  I built UHD 4.7 release X310_HG image successfully but when performing a binary compare between my image vs the prebuilt HG image, they are very different. The rpt output are also different.

Questions:

#1. What .yml file the UHD 4.7 prebuilt HG image is based on if uhd/fpga/usrp3/top/x300/x310_HG_rfnoc_image_core.yml was not used for the build?

#2. If UHD 4.7 version x310_HG_rfnoc_image_core.yml indeed doesn't support DUC and DDC, can I just use the UHD 4.6 version x310_rfnoc_image_core.yml? If the answer is no, how to add DUC and DDC back to x310_HG_rfnoc_image_core.yml?

Thanks,
Tom

wordimont, 2024-11-04:
That is the correct YAML file for the HG image and it does include the DDC and DUC by inheriting it from another file.

Notice the `inherit` section here:
https://github.com/EttusResearch/uhd/blob/UHD-4.7/fpga/usrp3/top/x300/x310_HG_rfnoc_image_core.yml#L14

So the DDC/DUC are included from here:
https://github.com/EttusResearch/uhd/blob/UHD-4.7/fpga/usrp3/top/x300/yaml_include/x3xx_radio_base.yml

You probably could also use the 4.6 version, but there may have been some other changes besides the YAML include files.

closed 2024-11-04 by wordimont

tomwts, 2024-11-05:
Hi Wordimont.

Thanks for the explanation. The x3xx_radio_base.yml caught me off guard. It makes sense now!

I am sorry to throw in a different issue #798 to this closed ticket. I opened the ticket for some time, but it seems nobody knows the root cause of the issue. If I specify the Rx radio and Tx radio to be the same radio block, the app failed with a no route found error. (See my issue ticket #798 for details). My app has a similar issue that it can't use the same radio block for Tx and Rx. I thought using example app rfnoc_radio_loopback that came with UHD 4.7 release would be easier for the support team to reproduce the issue.

Anyway, I enabled trace on rfnoc_radio_loopback.cpp, rebuilt it. Below is the trace output with both RX2 and TX/RX configured on the same 0/Radio#1 block

Creating the RFNoC graph with args: addr=192.168.40.2...
[INFO] [UHD] linux; GNU C++ version 13.2.0; Boost_108300; UHD_4.7.0.0-0-ga5ed1872
[DEBUG] [MPMD] Discovering MPM devices on port 49600
[INFO] [X300] X300 initialization sequence...
[DEBUG] [X300] Motherboard 0 has remote device ID: 1
[DEBUG] [X300] Setting up basic communication...
[DEBUG] [X300] Using FPGA version: 39.3 git hash: c37b318
[DEBUG] [X300] Loading values from EEPROM...
[DEBUG] [X300] Determining maximum frame size... 
[INFO] [X300] Maximum frame size: 8000 bytes.
[DEBUG] [X300] Setting up RF frontend clocking...
[DEBUG] [X300] x300_clock_ctrl::set_clock_delay: Which=8, Requested=0.000000, Digital Taps=5, Half Shift=OFF, Analog Delay=0 (OFF), Coerced Delay=0.000000ns
[DEBUG] [X300] x300_clock_ctrl::set_clock_delay: Which=4, Requested=0.000000, Digital Taps=5, Half Shift=OFF, Analog Delay=0 (OFF), Coerced Delay=0.000000ns
[DEBUG] [X300] x300_clock_ctrl::set_clock_delay: Which=5, Requested=0.000000, Digital Taps=5, Half Shift=OFF, Analog Delay=0 (OFF), Coerced Delay=0.000000ns
[DEBUG] [X300] x300_clock_ctrl::set_clock_delay: Which=0, Requested=0.000000, Digital Taps=5, Half Shift=OFF, Analog Delay=0 (OFF), Coerced Delay=0.000000ns
[DEBUG] [X300] x300_clock_ctrl::set_clock_delay: Which=2, Requested=0.000000, Digital Taps=5, Half Shift=OFF, Analog Delay=0 (OFF), Coerced Delay=0.000000ns
[INFO] [X300] Radio 1x clock: 200 MHz
[DEBUG] [X300] Device DNA: 0068ED012741A854
[DEBUG] [X300] Motherboard 0 has local device IDs: 
[DEBUG] [X300] * 2
[DEBUG] [RFNOC::MGMT] Starting topology discovery from device[local]:2/sep:1
[DEBUG] [RFNOC::MGMT] Discovered node device:1/xport:1
[DEBUG] [RFNOC::MGMT] Initialized node device:1/xport:1
[DEBUG] [RFNOC::MGMT] Discovered node device:1/xbar:0
[DEBUG] [RFNOC::MGMT] Initialized node device:1/xbar:0
[DEBUG] [RFNOC::MGMT] Discovered node device:1/sep:0
[DEBUG] [RFNOC::MGMT] Initialized node device:1/sep:0
[DEBUG] [RFNOC::MGMT] Discovered node device:1/sep:1
[DEBUG] [RFNOC::MGMT] Initialized node device:1/sep:1
[DEBUG] [RFNOC::MGMT] Discovered node device:1/sep:2
[DEBUG] [RFNOC::MGMT] Initialized node device:1/sep:2
[DEBUG] [RFNOC::MGMT] Discovered node device:1/sep:3
[DEBUG] [RFNOC::MGMT] Initialized node device:1/sep:3
[DEBUG] [RFNOC::MGMT] Discovered node device:1/sep:4
[DEBUG] [RFNOC::MGMT] Initialized node device:1/sep:4
[DEBUG] [RFNOC::MGMT] Discovered node device:1/sep:5
[DEBUG] [RFNOC::MGMT] Initialized node device:1/sep:5
[DEBUG] [RFNOC::MGMT] The following endpoints are reachable from device[local]:2/sep:1
[DEBUG] [RFNOC::MGMT] * 1:0
[DEBUG] [RFNOC::MGMT] * 1:1
[DEBUG] [RFNOC::MGMT] * 1:2
[DEBUG] [RFNOC::MGMT] * 1:3
[DEBUG] [RFNOC::MGMT] * 1:4
[DEBUG] [RFNOC::MGMT] * 1:5
[DEBUG] [RFNOC::LSM] Adding node device:1/xport:0 to topology graph outside of discovery.
[DEBUG] [RFNOC::LSM] Adding transport adapter on xbar port 0
[DEBUG] [RFNOC::GRAPH] Connecting the Host to Endpoint 1:0 through Adapter 0 (0 = no preference)... 
[DEBUG] [RFNOC::MGMT] Throttling stream endpoint to 100% (0x0)
[DEBUG] [RFNOC::MGMT] Bound stream endpoint with Addr=(1,0) to EPID=2
[DEBUG] [RFNOC] Started thread uhd_ctrl_ep0001 to process messages control messages on EPID 1
[DEBUG] [RFNOC::MGMT] Established a route from EPID=1 (SW) to EPID=2
[DEBUG] [RFNOC] Created ctrlport endpoint for port 0 on EPID 1
[DEBUG] [RFNOC::GRAPH] Connection to Endpoint 1:0 completed through Device 2. Using EPIDs 1 -> 2.
[DEBUG] [RFNOC] Created ctrlport endpoint for port 2 on EPID 1
[DEBUG] [0/DUC#0] Checking compat number for FPGA component `0/DUC#0': Expecting 0.1, actual: 0.1.
[DEBUG] [0/DUC#0] Loading DUC with 3 halfbands and max CIC interpolation 255
[DEBUG] [RFNOC::BLOCK_CONTAINER] Registering block: 0/DUC#0 (NOC ID=d0c00000)
[DEBUG] [RFNOC] Created ctrlport endpoint for port 3 on EPID 1
[DEBUG] [0/DDC#0] Checking compat number for FPGA component `0/DDC#0': Expecting 0.1, actual: 0.1.
[DEBUG] [0/DDC#0] Loading DDC with 3 halfbands and max CIC decimation 255
[DEBUG] [RFNOC::BLOCK_CONTAINER] Registering block: 0/DDC#0 (NOC ID=ddc00000)
[DEBUG] [RFNOC] Created ctrlport endpoint for port 4 on EPID 1
[... 107 lines cut ...]
[DEBUG] [0/Replay#0] Checking compat number for FPGA component `0/Replay#0': Expecting 1.2, actual: 1.2.
[DEBUG] [RFNOC::BLOCK_CONTAINER] Registering block: 0/Replay#0 (NOC ID=4e91a000)
[DEBUG] [0/DDC#1] Not setting frequency until sampling rate is set.
[DEBUG] [0/DDC#1] Not setting frequency until sampling rate is set.
[DEBUG] [0/DUC#1] Not setting frequency until sampling rate is set.
[DEBUG] [0/DDC#0] Not setting frequency until sampling rate is set.
[DEBUG] [0/DDC#0] Not setting frequency until sampling rate is set.
[... cut ...]
[DEBUG] [RFNOC::MGMT] Programmed a route from EPID=3 to EPID=2
[DEBUG] [RFNOC::MGMT] Throttling stream endpoint to 100% (0x0)
[DEBUG] [RFNOC::MGMT] Setup a stream from EPID=3 to EPID=2
[DEBUG] [RFNOC::MGMT] Programmed a route from EPID=3 to EPID=2
[DEBUG] [RFNOC::MGMT] Throttling stream endpoint to 100% (0x0)
[DEBUG] [RFNOC::MGMT] Setup a stream from EPID=3 to EPID=2
[DEBUG] [RFNOC::GRAPH] Data stream between EPID 3 and EPID 2 established where downstream buffer can hold 524287 bytes and 16777215 packets

wordimont, 2024-11-05:
If you're looking for support, I suggest you reach out to Ettus/NI support and utilize the mailing list. I don't know if the issue you referenced is a bug or not, but it seems like you have some questions and you're more likely to get an answer from the mailing list than the issue tracker, which is mainly used to report bugs.