EttusResearch/uhd #813: rfnoc_modtool create and add requires some work after

zacaric, 2024-11-20:
## Issue Description ##

I am using `rfnoc_modtool` to create an oot module called `core` and a block called `square` (squares the IQ data). 

When doing this, there are a few issues that causes things to not work right away and requires changes right out of the box (See Actual Behavior to see what changes are required to get working). Some of the items in the list I can understand needing to do, but other items I feel should have been handled by `rfnoc_modtool` when generating the files.

## Setup Details ##

`UHD 4.7.0.0-149-g635ad362`, `Ubuntu 22.04`, `Vivado 2021.2 AR76780`

## Expected Behavior ##

Expected for files generated from `rfnoc_modtool` to work out of the box with little modification. Also expected adding an icore file would also work out of the box. Little modification is understandable, but some documentation on that would be helpful.

## Actual Behavior ##

The following files require modification:
- `rfnoc-core/CMakeLists.txt` needs `find_package (Python3 COMPONENTS Interpreter Development)` added (I added it after line 43 and it worked)
- `rfnoc-core/python/square_block_control_python.hpp` needs to have line 12 changed from `using namespace rfnoc::gain;` to `using namespace rfnoc::core;`
- `rfnoc-core/fpga/core/CMakeLists.txt` needs `add_subdirectory(rfnoc_block_square)` added
- Create the file `rfnoc-core/fpga/core/rfnoc_block_square/CMakeLists.txt` and add the following:
    ```cmake
    #
    # Copyright 2019 Ettus Research, a National Instruments Brand
    #
    # SPDX-License-Identifier: GPL-3.0-or-later
    #

    # This macro will tell CMake that this directory contains an RFNoC block. It
    # will parse Makefile.srcs to see which files need to be installed, and it will
    # register a testbench target for this directory.
    RFNOC_REGISTER_BLOCK_DIR()

    # This will do the same, but it will skip the testbench target.
    #RFNOC_REGISTER_BLOCK_DIR(NOTESTBENCH)
    ```
- `rfnoc-core/fpga/core/rfnoc_block_square/Makefile.srcs` needs the file extenstions to be change from `*.v` to `*.sv`
- `rfnoc-core/icores/CMakeLists.txt` requires `RFNOC_REGISTER_BLOCK` to be changed to `RFNOC_REGISTER_IMAGE_CORE`

Even after doing this, I am still getting the following error:
```
⛔   Error evaluating square0: File fpga/square/rfnoc_block_square/Makefile.srcs not found in include paths: ['/workspaces/rfnoc-project-example/src/oot/rfnoc-core', '/usr/local/share/uhd/rfnoc']
⛔   Image configuration contains issues: Skipping build. Use --ignore-warnings to build despite warnings.
```

## Steps to reproduce the problem ##

```bash
rfnoc_modtool create core
cd rfnoc-core
rfnoc_modtool add square

# Next, do these steps
# 1. copy x310_rfnoc_image_core.yml from the gain example
# 2. change any instance of "gain" to "square"

mkdir build
cd build
cmake -DUHD_FPGA_DIR=/uhd/fpga ../
make # This fails require files to be modified
make install
make x310_rfnoc_image_core # This fails requiring even more files to be modified
```

## Additional Information ##

I am running all of this in a docker image (using Ubuntu 22.04 as the base). I was able to get the rfnoc_gain example working in this docker image.

mbr0wn, 2024-11-26:
@zacaric Thanks for this super helpful report. FYI, we generate the OOT sources from `rfnoc-gain`, so that explains some of the issues, but not all of them. We're investigating and will provide a fix soon.

zacaric, 2024-11-26:
Another modification I had to make afterwards was to add a comma on line 182 of `rfnoc-core/fpga/core/rfnoc_block_square/rfnoc_block_square.sv`. I will add this to the description as well to keep these needed modifications in one place

mbr0wn, 2024-11-26:
@zacaric do you have f0066cf16d3050c904edef8b5b70a7e2a34cd2ec? That should take care of the noc_shell fixes.

zacaric, 2024-11-26:
It seems like I do have those changes. I have my docker image setup to grab the latest code on the `master` branch every time it gets built and I rebuilt the image yesterday. Do it should be running to most up to date version. 

I also found some errors with double quotes (`"`) not being around `HDL_IP` in a couple of the files. I added that to the issue description too

mbr0wn, 2024-11-28:
@zacaric I hope you don't if I edit your list into a checklist so I can keep track of things that I've locally fixed.

closed 2024-12-10 by joergho