EttusResearch/uhd #815: uhd returns wrong path to image downloader

JustusBab, 2024-11-27:
## Issue Description ##
When using `uhd_usrp_probe`, I got the following error:
```
Error: RuntimeError: Expected FPGA compatibility number 39.0, but got 38.0:
The FPGA image on your device is not compatible with this host code build.
Download the appropriate FPGA images for this version of UHD.
As an Administrator, please run:

"C:\Program Files\UHD\bin\uhd\utils\uhd_images_downloader.py"
```
However, the location of the `uhd_images_downloader.py` is actually in `C:\Program Files\UHD\lib\uhd\utils\uhd_images_downloader.py`.
The correct path is used in the [wiki](https://uhd.readthedocs.io/en/latest/page_images.html) under UHD Image Downloader.

Otherwise, the downloading and burning of the new image works. Just this path in the error message is wrong.

## Setup Details ##
`UHD_4.6.0.0-0-g0c4637ab`, `Windows 11 22H2`

## Expected Behavior ##
Returns the correct path in the error message.

## Actual Behavior ##
Returns the wrong path in the error message.

joergho, 2024-11-29:
Thanks for reporting, @JustusBab! We will address this in the upcoming release of UHD.

joergho, 2025-02-07:
The fix will be in public `master` soon. Unfortunately the fix did not make it into the UHD 4.8.0.0 release but it will be part of the next release.

closed 2025-02-07 by joergho