EttusResearch/uhd #816: [Feature request] Adding support of 2.5Gb Ethernet PHY

ohault, 2024-11-28:
If UHD FPGA gateware could be updated to support 2.5Gb Ethernet PHY, this would be very nice for smaller labs.

mbr0wn, 2024-12-13:
Hi @ohault, this is very low on our priority list. 2.5GbE may be becoming more common, but it's not even fast enough for many of our devices (at sc16 samples, it can maybe sustain 75 Msps). Also, there are no free Xilinx PHY cores we can use.

ohault, 2024-12-13:
Hi @mbr0wn , you will find some motiviation at https://github.com/enjoy-digital/liteeth.
This question introduce another one, even more interresting about LiteX FPGA developpments for Etthus Research devices. USRP family could perhaps join https://github.com/litex-hub/litex-boards ? :-)

ohault, 2024-12-13:
@mbr0wn, I never ask about switching your entire archictecture, but only the opportunity to "sponsor" the fact that Etthus Research boards and devices could also be supported by LiteX framework. This coud be a great community opportunity/initiative for Etthus Research as SDR are starting to become more ubiticious. Feel free to contact Florent from [Florent. ](http://www.enjoy-digital.fr/)

mbr0wn, 2025-06-24:
B2xx series, no. The FPGAs (Spartan-6) are different and porting RFNoC to those would not only be significant work, it would also change the way these devices work and that would break thousands of people's use cases. So while I'm personally a bit sad we can't have "RFNoC-Lite-for-B2xx", I think many people are also happy that we're not breaking their systems.

3rd party, sure, it's possible and has been done (e.g., here: https://github.com/ptrkrysik/uhd for ZCU111). It's not supported by us, though.